zephyrproject-rtos/zephyr #62218: icm42605: kernel panics when reconfiguring NVIC on STM32H747

mgritz, 2023-09-01:
**Describe the bug**
We are using an STM32H747 with an ICM42605 connected to SPI5 and PC13 for the IRQ line, both managed by the CM7 of that asymmetric dualcore MCU.

During boot and initialization of both cores we randomly encounter kernel panics originating from the icm42605_trigger.c trying to turn the GPIO interrupt off an on whenever an interrupt occurred.
The gpio driver is unsuccessfully pending on the according STH32H7 hardware semaphore to access the NVIC in this case.
The hangup happens for about 2-3 in 10 resets and seems to be a race condition between the two cores starting up, with the IMU driver already trying to service interrupts while the other core is still configuring its peripherals.

A successful workaround is to simply keep the GPIO interrupt enabled all the time as I put into the appended patch for zephyr v3.3.0.

**To Reproduce**
* This should be reproducible using any STM32H7 dualcore board, having applications on both cores that both configure any reasonable amount of chip peripherals or otherwise use the NVIC.
* In this scenario, enable the `ICM42605_TRIGGER` variant of the driver.
* Configure the IMU to generate a FIFO-threshold-interrupt about every 80ms.

**Expected behavior**
STM32 boots and starts servicing the IMU interrupt.

**Impact**
This was a showstopper for us.

**Environment (please complete the following information):**
 - OS: Debian12
 - Toolchain: arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205
 - 07c6af3b8c35c1e49186578ca61a25c76e2fb308 (zephyr-v3.3.0)

mgritz, 2023-09-01:
Here's the patch I propose as a fix:

```patch
From 2c20efa544e2e77021c4ed82fc4baa925fa2d438 Mon Sep 17 00:00:00 2001
From: Martin Gritzan <martin.gritzan@gmail.com>
Date: Fri, 1 Sep 2023 14:59:39 +0200
Subject: [PATCH] drivers: sensors: icm42605: keep GPIO interrupt on

Do not un- and re-register the GPIO interrupt for the IMU's IRQ line for
each received interrupt.

On an STM32H747 (dualcore) MCU reconfiguring the NVIC all the time is
harmful as it requires acquiring the according HSEM, which messes with
timing and sometimes even times out if the other core is doing its
initialization.

In the situation at hand the IRQ can just remain enabled. If the IRQ is
asserted a second time before the icm42605_thread can service the
gpio_sem the semaphore is given again, causing the thread to process it
a second time and so on.

---
 drivers/sensor/icm42605/icm42605_trigger.c | 8 --------
 1 file changed, 8 deletions(-)

diff --git a/drivers/sensor/icm42605/icm42605_trigger.c b/drivers/sensor/icm42605/icm42605_trigger.c
index 08da99e40e..a2f584d94b 100644
--- a/drivers/sensor/icm42605/icm42605_trigger.c
+++ b/drivers/sensor/icm42605/icm42605_trigger.c
@@ -62,20 +62,14 @@ static void icm42605_gpio_callback(const struct device *dev,
 {
 	struct icm42605_data *drv_data =
 		CONTAINER_OF(cb, struct icm42605_data, gpio_cb);
-	const struct icm42605_config *cfg = drv_data->dev->config;
-
 	ARG_UNUSED(pins);
 
-	gpio_pin_interrupt_configure_dt(&cfg->gpio_int, GPIO_INT_DISABLE);
-
 	k_sem_give(&drv_data->gpio_sem);
 }
 
 static void icm42605_thread_cb(const struct device *dev)
 {
 	struct icm42605_data *drv_data = dev->data;
-	const struct icm42605_config *cfg = dev->config;
-
 	if (drv_data->data_ready_handler != NULL) {
 		drv_data->data_ready_handler(dev,
 					     &drv_data->data_ready_trigger);
@@ -85,8 +79,6 @@ static void icm42605_thread_cb(const struct device *dev)
 	    drv_data->double_tap_handler != NULL) {
 		icm42605_tap_fetch(dev);
 	}
-
-	gpio_pin_interrupt_configure_dt(&cfg->gpio_int, GPIO_INT_EDGE_TO_ACTIVE);
 }
 
 static void icm42605_thread(void *p1, void *p2, void *p3)
-- 
2.39.2
```

erwango, 2023-09-01:
@mgritz Thanks for reporting this issue and the proposed patch. We don't take patch, but maybe you'll be able to [submit a Github Pull Request ](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request)?
See https://docs.zephyrproject.org/latest/contribute/guidelines.html

mgritz, 2023-09-01:
@erwango Thanks for the quick reply! Sure thing, will do within next week.

mgritz, 2023-09-06:
PR https://github.com/zephyrproject-rtos/zephyr/pull/62303 is on for this.

erwango, 2023-11-06:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/64075

closed 2023-11-06 by erwango