zephyrproject-rtos/zephyr #62228: RFC: Definition and support for no-erase flag in flash definitions

de-nordic, 2023-09-01:
## Introduction

There is wide range of SPI devices that have SPI NOR compatibility, in command set and operation, that do not require erase,
for example Fujitsu FRAM and ReRAM devices.
These devices work perfectly fine with SPI NOR implementations we have in Zephyr, except that they do not understand erase commands.
I have been testing some of these devices and, after blocking or emulating erase, I was able to successfully run some of tests and samples normally designated for flash devices (where SPI NOR is supported).

### Problem description

Erase is required only by devices that need it to reset to bias (0 or 1) and takes time.
Zephyr subsystems currently assume that erase is always needed, but being able to pass information on whether it is really needed or not should allow us to improve these systems or provide optimizations like for example only emulate erase on minimal needed size to terminate writes.
We have to start somewhere, to be able to figure out that device does not require erase, before we can make our subsystems work with such devices.

### Proposed change

My proposed change, for beginning, is implementation of `flash_get_info_word` API call that should allow to return device flags, and no-erase has been first such flag.
I propose also to add `no-erase` DTS bolean flag to JEDEC SPI NOR definition to allow indicating that instance of a device does not require or provide erase.

### Proposed change (Detailed)

Proposed definition of `no-erase` flag for DTS device definitions.
Proposed addition of `flash_get_info_word` API call allowing to obtain device flags, with the first flag defined:
`FLASH_INFO_WORD_F_NO_ERASE`
Proposed addition of `FLASH_NEEDS_ERASE_BEFORE_WRITE` macro that allows to check whether device requires erase.
The `flash_get_info_word` takes single parameter which is `const struct device` pointer and returns the info word as `uint32_t`.

### Dependencies

Currently none.

### Concerns and Unresolved Questions

Should emulated erase be provided by such devices?
I have no QSPI device to check for QSPI compatibility, only SPI 0 and 3 devices.

## Alternatives

Not supporting deices or moving them to another compatible which would basically duplicate code at this point.

### Draft of proposed changes

Here is PR https://github.com/zephyrproject-rtos/zephyr/pull/62227 that is proposing required changes.
The PR at this point is not complete as only the SPI NOR driver has been given the `flash_get_info_word` implementation.

Laczen, 2023-09-02:
@de-nordic If these devices do not have an erase, would it not be simpler to threat them as eeprom devices?

Laczen, 2023-09-06:
@de-nordic, I don't think it is a good idea to define these devices as flash devices.

There are cases where a lot of extra code will be required to get the wanted behavior. As an example suppose an image is stored on such a device and mcuboot is started to do the swap. The swap state will be written to the image, but to be valid all previous states need to be removed. To catch this it will be needed to retrieve the no-erase flag property, fake erase the area, ...

When using flash we are making sure that we have erased the area before we write (to avoid an error), that an area is erased allows us to take certain decisions. When we don't need to make sure that we have erased the area it will be needed to go over every flash_write, see whether it makes some assumptions and if it does adapt the code.

Laczen, 2023-09-17:
It seems that a similar issue (multiple devices use spi in a similar matter) existed in linux, this was solved by introducing a spi-mem layer.

henrikbrixandersen, 2024-01-23:
Architecture WG meeting, 2024-01-23:
- @de-nordic is looking for guidance if the suggested solution in this PR is an acceptable path forward.
- As a few different solutions to handling flash with/without erase has already been proposed and more or less rejected, it would be nice to hear from other silicon vendors how this meets their hardware capabilities.