zephyrproject-rtos/zephyr #62233: CPU Die Temperature polling sample broken for STM32F4 boards after bump to v3.4.0

maxmclau, 2023-09-02:
**Describe the bug**
The stm32-temp driver appears to be returning incorrect temperatures after bumping from Zephyr v3.3.0 to v3.4.0. I have a few custom boards using various STM32F411s and STM32F412s that are reporting -184 for die_temp after the version bump. I was able to recreate the error by build and running _samples/sensor/die_temp_polling_ for a **blackpill_f411ce**. I have not been able to narrow down what commit may have caused this.

**Impact**
This is an annoyance for my use case, but could be more serious for others.

**Logs and console output**
CPU Die temperature[dietemp]: -184.3 °C

**Environment (please complete the following information):**
Zephyr 3.4.0

benjaminbjornsson, 2023-09-04:
@maxmclau STM32F40x and STM32F41x are using the wrong ADC channel (should be 16 not 18 according to RM0090 13.10).

maxmclau, 2023-09-04:
@benjaminbjornsson strange, channel 18 is the channel used in Zephyr v3.2.0 which appears to be working. I do see that note in the reference manual saying otherwise though. I've also confirmed that vbat is disabled on my board. Using a debugger to break right after the `adc_read()` within `stm32_temp_sample_fetch()` it appears both the working and non-working builds are using the exact same **adc_cfg** and **adc_seq** to make readings.
### Zephyr v3.2.0
[image]

### Zephyr v3.3.0
[image]

You can see the reading loaded into `data->sample_buffer` is very different between builds. Stranger still I've been able to confirm that the sample being read into the buffer does correlate to temperature, though it's a negative correlation reading well below -200 if I apply heat to the IC. I'll continue poking around.

maxmclau, 2023-09-04:
Managed to git bisect down to https://github.com/zephyrproject-rtos/zephyr/commit/ac7f2dad4ecba95ace2bc8473ff630e7707bf171 which appears to be the offending commit.  I don't know the relationship between the adc vbat-channel and the vbat io-channel, but commenting out this line appears to fix the temp-sensor regression. I'm sure it removes vbat sensing functionality as well though.
https://github.com/zephyrproject-rtos/zephyr/blob/5ba4288fea6dd38159d500613d08c8e8967d7239/dts/arm/st/f4/stm32f4.dtsi#L517

I'm hoping someone more familiar with the STM32 ADC configuration can swoop in and lend a hand.

Another thing to note - the commit also updates the `dietemp > io-channels` phandle from 16 to 18. Seems like there may be an error with the ST reference manual, because 18 appears to be correct.

benjaminbjornsson, 2023-09-04:
> Managed to git bisect down to [ac7f2da](https://github.com/zephyrproject-rtos/zephyr/commit/ac7f2dad4ecba95ace2bc8473ff630e7707bf171) which appears to be the offending commit. I don't know the relationship between the adc vbat-channel and the vbat io-channel, but commenting out this line appears to fix the temp-sensor regression. I'm sure it removes vbat sensing functionality as well though.
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/5ba4288fea6dd38159d500613d08c8e8967d7239/dts/arm/st/f4/stm32f4.dtsi#L517
> 
> I'm hoping someone more familiar with the STM32 ADC configuration can swoop in and lend a hand.
> 
> Another thing to note - the commit also updates the `dietemp > io-channels` phandle from 16 to 18. Seems like there may be an error with the ST reference manual, because 18 appears to be correct.

Looks as if the common path configuration is applied for vbat as long as the vbat-channel property exists in the adc node:

https://github.com/zephyrproject-rtos/zephyr/blob/5ba4288fea6dd38159d500613d08c8e8967d7239/drivers/adc/adc_stm32.c#L659

benjaminbjornsson, 2023-09-05:
IMO It would make more sense to have the different sensor drivers do the setup / teardown.

erwango, 2023-09-05:
@gautierg-st Can you have a check?

gautierg-st, 2023-09-06:
I'm looking into it

closed 2023-09-22 by carlescufi

maxmclau, 2023-09-22:
thanks @gautierg-st!

PR #62345 (2023-09-22): Fix STM32F4 ADC temperature and Vbat measurement
This PR fixes a problem with the measurement of the temperature on STM32F4 due to the sharing of the channel with the Vbat sensor.

It moves the setting of the sensor internal channels in the corresponding sensor driver instead of doing it in the ADC driver.

Fixes #62233