zephyrproject-rtos/zephyr #62258: drivers: serial: STM32L0 LPUART does not support interrupt

nguyenmthien, 2023-09-04:
**Describe the bug**
The LPUART driver for the STM32L0 line does not support interrupt-based communication. 

**To Reproduce**
We will investigate the `nucleo_l073rz` board with the `echo_bot` example:

Steps to reproduce the behavior:
1. Create `nucleo_l073rz.overlay` in `zephyr/samples/drivers/uart/echo_bot`: 

    ```
    / {
        chosen {
            zephyr,console = &lpuart1;
        };
    };

    &lpuart1 {
        status = "okay";
        pinctrl-0 = <&lpuart1_tx_pc4 &lpuart1_rx_pc5>;
        pinctrl-names = "default";
        current-speed = <115200>;
    };
    ```
3. Build the `echo_bot` example:
    ```
    west build -b nucleo_l073rz zephyr/samples/drivers/uart/echo_bot
    ````
4. Flash the device with `west flash`
5. Connect the PC4 and PC5 to the STLink TX and RX pins
6. Using a serial monitor (e.g. Putty) to see that the device does not ping back what is entered into the serial monitor. 

**Expected behavior**
This example based on UART interrupt for the RX buffer and polling for the TX buffer. 

Without step 1, the device's behavior is very predictable: whatever is typed into the monitor before the return key is pinged back to the UART output. 

With overlay file, the device only have an introduction line and nothing else. 

This only happen because of no support of LPUART interrupt in the HAL. 

**Impact**
This will affect all drivers and applications that use LPUART interrupt on the L0 family. 
We are currently unable to confirm if this is local to the L0 line alone. 

**Environment (please complete the following information):**
 - Our build environment is the lastest version of the CI Zephyr image. (fe38bd9)
 - Toolchain: Zephyr SDK (0.16.1)
 - Commit used: 5c901052ec2a105594a91e33b8fb43d40c34b88f

Desvauxm-st, 2023-09-06:
@nguyenmthien

Hello
I try to reproduce the issue but I can’t ( or I did something wrong)

for me is not easy to connect the PC4 and PC5 to the STLink TX and RX pins

I prefer to use the console .
The console is link to PA2 and PA3 on this board instead of PC4 and PC5.

So I replace pin like that

pinctrl-0 = <&lpuart1_tx_pc4 &lpuart1_rx_pc5>;
new:
pinctrl-0 = <&lpuart1_tx_pa2 &lpuart1_rx_pa3>;

I give you the nucleo_l073rz.overlay I used

Is like that:

/ {
    chosen {
        zephyr,console = &lpuart1;
        zephyr,shell-uart = &lpuart1;
    };
};

/*
&lpuart1 {
    status = "okay";
    pinctrl-0 = <&lpuart1_tx_pc4 &lpuart1_rx_pc5>;
    pinctrl-names = "default";
    current-speed = <115200>;
};
*/

&lpuart1 {
        status = "okay";
        pinctrl-0 = <&lpuart1_tx_pa2 &lpuart1_rx_pa3>;
        pinctrl-names = "default";
        current-speed = <115200>;
};

And in the console the echo works.

see attachement:

[Lpuart_nucleo_l073rz_issue.docx](https://github.com/zephyrproject-rtos/zephyr/files/12535740/Lpuart_nucleo_l073rz_issue.docx)

[image]

[nucleo_l073rz.overlay.txt](https://github.com/zephyrproject-rtos/zephyr/files/12535827/nucleo_l073rz.overlay.txt)

nguyenmthien, 2023-09-11:
Hi, 

> hello
> 
> I try your setting
> 
> [image]
> 
> the result is like that
> 
> [image]
> 
> I did like that:
> 
> remove SB13 and SB14.
> 
> connect CN3:TX with CN10:6 connect CN3:rx with CN10:34
> 
> [image]

Hi, I double checked the soldering bridges and it worked now 👍

closed 2023-09-11 by nguyenmthien

Desvauxm-st, 2023-09-12:
good  new  
like this the problem is solved
great