zephyrproject-rtos/zephyr #62282: ieee802154: Build failes when including the file net/ieee802154_radio.h in a cplusplus file.

Maartenwn, 2023-09-05:
**Describe the bug**
Build fails when including the file zephyr/net/ieee802154_radio.h in a cplusplus file.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir net_ieee802154_radio_build_fail; cd net_ieee802154_radio_build_fail
2. west init -m https://github.com/Maartenwn/net_ieee802154_radio_build_fail.git --mr master
3. west update
4. cd net_ieee802154_radio_build_fail
5. west build -b nrf52840dk_nrf52840 -p
7. See error
8. change target_sources from main.cpp to main.c
9. west build 
10. See no error

**Expected behavior**
Including this header in a cplusplus file should compile.

**Impact**
Because of this issue we either have to fork zephyr or compile the file with fPermissive.

**Logs and console output**
`zephyr/include/zephyr/net/ieee802154_pkt.h:95:26: error: invalid conversion from 'void*' to 'net_pkt_cb_ieee802154*' [-fpermissive]
   95 |         return net_pkt_cb(pkt);
      |                ~~~~~~~~~~^~~~~
      |                          |
      |                          void*
`

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 8350fc2c63b10836bf3bf18a55f692fc6077e864

fabiobaltieri, 2023-09-05:
cc @cfriedt

cfriedt, 2023-09-05:
> @cfriedt @Maartenwn Already got a (trivial) fix - see https://github.com/fgrandel/zephyr/commits/fix/62282-ieee802154-pkt-cpp. Just cannot create a PR right now as github seems to have a problem. Will try later again.

I noticed GitHub was weird today - thanks Florian!

Maartenwn, 2023-09-06:
Yes that solves the problem, thank you for fixing it so quickly.

PR #62314 (2023-09-06): net: l2: ieee802154: pkt: fix cpp build failure
ieee802154_pkt.h: Add cast to support CPP.

Fixes: #62282
commit 718fe844 (2023-09-06): net: l2: ieee802154: pkt: fix cpp build failure

CPP needs an additional cast.

Fixes: #62282
closed 2023-09-06 by carlescufi