zephyrproject-rtos/zephyr #62327: Add documentation for new modem subsystem

kartben, 2023-09-06:
**Is your enhancement proposal related to a problem? Please describe.**

The recently added modem subsystem does not show up in the documentation, making it potentially difficult to know of its existence in the first place, and to learn how to use it for folks who'd be looking for that information.

**Describe the solution you'd like**

* Add documentation in OS Services section of the documentation
* The Doxygen documentation for chat, cmux, etc API, needs to be completed, in particular add groups and make sure they are "mounted" properly in a relevant parent group (`connectivity`?)
   * Please make sure that internal stuff is flagged appropriately (`@cond INTERNAL_HIDDEN`) so that it does not show up in the generated docs, and that all other public API has documentation.
* Update the cellular_modem sample using the change introduced in #62029 (when/if merged) so that it references the doxygen group(s) that are relevant.

**Describe alternatives you've considered**

N/A

**Additional context**

Sligthly related: function that have multiple return values should use `@retval`, not `@returns` in their Doxygen docstring.

Thanks!

bjarki-trackunit, 2023-09-11:
Just letting you know I have indeed seen the issue and plan to to start work on the it within a couple of weeks :)

bjarki-trackunit, 2023-10-14:
In progress #63949

bjarki-trackunit, 2023-10-19:
@kartben Did you have some more nits before we close this issue? :)

kartben, 2023-10-19:
Ahah no, thanks, this is great :)

closed 2023-10-19 by kartben