zephyrproject-rtos/zephyr #62330: tests: net: lib: lwm2m: content_plain_text: net.lwm2m.content_plain_text fails

katgiadla, 2023-09-06:
**Describe the bug**
The test `net.lwm2m.content_plain_text` fails.

Observed for `nrf52840dk_nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/net/lib/lwm2m/content_plain_text -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-3264-g5ba4288fea6d ***
Running TESTSUITE net_content_plain_text
===================================================================
START - test_get_bool
PASS - test_get_bool in 0.001 seconds
===================================================================
START - test_get_float
PASS - test_get_float in 0.001 seconds
===================================================================
START - test_get_objlnk
PASS - test_get_objlnk in 0.001 seconds
===================================================================
START - test_get_opaque
PASS - test_get_opaque in 0.001 seconds
===================================================================
START - test_get_s32
PASS - test_get_s32 in 0.001 seconds
===================================================================
START - test_get_s64
PASS - test_get_s64 in 0.001 seconds
===================================================================
START - test_get_string
PASS - test_get_string in 0.001 seconds
===================================================================
START - test_put_bool
PASS - test_put_bool in 0.001 seconds
===================================================================
START - test_put_float
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/net/lib/lwm2m/content_plain_text/src/main.c:231: net_content_plain_text_test_put_float: (ret not equal to strlen(expected_payload[i]))
Invalid length returned
FAIL - test_put_float in 0.021 seconds
===================================================================
START - test_put_objlnk
PASS - test_put_objlnk in 0.001 seconds
===================================================================
START - test_put_s16
PASS - test_put_s16 in 0.001 seconds
===================================================================
START - test_put_s32
PASS - test_put_s32 in 0.001 seconds
===================================================================
START - test_put_s64
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/net/lib/lwm2m/content_plain_text/src/main.c:171: net_content_plain_text_test_put_s64: (ret not equal to strlen(expected_payload[i]))
Invalid length returned
FAIL - test_put_s64 in 0.021 seconds
===================================================================
START - test_put_s8
PASS - test_put_s8 in 0.001 seconds
===================================================================
START - test_put_string
PASS - test_put_string in 0.001 seconds
===================================================================
TESTSUITE net_content_plain_text failed.
Running TESTSUITE net_content_plain_text_nodata
===================================================================
START - test_get_bool_nodata
PASS - test_get_bool_nodata in 0.001 seconds
===================================================================
START - test_get_float_nodata
PASS - test_get_float_nodata in 0.001 seconds
===================================================================
START - test_get_objlnk_nodata
PASS - test_get_objlnk_nodata in 0.001 seconds
===================================================================
START - test_get_opaque_nodata
PASS - test_get_opaque_nodata in 0.001 seconds
===================================================================
START - test_get_s32_nodata
PASS - test_get_s32_nodata in 0.001 seconds
===================================================================
START - test_get_s64_nodata
PASS - test_get_s64_nodata in 0.001 seconds
===================================================================
START - test_get_string_nodata
E: Buffer too small to accommodate string, truncating
PASS - test_get_string_nodata in 0.005 seconds
===================================================================
TESTSUITE net_content_plain_text_nodata succeeded
Running TESTSUITE net_content_plain_text_nomem
===================================================================
START - test_put_bool_nomem
PASS - test_put_bool_nomem in 0.001 seconds
===================================================================
START - test_put_float_nomem
PASS - test_put_float_nomem in 0.001 seconds
===================================================================
START - test_put_objlnk_nomem
PASS - test_put_objlnk_nomem in 0.001 seconds
===================================================================
START - test_put_s16_nomem
PASS - test_put_s16_nomem in 0.001 seconds
===================================================================
START - test_put_s32_nomem
PASS - test_put_s32_nomem in 0.001 seconds
===================================================================
START - test_put_s64_nomem
PASS - test_put_s64_nomem in 0.001 seconds
===================================================================
START - test_put_s8_nomem
PASS - test_put_s8_nomem in 0.001 seconds
===================================================================
START - test_put_string_nomem
PASS - test_put_string_nomem in 0.001 seconds
===================================================================
TESTSUITE net_content_plain_text_nomem succeeded
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  86.67% [net_content_plain_text]: pass = 13, fail = 2, skip = 0, total = 15 duration = 0.055 seconds
- PASS - [net_content_plain_text.test_get_bool] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_float] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_objlnk] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_opaque] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_s32] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_s64] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_get_string] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_put_bool] duration = 0.001 seconds
- FAIL - [net_content_plain_text.test_put_float] duration = 0.021 seconds
- PASS - [net_content_plain_text.test_put_objlnk] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_put_s16] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_put_s32] duration = 0.001 seconds
- FAIL - [net_content_plain_text.test_put_s64] duration = 0.021 seconds
- PASS - [net_content_plain_text.test_put_s8] duration = 0.001 seconds
- PASS - [net_content_plain_text.test_put_string] duration = 0.001 seconds
SUITE PASS - 100.00% [net_content_plain_text_nodata]: pass = 7, fail = 0, skip = 0, total = 7 duration = 0.011 seconds
- PASS - [net_content_plain_text_nodata.test_get_bool_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_float_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_objlnk_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_opaque_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_s32_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_s64_nodata] duration = 0.001 seconds
- PASS - [net_content_plain_text_nodata.test_get_string_nodata] duration = 0.005 seconds
SUITE PASS - 100.00% [net_content_plain_text_nomem]: pass = 8, fail = 0, skip = 0, total = 8 duration = 0.008 seconds
- PASS - [net_content_plain_text_nomem.test_put_bool_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_float_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_objlnk_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_s16_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_s32_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_s64_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_s8_nomem] duration = 0.001 seconds
- PASS - [net_content_plain_text_nomem.test_put_string_nomem] duration = 0.001 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: d89b0029cff77d77925f7a085676cba5
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.0
- Commit SHA or Version used: zephyr-v3.4.0-3264-g5ba4288fea6d

katgiadla, 2023-09-07:
The same errors occurred for `nrf5340dk_nrf5340_cpunet`.

SeppoTakalo, 2023-09-11:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/62431

I know the title says that it fixes floating points, but it also fixes 64bit numbers as well.

katgiadla, 2023-09-12:
It is fixed. Thanks

closed 2023-09-12 by katgiadla