zephyrproject-rtos/zephyr #62346: boot_write_img_confirmed not working on stm32g474ce

st0ne, 2023-09-06:
**Describe the bug**
After updating the firmware with mcuboot and dfu-util the firmware will start after swapping the images.
But it gets Error -5 on confirming the image.

I am using a custom board with stm32g474ce.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
confirming the new running image should work as expected.

**Impact**
can not update my firmware after deployment of hardware.

**Logs and console output**
With CONFIG_FLASH_LOG_LEVEL_DBG=y i will get the following output:
```
[00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311280, len: 16
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311256, len: 1
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311264, len: 1
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311272, len: 1
 [00:00:03.030,000] <inf> main: Confirming Boot Image
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311280, len: 16
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311256, len: 1
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311264, len: 1
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_read: Read offset: 311272, len: 1
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_write: Write offset: 311272, len: 8
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_write_protection: Disable write protection
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_check_status: Status: 0x000000c0
 [00:00:03.030,000] <dbg> flash_stm32: flash_stm32_write_protection: Enable write protection
```

nordicjm, 2023-09-07:
You have not included the zephyr commit you are using, please update with this information

nordicjm, 2023-09-07:
CC: @de-nordic

st0ne, 2023-09-07:
> You have not included the zephyr commit you are using, please update with this information

added the missing information :-)

erwango, 2023-10-03:
@st0ne Can you provide the flash partition you're using ?
Also, the error you mention doesn't appear clearly in the log. Would you  have a way to provide full extent log ?

st0ne, 2023-10-03:
@erwango i will try to get a better log. The main error is return of -5 when boot_write_img_confirmed is called. 
Therefore the new image will be switched back after reboot.

flash partition:

`&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x0 DT_SIZE_K(96)>;
			read-only;
		};

		slot0_partition: partition@18000 {
			label = "image-0";
			reg = <0x00018000 DT_SIZE_K(208)>;
		};

		slot1_partition: partition@4C000 {
			label = "image-1";
			reg = <0x0004C000 DT_SIZE_K(208)>;
		};

		/*scratch_partition: partition@70000 {
			reg = <0x00070000 DT_SIZE_K(16)>;
		};*/

		/* Set 4Kb of storage at the end of the 512Kb of flash */
		/*storage_partition: partition@70000 {
			label = "storage";
			reg = <0x00070000 DT_SIZE_K(64)>;
		};*/
	};
};`

st0ne, 2023-10-03:
no it is not needed anymore. They have a new swap algorithm which does not wear out the flash at the scratch partition.

st0ne, 2023-10-05:
ah, maybe this could be a reason, i have not used mcumgr. I used to start into the bootloader with a button and then used dfu_util for updating.
Is it in function different than mcumgr?

Reason to do this was because of saving space for the main app.

erwango, 2023-10-12:
Sorry, I won't be able to help on reasons why it is different.
Anyway, I would tend to close the issue as using mcumgr demonstrates mcuboot is working on G4 with the provided flash partition.
@st0ne Thoughts ?

de-nordic, 2023-10-12:
> ah, maybe this could be a reason, i have not used mcumgr. I used to start into the bootloader with a button and then used dfu_util for updating. Is it in function different than mcumgr?
> 
> Reason to do this was because of saving space for the main app.

dfu-util works with USB, mcumgr relies on Serial Recovery feature of MCUboot.
USB support in MCUboot make bigger MCUboot, serial recovery is smaller in size.

You upload image using dfu-util and then confirm it using MCUmgr?

st0ne, 2023-10-12:
I upload the image with dfu-util, which is working. And then after a reboot it should get confirmed when it is running.

The reason for this is, that i have not enough space for my application. In this case, only mcuboot has to do the update and everything elese.

erwango, 2024-01-16:
@st0ne Is there anything more that can be done or can we close the issue ?

st0ne, 2024-02-05:
Hi,
did not have time to work on the project for some time. But i got another idea. STM32G474 is dual bank by default. Could this be the reason why it is not working correctly? Do i have to set DBANK=0 before flashing zephyr?
I may have time to test this next week.

closed 2024-04-21 by github-actions[bot]