zephyrproject-rtos/zephyr #62372: twister build speed optimization

hakehuang, 2023-09-07:
**Is your enhancement proposal related to a problem? Please describe.**
we run `twister -p <platform> --build-only`
which usually requires 50 minutes for 8 core system.

some analysis find the build log output may takes additional time and as we are using the ipc to control the mutl-process, this maybe need optimized

**Describe the solution you'd like**
analyze the build bottleneck and find out a easy way to optimize.

**Describe alternatives you've considered**
make qverage build time equal to simple build

hakehuang, 2023-09-20:
some analysis:
testing with twister `-j 1` with platfrom mimxrt595_evk_cm33, it will takes 5hrs to finish(~710 builds) whereas ~1hr minutes for '-j 4', without ccache. so the parallaization mechanism looks right in twister, not the problems, every build will takes about 30s per cases, in zephyr sdk tool chain.
I have no try the ccache method yet, but I'd like to analysis whether there is any way to improve the build speed alone.

nashif, 2023-09-20:
and how exactly is this a twister problem? Try to build 710 builds with west or cmake alone and see how long it takes... Twister is just a wrapper. if you want to speed thing up with builds, you need to look at the build system, not twister.

hakehuang, 2023-09-20:
> and how exactly is this a twister problem? Try to build 710 builds with west or cmake alone and see how long it takes... Twister is just a wrapper. if you want to speed thing up with builds, you need to look at the build system, not twister.

it is not twister problem, according to my test. I just want to find a way to analysis the build efficiency, by this tools, https://github.com/aras-p/ClangBuildAnalyzer)/aras-p/ClangBuildAnalyzer, and see whether there is a way to improve the build speed.

hakehuang, 2023-09-25:
with below step, I eanble the ClangBuildAnalyzer and result as below
```

$ export ZEPHYR_TOOLCHAIN_VARIANT=llvm
$ cd to <zephyr>/tests/benchmarks/latency_measure
$ ClangBuildAnalyzer --start /home/shared/disk/zephyr_project/buildanalyzer_folder/
$ west build -b native_posix -- -DEXTRA_CFLAGS=-ftime-trace
$ find build/ -name *.json -exec cp {} /home/shared/disk/zephyr_project/buildanalyzer_folder/ \;
$ ClangBuildAnalyzer --stop /home/shared/disk/zephyr_project/buildanalyzer_folder/ /home/shared/disk/zephyr_project/buildanalyzer_folder/ana_result
$ ClangBuildAnalyzer --analyze /home/shared/disk/zephyr_project/buildanalyzer_folder/ana_result

```

```
ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure$ ClangBuildAnalyzer --analyze /home/shared/disk/zephyr_project/buildanalyzer_folder/ana_result
Analyzing build trace from '/home/shared/disk/zephyr_project/buildanalyzer_folder/ana_result'...
**** Time summary:
Compilation (143 times):
  Parsing (frontend):            5.1 s
  Codegen & opts (backend):      2.9 s

**** Files that took longest to parse (compiler frontend):
   135 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//bitarray.c.json
   126 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//log_msg.c.json
   123 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//fdtable.c.json
   112 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//sched.c.json
   109 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//heap_malloc_free.c.json
   102 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//cbprintf_complete.c.json
    99 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//log_core.c.json
    96 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//mutex.c.json
    94 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//log_output.c.json
    92 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//init.c.json

**** Files that took longest to codegen (compiler backend):
   227 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//cbprintf_complete.c.json
   139 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//sched.c.json
   132 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//cbprintf_packaged.c.json
   131 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//rb.c.json
   117 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//heap.c.json
   104 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//heap-validate.c.json
    96 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//log_core.c.json
    93 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//work.c.json
    90 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//bitarray.c.json
    89 ms: /home/shared/disk/zephyr_project/buildanalyzer_folder//log_output.c.json

**** Functions that took longest to compile:
    62 ms: z_cbvprintf_impl (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/cbprintf_complete.c)
    28 ms: _ConfigAbsSyms (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/build/zephyr/misc/generated/configs.c)
    27 ms: cbvprintf_package (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/cbprintf_packaged.c)
    22 ms: cbprintf_package_convert (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/cbprintf_packaged.c)
    20 ms: rb_remove (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/rb.c)
    19 ms: log_cache_init (/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/logging/log_cache.c)
    18 ms: native_handle_cmd_line (/home/shared/disk/zephyr_project/zephyr_test/zephyr/boards/posix/native_posix/cmdline.c)
    16 ms: idle (/home/shared/disk/zephyr_project/zephyr_test/zephyr/kernel/idle.c)
    16 ms: log_output_process (/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/logging/log_output.c)
    14 ms: z_get_fd_obj (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/fdtable.c)
    13 ms: sys_heap_aligned_realloc (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/heap.c)
    12 ms: sys_heap_validate (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/heap-validate.c)
    10 ms: mutex_lock_unlock (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/mutex_lock_unlock.c)
    10 ms: sema_test_signal (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/sema_test_signal_release.c)
     9 ms: heap_print_info (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/heap-validate.c)
     9 ms: thread_switch_yield (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/thread_switch_yield.c)
     8 ms: sys_heap_stress (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/heap-validate.c)
     8 ms: sys_heap_aligned_alloc (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/heap.c)
     8 ms: suspend_resume (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/thread.c)
     8 ms: heap_malloc_free (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/heap_malloc_free.c)
     7 ms: rb_insert (/home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/os/rb.c)
     7 ms: coop_ctx_switch (/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/benchmarks/latency_measure/src/coop_ctx_switch.c)
[... cut ...]
156 ms: /home/shared/disk/zephyr_project/zephyr_test/zephyr/kernel/include/kernel_internal.h (included 19 times, avg 8 ms), included via:
  14x: ksched.h
  3x: <direct include>
  1x: kswap.h ksched.h
  1x: kernel_offsets.h

151 ms: /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/logging/log_core.h (included 13 times, avg 11 ms), included via:
  6x: log.h
  2x: log_internal.h
  2x: log_ctrl.h log_internal.h
  1x: stack.h log.h
  1x: tc_util.h log_ctrl.h log_internal.h
  1x: <direct include>
  ...

  done in 0.0s.

```