zephyrproject-rtos/zephyr #62380: CI broken: samples/tfm_integration has invalid twister settings

benediktibk, 2023-09-07:
**Describe the bug**
CI currently fails due to faulty twister settings in samples/tfm_integration:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/6106995843/job/16573120789?pr=60323

**To Reproduce**
Trigger rebuild for one of these PRs:
https://github.com/zephyrproject-rtos/zephyr/pull/60328
https://github.com/zephyrproject-rtos/zephyr/pull/60347
https://github.com/zephyrproject-rtos/zephyr/pull/60323

**Expected behavior**
The builds should succeed.

**Impact**
I am unable to run the CI pipeline successfully at the moment.

**Logs and console output**

    INFO: Error found: samples/tfm_integration/psa_protected_storage/sample.tfm.protected_storage on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_ipc/sample.tfm_ipc on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_protected_storage on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_internal_trusted_storage on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_crypto on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)
    INFO: Error found: samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition on mps2_an521_ns (Excluded tags per platform (only_tags) but is one of the integration platforms)

nordicjm, 2023-09-07:
@PerMac any clue how to resolve this?

martinjaeger, 2023-09-07:
Probably something in #62123 is the root cause.

benediktibk, 2023-09-07:
#62381 has the same issue

aescolar, 2023-09-07:
It is just the same type of issue as https://github.com/zephyrproject-rtos/zephyr/issues/62215

PerMac, 2023-09-07:
I can provide a proper fix rather quick. I believe I know what's the issue. Quick fix won't be needed then.

PerMac, 2023-09-07:
The reason:
- test plan is generated by twister with --integration arg. It means, when applicable, platforms from integration_platforms in yamls are selected as a scope. In addition, their definition is that they are the minimal scope that should work -> skips are not allowed on integration platform (only CLI dependent are allowed) and causing errors.
-  some platforms have limitations defined in their yamls. "only_tag" entry should limit the usage and causes skips if such platform is selected for a test which doesn't have a given tag
- New tag was added and test plan is optimizing the execution scope in/excluding tests with a given tag. 
- mps2_an521_ns doesn't have this tag in its "tag allow" list https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/mps2_an521/mps2_an521_ns.yaml#L14
- this causes the platform to be skipped, and with --integration and it being an integration+platform causing an error

Solutions:
1.) remove mps2_an521_ns  from integration_platforms in failing tests
2.)  add "trusted-firmware-m" tag to "only_tags" in the board definition

I opt for 2

sjanc, 2023-09-07:
Despite #62385 this stil seems to be failing,  eg  https://github.com/zephyrproject-rtos/zephyr/pull/61844

aescolar, 2023-09-07:
Yes, Unfortunately there is another 2 issues apart from that quick fix.
@PerMac "mcuboot" is also required after that PR 
So I would go with 1) as this sample seems to not actually build for this integration platform(?)

nordicjm, 2023-09-07:
The point of the PR was to run the TF-M tests when mcuboot was updated and mps2_an521_ns is a supported board that should be built

aescolar, 2023-09-07:
> The point of the PR was to run the TF-M tests when mcuboot was updated and mps2_an521_ns is a supported board that should be built

That is ok, but the problem was that the PR forgot those tags are also used to filter boards. They tags are listed in the boards yaml as the capabilities they support, and filtered by that

PerMac, 2023-09-07:
I linked a fix. With it I got:
```
12:20 $ scripts/twister -p mps2_an521_ns --integration -T samples/tfm_integration/ -v
ZEPHYR_BASE unset, using "/home/maciej/zephyrproject/zephyr"
Renaming output directory to /home/maciej/zephyrproject/zephyr/twister-out.21
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3384-g46306aad7907
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/maciej/zephyrproject/zephyr/twister-out/testplan.json
INFO    - JOBS: 4
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  4/12 mps2_an521_ns             samples/tfm_integration/psa_protected_storage/sample.tfm.protected_storage PASSED (qemu 2.684s)
INFO    -  5/12 mps2_an521_ns             samples/tfm_integration/tfm_secure_partition/sample.tfm.secure_partition PASSED (qemu 2.736s)
INFO    -  6/12 mps2_an521_ns             samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_crypto PASSED (qemu 9.910s)
INFO    -  7/12 mps2_an521_ns             samples/tfm_integration/psa_crypto/sample.psa_crypto PASSED (qemu 3.841s)
INFO    -  8/12 mps2_an521_ns             samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_internal_trusted_storage PASSED (qemu 3.259s)
INFO    -  9/12 mps2_an521_ns             samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_storage PASSED (qemu 3.405s)
INFO    - 10/12 mps2_an521_ns             samples/tfm_integration/tfm_ipc/sample.tfm_ipc.no_bl2 PASSED (qemu 2.298s)
INFO    - 11/12 mps2_an521_ns             samples/tfm_integration/tfm_psa_test/sample.tfm.psa_test_protected_storage PASSED (qemu 3.676s)
INFO    - 12/12 mps2_an521_ns             samples/tfm_integration/tfm_ipc/sample.tfm_ipc     PASSED (qemu 2.178s)

INFO    - 12 test scenarios (12 test instances) selected, 3 configurations skipped (3 by static filter, 0 at runtime).
INFO    - 9 of 12 test configurations passed (100.00%), 0 failed, 0 errored, 3 skipped with 0 warnings in 198.06 seconds
INFO    - In total 9 test cases were executed, 3 skipped on 2 out of total 608 platforms (0.33%)
INFO    - 9 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/maciej/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/maciej/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/maciej/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

```

PR #62393 (2023-09-07): ci: boards: Allow mps2_an521_ns to execute tfm tests
Reverts a quickfix which removed a "trusted-firmware-m" tag.
Allow to run tests with the given tag on mps2_an521_ns

Fixes: #62380
commit 97a77438 (2023-09-07): boards: Allow mps2_an521_ns to execute tfm tests

Allow mps2_an521_ns to execute tests with trusted-firmware-m tag.
Resolve an error when this platform is selected as an integration
platform in tests marked with the given tag.

Fixes: #62380
closed 2023-09-07 by fabiobaltieri