zephyrproject-rtos/zephyr #62396: Required Zephyr SDK version (cmake check) is not sufficient

aescolar, 2023-09-07:
**Describe the bug**

Currently the cmake code is checking for zephyr's SDK 0.15, but some tests will not build without 0.16

**To Reproduce**
Steps to reproduce the behavior:
1. get the 0.15.2 sdk (and set your workspace to use it)
2. west twister -p mps2_an521_ns -s samples/tfm_integration/psa_protected_storage/sample.tfm.protected_storage
4. See error below

**Expected behavior**
A cmake warning/error requiring an up to date SDK 

**Impact**
Confused developers when tests fail, as they assume the required SDK version is checked by cmake

**Logs and console output**
```
arm-zephyr-eabi-gcc: fatal error: cannot read spec file 'picolibc.specs': No such file or directory
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04)
 - Toolchain:  zephyr-sdk-0.15.2
 - Commit SHA or Version used: main

nashif, 2023-09-07:
@keith-packard FYI

nordicjm, 2023-09-07:
https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/FindHostTools.cmake#L53 needs changing to 0.16

keith-packard, 2023-09-07:
Do we want to fix the tests or bump the SDK version requirement?

nordicjm, 2023-09-08:
We're coming up for a 0.16.3 toolchain release and want to make picolibc the default C library, I think it's about time we bumped it to 0.16 minimum so that people will already be ready when that change is made

PR #62429 (2023-09-12): cmake: modules: host tools: Bump minimum version to 0.16
Some tests require the zephyr toolchain version 0.16 or newer to be able to use picolibc functionality.

Fixes #62396
closed 2023-09-12 by carlescufi