zephyrproject-rtos/zephyr #62405: samples/subsys/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization fails for qemu_arc_hs5x

fabiobaltieri, 2023-09-07:
**Describe the bug**
`samples/subsys/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization` is failing for `qemu_arc_hs5x` since f0daf904bb0202c9247cdcf4d1186e0762a9db67 (i.e. it does not seem to work with picolibc). Rolling back fixes it.

Seems to work on other arc platform so I'll file a PR to skip the test for this specific one while it gets debugged.

**To Reproduce**
`west build -t run -p -b qemu_arc_hs5x -T samples/subsys/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization` <- no output, building with `CONFIG_NEWLIB_LIBC` works.

**Expected behavior**
Test running

**Impact**
CI brekage

**Environment (please complete the following information):**
Current main (0371708ba4d067319c4a66f33489ebec8aedd0fc), Linux, Zephyr SDK 16.1

keith-packard, 2023-09-07:
Yes, this is likely caused by the qemu bug in delay slot handling. Disabling use of delay slots with the `-fno-delayed-branch` compiler option should also fix this. If you can try that with the older SDK, you'll be able to verify whether this is the problem. See #60071

fabiobaltieri, 2023-09-07:
> Yes, this is likely caused by the qemu bug in delay slot handling. Disabling use of delay slots with the `-fno-delayed-branch` compiler option should also fix this. If you can try that with the older SDK, you'll be able to verify whether this is the problem. See #60071

Brilliant, that works, thanks @keith-packard. So far I've excluded the platform from the test, I guess if it's just an qemu issue there's no point setting the flag for the whole arch. @ruuddw can we leave the exclusion on until the new SDK is released? Alternatively I suppose there would no harm to use:

```diff
diff --git a/boards/arc/qemu_arc/CMakeLists.txt b/boards/arc/qemu_arc/CMakeLists.txt
index 28d0f0d11c..59a6b402c1 100644
--- a/boards/arc/qemu_arc/CMakeLists.txt
+++ b/boards/arc/qemu_arc/CMakeLists.txt
@@ -1,3 +1,5 @@
 # SPDX-License-Identifier: Apache-2.0
 
 zephyr_sources_ifdef(CONFIG_ARC_MPU_ENABLE arc_mpu_regions.c)
+
+zephyr_cc_option(-fno-delayed-branch)
```

closed 2023-11-26 by github-actions[bot]