zephyrproject-rtos/zephyr #62427: modules: loramac-node: sx126x: shadowed declaration related to macro expansion

adolfogc, 2023-09-08:
**Describe the bug**
Expansion of macro `CRITICAL_SECTION_BEGIN` results in a shadowed declaration. Affects boards using the sx126x transceivers. Can be observed when building the `driver/lora/send` and `driver/lora/receive` samples.

**To Reproduce**
Build the mentioned samples for a board that uses the sx126x transceivers. For example:

```
west build -b heltec_wireless_stick_lite_v3 samples/drivers/lora/send --pristine
```

**Expected behavior**
Building the samples, or code using the `loramac-node` module, shouldn't raise `-Wshadow` warnings.

**Impact**
Triggers CI errors.

**Logs and console output**
```bash
[13/195] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (/Users/user/zephyrproject/zephyr), build: zephyr-v3.4.0-3265-g1ddabfa8d8a6
[138/195] Building C object zephyr/drivers/lora/CMakeFiles/loramac-node.d.../user/zephyrproject/modules/lib/loramac-node/src/radio/sx126x/radio.c.obj
In file included from /Users/user/zephyrproject/modules/lib/loramac-node/src/radio/sx126x/radio.c:25:
/Users/user/zephyrproject/modules/lib/loramac-node/src/radio/sx126x/radio.c: In function 'RadioIrqProcess':
/Users/user/zephyrproject/modules/lib/loramac-node/src/boards/utilities.h:183:44: warning: declaration of 'mask' shadows a previous local [-Wshadow]
  183 | #define CRITICAL_SECTION_BEGIN( ) uint32_t mask; BoardCriticalSectionBegin( &mask )
      |                                            ^~~~
/Users/user/zephyrproject/modules/lib/loramac-node/src/radio/sx126x/radio.c:1264:9: note: in expansion of macro 'CRITICAL_SECTION_BEGIN'
 1264 |         CRITICAL_SECTION_BEGIN( );
      |         ^~~~~~~~~~~~~~~~~~~~~~
/Users/user/zephyrproject/modules/lib/loramac-node/src/boards/utilities.h:183:44: note: shadowed declaration is here
  183 | #define CRITICAL_SECTION_BEGIN( ) uint32_t mask; BoardCriticalSectionBegin( &mask )
      |                                            ^~~~
/Users/user/zephyrproject/modules/lib/loramac-node/src/radio/sx126x/radio.c:1252:5: note: in expansion of macro 'CRITICAL_SECTION_BEGIN'
 1252 |     CRITICAL_SECTION_BEGIN( );
      |     ^~~~~~~~~~~~~~~~~~~~~~

```

**Environment (please complete the following information):**
 - OS: macOS
 - Zephyr latest
 - Commit SHA: 1ddabfa8d8a6d807717dcd4ce0d6a5ee723a89fb

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

adolfogc, 2023-10-06:
Hi @fabiobaltieri, thank you! I think this fix did help with an issue I was observing of the lora/receive sample firmware blocking on synchronous reception.

fabiobaltieri, 2023-10-06:
> Hi @fabiobaltieri, thank you! I think this fix did help with an issue I was observing of the lora/receive sample firmware blocking on synchronous reception.

Cool, but I'd be surprised if it does :-) it should only have fixed a compiler error.

adolfogc, 2023-10-06:
Are there any other recent changes that could be related?

fabiobaltieri, 2023-10-06:
Not sure, maybe https://github.com/zephyrproject-rtos/zephyr/pull/63032?

adolfogc, 2023-10-08:
Thank, I'll take a look