zephyrproject-rtos/zephyr #62440: Microchip's TC6 OA library for T1S

lmajewski, 2023-09-08:
## Origin

Microchip's TC6 OA library for T1S:
https://github.com/MicrochipTech/oa-tc6-lib/tree/release-v3.1.3

## Purpose

This is the OPEN Alliance TC6 Protocol Driver for LAN8650/1. The ETH driver (which uses the SPI
for communication) is under development and RFC code shall be provided soon.

It also may serve as a generic library for OA driver for T1S Ethernet protocol (for example the already
in-tree present ADIN2111 and ADIN1111 drivers have the possibility to use OA standardized registers).

## Mode of integration

I would prefer to have it as a module - something similar to `tinycrypt` as it shall be maintained by Microchip.

In Zephyr, we would just use stable releases - like `v3.1.3`.

## Maintainership

GitHub ID: lmajewski (https://github.com/lmajewski)

## Pull Request

https://github.com/MicrochipTech/oa-tc6-lib/tree/release-v3.1.3

## Description

In short - this code would facilitate the integration and support for T1S ETH support
for LAN865x devices.

What is its primary functionality ?

Provide library to communicate with T1S ETH module with using Open Alliance standard.

What problem are you trying to solve? 

I would like to avoid copying the tc6* files to ./samples directory.
It is OK, for development, but will be impractical for long term.

Why is this the right component to solve it.

ADIN2111 has the support for OA register's map (external pin configuration), however, for now
uses ADI's own protocol. With OA library - it shall be possible to re-use the code. Other
ICs, which are going to provide T1S communication can reuse it.

## Dependencies

What other components does this package depend on?

None.

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?

This is implementation of the standard. 

## Revision

v3.1.3 (SHA1: 390181895faa797727f4b1e2dc0be2e7910239f1)

## License

I've just found following License information:
https://github.com/MicrochipTech/oa-tc6-lib/blob/release-v3.1.3/Microchip_SLA001.md

and I don't know the proper SPDX identifier for it. I've posted question to its development [page](https://github.com/MicrochipTech/oa-tc6-lib/issues/13), but no reply.

henrikbrixandersen, 2023-09-11:
Please note the following license restriction from https://docs.zephyrproject.org/latest/contribute/external.html#software-license: "The board will automatically reject licenses that have not been approved by the [Open Source Initiative (OSI)](https://opensource.org/licenses/alphabetical)."

lmajewski, 2023-09-11:
Thanks for the information. The Microchip's code is released under their "own" (even not listed in the OSI page) license. Then, I do conclude, that this source code would not be accepted until released on Apache 2.0?

uwearzt, 2023-09-12:
@lmajewski I am also interested in a driver for the LAN8651 chip. Did you already start development? Would you also be interested in a clean, from scratch development under Apache License?

lmajewski, 2023-09-12:
@uwearzt - The current status is as follows:

 I've started working on the LAN8651 driver with having in my mind the generic OA_TC6 implementation. I can communicate with the chip (OA registers R/W via SPI) and now struggle with having proper reset IRQ deassertion (the device is not pulling up the IRQ line after following the documentation guidelines).

Please also be informed, that recently [RFC](https://lore.kernel.org/lkml/20230908142919.14849-3-Parthiban.Veerasooran@microchip.com/) for Linux driver has been posted (as GPLv2+).

Microchip's TC6 library implementation as is is not acceptable to Zephyr. I've been asked to fill documents and ask for re-licensing to Apache 2.0 - https://github.com/MicrochipTech/oa-tc6-lib/issues/13.

uwearzt, 2023-09-12:
@lmajewski That sounds great, do you have a branch somewhere so i can try to help?

lmajewski, 2023-09-13:
@uwearzt - It is a really early (i.e. hackish) code. I need to polish it, so it will be eligible "to show to the people".  Probably by next week I shall have some code.

Now, I'm struggling with having proper IRQ_N line deassertion after reset of LAN8651. I will keep you updated.

lmajewski, 2023-09-13:
@henrikbrixandersen - And if I may ask - what is the attitude of Zephyr "board" members to code released with `GPLv2+` (as it is the most pervasive license available in Linux)?

lmajewski, 2023-10-06:
> @lmajewski That sounds great, do you have a branch somewhere so i can try to help?

@uwearzt - please find RFC code: https://github.com/zephyrproject-rtos/zephyr/pull/63614

lmajewski, 2023-10-23:
> > And if you don't mind - where shall I place such code? To ./zephyr/modules or to ./modules (and then I shall have a separate repo in mine github?). Personally I would prefer the former option (so the code would be in zephyr's repository from the outset).
> 
> @lmajewski Once you have this code in a suitable license, then you can choose between the two.
> 
> Please read through the whole page here: https://docs.zephyrproject.org/latest/contribute/external.html

@carlescufi - I've already developed from scratch usable, small memory footprint library to handle OA TC6 communication with lan8651 from Microchip: https://github.com/zephyrproject-rtos/zephyr/pull/63614

Now, it is just up to the community to decide when it will be pulled.

carlescufi, 2023-10-23:
> @carlescufi - I've already developed from scratch usable, small memory footprint library to handle OA TC6 communication with lan8651 from Microchip: #63614

Oh great! So then I guess we can close this issue? I see that the review is progressing nicely in the PR so this should be merged soon.

lmajewski, 2023-10-23:
@carlescufi - Yes, it can be closed.

closed 2023-10-23 by lmajewski