zephyrproject-rtos/zephyr #62441: [RFC][RISCV?] Support for multi-instance & multi-level interrupt controller driver

ycsin, 2023-09-08:
## Introduction

Hi all, I'm looking at our custom [`intc_plic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/interrupt_controller/intc_plic.c) implementation and really feel like we should do something upstream to minimize the gap between upstream & our codebase, our [`intc_plic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/interrupt_controller/intc_plic.c) has partial implementation to support multi-instance, alongside with some hacks to the RISCV's [`irq_manage.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/irq_manage.c) to complete that change.

### Problem description

As far as I can tell, the upstream build system and/or the RISCV's [`arch_irq_connect_dynamic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/irq_manage.c#L35) install approach do not support such multi-instance & multi-level interrupt controller driver. For [`intc_plic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/interrupt_controller/intc_plic.c) to support multiple instances, we need to be able to discern the IRQs between the instances, and install them accordingly into the [`_sw_isr_table`](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sw_isr_table.h#L44).

### Proposed change

Some ideas, starting from the easiest one..

1. Allow soc/driver implementation of the (RISCV's) [`arch_irq_connect_dynamic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/irq_manage.c#L35), to facilitate custom implementation downstream
2. Update the RISCV's [`arch_irq_connect_dynamic`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/irq_manage.c#L35) upstream, such that it used a arch-specific `isr_install` to install the ISR based on which controller the IRQ was intended for.
3. Update the `arch_irq_connect_dynamic` & [`z_isr_install`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/common/sw_isr_common.c#L74) upstream, such that `arch_irq_connect_dynamic` pass in additional info into `z_isr_install` for it to install the ISR into the table properly. Probably requires additional properties in the devicetree bindings to partition the `_sw_isr_table`
4. Update [scripts/build/gen_isr_tables.py](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/build/gen_isr_tables.py) to generate one `_sw_isr_table` for every interrupt controller, so that we don't need to partition the table, is that possible?

ycsin, 2023-09-08:
cc @carlocaione @cfriedt @jgl-meta

carlocaione, 2023-09-08:
Thanks for opening this. A couple of questions:
1. When you say multi-instance & multi-level you mean different interrupt controllers, possibly in cascade managing different IRQ ranges?
2. Why are you interested only in the dynamic IRQs? Isn't this problem also valid for statically installed handlers with `IRQ_CONNECT`?

cfriedt, 2023-09-08:
The PLIC definitely needs to be made multi-instance (if only to support `build_all` tests with different configurations).

So I agree to that.

I think the scenario that @ycsin is talking about does have more than one "root" interrupt controller while also supporting multi-level interrupts.

To address that, it could be useful to make a variant of `arch_irq_connect_dynamic()`, say `arch_irq_connect_dynamic_dev(..., const struct device *dev)`, such that the former could wrap around the latter. If the `dev` parameter were non-`NULL` it could override whatever interrupt controller is normally used.

However, I'm also interested in the proposal by @dcpleung and @andyross about chained interrupts / a DAG, which is IIRC how it's done in Linux. We may be able to get that done for LTSv3? Hard to say..

cfriedt, 2023-09-08:
Also, cc @fkokosinski @npitre

PR #62907 (2023-11-09): drivers: intc: plic: add support for multiple instances
1. Refactored the build_all test added in #62541
2. Refactored the `intc_plic` driver to be multi-instance ready
3. Updated the `sw_isr_*` so that it also records the device pointer of the interrupt controllers & provide functions to do a lookup for the controller using the IRQN, as well as getting the ISR table offset for a controller
4. Updated the `intc_plic` driver to use the functions in the `sw_isr` so that it can differentiate parent controller from IRQN, and invoke the correct ISR with the correct offset from `sw_isr_*`
   - Modif [... cut]
closed 2023-11-09 by carlescufi