zephyrproject-rtos/zephyr #62444: 64 bit number formatting broken on Picolib

SeppoTakalo, 2023-09-08:
**Describe the bug**

When trying to print or format a string that contains 64bit signed numbers, Picolib does not produce correct output, but instead truncates values into 32bit before conversion.

**To Reproduce**

Use following code:
```c
int main(void)
{
	printf("Hello World! %s\n", CONFIG_BOARD);

	printf("%" PRId64 "\n", INT64_MAX);
	printf("%" PRIx64 "\n", INT64_MAX);

	printf("%" PRId32 "\n", INT32_MAX);
	printf("%" PRIx32 "\n", INT32_MAX);

	return 0;
}
```

and build it with PICOLIB enabled and in any of the 32bit platforms. I tried `qemu_cortex_m3`.

Output is
```
Hello World! qemu_cortex_m3
-1
ffffffff
2147483647
7fffffff
QEMU: Terminated
```

**Expected behavior**
Expected output would be
```
Hello World! qemu_cortex_m3
9223372036854775807
7fffffffffffffff
2147483647
7fffffff
```
When I compile the same code using NewlibC, it produces the correct output.

**Impact**
Breaks formatting of big numbers. For example LwM2M plain text encoding.

Root causes of issue  https://github.com/zephyrproject-rtos/zephyr/issues/62330

nordicjm, 2023-09-08:
This is known, it is due to using the bundled picolibc module, if you build from source then it is not a problem. @keith-packard can probably link the issue I saw it in earlier

keith-packard, 2023-09-08:
You can also enable floating point output by selecting `CONFIG_PICOLIBC_IO_FLOAT` -- that includes 64-bit integer output.

We might want to change the SDK picolibc configuration, although adding long long support increases the size of the library by a small amount.

SeppoTakalo, 2023-09-11:
@keith-packard Thank you for the information.
I tried, and yes it definitely looks like `CONFIG_PICOLIBC_IO_FLOAT` would fix our number formatting tests.

Now, I assume that this problem is known and will not be fixed. But is there an open bug report? How is this documented?
I assume that this configuration option will impact on flash size, so developers might consider leaving if off knowing that some clear text formatting corner cases are not working.

keith-packard, 2023-09-11:
> @keith-packard Thank you for the information. I tried, and yes it definitely looks like `CONFIG_PICOLIBC_IO_FLOAT` would fix our number formatting tests.
>
> Now, I assume that this problem is known [...]

It's all about flash space trade-offs for the default configuration. Are long long formatting requests common enough that the default (SDK provided) library should enable them? Or is the space savings important enough to leave them out? Including long long support in printf costs 840 bytes on cortex m3 targets, most of which is pulling in the 64-bit division code from libgcc (the change in printf is only 32 bytes).

If you need for long long printf support for your project, you can switch from using the SDK bundled picolibc version to the picolibc module. That provides finer grained selection over printf features, including the ability to enable long long support without also enabling floating point support.

Alternatively, if you feel that the SDK should include long long printf support by default, that's also possible, but would need some discussion to build rough consensus for the change.

keith-packard, 2023-09-11:
I did a bit more poking at this to try and provide a bit more data here. As I mentioned above, the reason for eliding long long support is mostly about not using the 64-bit software divide code; the printf code itself grows by a tiny amount (32 bytes).

The alternative (and one which picolibc used in the past) was to avoid the 64-bit divide by doing division and modulus by 10 with fancy code. I just re-implemented that with a modest chunk of code which does the division by 10 using reciprocal multiply; the upper 64-bits of a multiply by 0xcccccccccccccccdULL yields an approximation to /10, which can be adjusted and then the remainder computed. That increases the size of vfprintf by 116 bytes, but avoids pulling in __aeabi_uldivmod for a net savings of 632 bytes.

That would be good for targets where the only 64-bit division operations were in the bowels of printf. However, a target which does a bunch of 64-bit math and uses __aeabi_uldivmod for its own needs now pays the 116 byte additional cost in vfprintf, without the savings from not linking __aeabi_uldivmod. 

All of this also presumes we can tell which platforms perform division in software, which is actually pretty hard to detect at build time.

mniestroj, 2023-11-19:
I have encountered this problem with my downstream application. All `long long` prints are truncated to 32-bit even though `CONFIG_PICOLIBC_IO_LONG_LONG` is selected. If SDK bundled picolibc does not have `long long` support by default, then we should disable `CONFIG_PICOLIBC_IO_LONG_LONG`, so it does not confuse users that try to get `long long` prints working.

Of course building picolibc from module works fine, as well as enabling `CONFIG_PICOLIBC_IO_FLOAT` as suggested above. But still, this is not enough.

Possible approaches:
1. Enable `long long` support by default in Zephyr SDK, since `CONFIG_PICOLIBC_IO_LONG_LONG=y` is selected by default as well (due to `CONFIG_CBPRINTF_FULL_INTEGRAL=y`).
2. Select `CONFIG_PICOLIBC_IO_FLOAT` automatically at Kconfig level whenever `CONFIG_PICOLIBC_USE_MODULE=n` and `CONFIG_PICOLIBC_IO_LONG_LONG=y` 
3. Change `if(CONFIG_PICOLIBC_IO_FLOAT)` to `if(CONFIG_PICOLIBC_IO_FLOAT OR (NOT CONFIG_PICOLIBC_USE_MODULE AND CONFIG_PICOLIBC_IO_LONG_LONG))` in `lib/libc/picolibc/CMakeLists.txt`
4. Select `PICOLIBC_IO_FLOAT` instead of `PICOLIBC_IO_LONG_LONG` when `!PICOLIBC_USE_MODULE`.

2-4 are not optimal, since they will pull float/double printf/scanf by default with simplest Zephyr applications (like hello_world). But at least this results in expected (and selected by configuration) behavior, right? Or maybe we don't need to (try to) enable `long long` support with `CBPRINTF_FULL_INTEGRAL` and in most cases float variant won't be used?

mniestroj, 2023-11-19:
Nevermind above comment, I just read linked "picolibc 1.8.5" which solves this issue.

PR #62882 (2023-11-20): Picolibc 1.8.5
Merge in picolibc version 1.8.5.

There is a matching SDK PR pending, https://github.com/zephyrproject-rtos/sdk-ng/pull/707

This offers two new printf variants, 'long long', which provides an integer-only mode with long long support included and a 'minimal' variant which cuts out a bunch of format modifier support to make things very small.

hello-world on cortex-m3 FLASH sizes:
```
float		13378
float inexact	10666
long long	 7950
integer		 7798
minimal		 6834
```
For comparison, here are the minimal C lib sizes:
```
float		11082
long long	 9194
integer		 8426
nano		 7386
```
W [... cut]
closed 2023-11-20 by cfriedt