zephyrproject-rtos/zephyr #62460: too few arguments in invocation of macro "Z_FOR_LOOP_1"

dpkristensen, 2023-09-08:
**Describe the bug**
When using LOG_* macros with a single argument, linters and compilers at high warning levels can emit show warnings like this:
~~~~
too few arguments in invocation of macro "Z_FOR_LOOP_1"
~~~~

This is mentioned in comments on #21119; the ticket opened in https://github.com/microsoft/vscode-cpptools/issues/9187 was closed due to inactivity.

**To Reproduce**
Steps to reproduce the behavior via VS Code:

1. Ensure `ms-vscode.cpptools-extension-pack` is installed.
2. Open vscode to sample app `zephyr/samples/subsys/shell/shell_module`
3. Run the build to generate the configuration: `west build -b nrf52dk_nrf52832`
4. Populate `.vscode/c_cpp_configuration.json` file:
~~~~
{
    "configurations": [
        {
        "name": "Test",
        "compileCommands": "build/compile_commands.json",
        }
    ],
    "version": 4
}
~~~~
5. Open src/main.c and look at the `problems` output

**Expected behavior**
No problems reported

**Impact**
Many problems reported makes it more difficult to find actual problems with the code.

**Logs and console output**
Problems output:
~~~~
too few arguments in invocation of macro "Z_FOR_LOOP_1"
~~~~

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK, gcc 11.4.0
 - Zephyr commit 8ea58fc43b77859df713712cc5be2c33a10b0e7e

**Additional context**
I have seen this specifically with nRF Connect SDK v2.2.0 (v3.2.99-ncs1), but the problem is generic to Zephyr code itself.  I was able to reproduce it with Zephyr only using the steps above on a computer in a standalone zephyr installation.

I've narrowed it down to this section of Z_LOG_MSG_CREATE3():
~~~~
	bool has_rw_str = CBPRINTF_MUST_RUNTIME_PACKAGE( \
					Z_LOG_MSG_CBPRINTF_FLAGS(_cstr_cnt), \
					__VA_ARGS__); \
~~~~

It seems there is some old logic in CBPRINTF_MUST_RUNTIME_PACKAGE where it used to take separate fmt and ... args, but now just takes ... instead.  In other parts of this macro, `Z_LOG_FMT_ARGS(_fmt, ##__VA_ARGS__)` is used instead to reduce the number of arguments to just format when there are no subsequent arguments.

There's probably some other history on this, and I haven't had time to dig further.

mrclark25, 2023-10-09:
This impacts my project as well.

ryland-mueller, 2024-01-09:
This has been an issue for years, can we please keep this issue open (remove stale label)

dpkristensen, 2024-01-18:
I don't believe I have access to remove the label or re-open the issue; one of the maintainers will have to do this.

nordic-krch, 2024-01-19:
Can you check if that was the fix? #46532

dpkristensen, 2024-01-19:
I applied the change from #46532 and it silenced the warning from the checker.  Also both single and multi-argument versions of LOG_INF() macro produced the expected output.  So I think it will fix it.

nordic-krch, 2024-01-23:
@dpkristensen @ryland-mueller I've reopened the PR with the fix.

PR #67980 (2024-01-29): sys: util: Accept empty FOR_EACH
The `##` part of `##__VA_ARGS__` in the `Z_FOR_EACH_ENGINE` macro technically
breaks the invocation of empty `FOR_EACH` sequences, as the empty
__VA_ARGS__ gets squashed with in the invocation of `Z_FOR_LOOP_1()` in
`Z_FOR_LOOP_2()`, which makes the macro only pass 4 arguments to
`Z_FOR_LOOP_1`.

This breaks IntelliSense in Microsoft's C/C++ extension for VS Code,
which is strict about the amount of arguments you can pass to a variadic
macro.

Fixes #62460.
closed 2024-01-29 by carlescufi