zephyrproject-rtos/zephyr #62464: ZRESTRICT conflict between toolchain/common.h and toolchain/gcc.h

marc-hb, 2023-09-08:
The addition of this `#include toolchain/common.h` just revealed a funny ZRESTRICT conflict between `include/zephyr/toolchain/common.h` and `include/zephyr/toolchain/gcc.h` when compiling C++

- #27245

As a datapoint/evidence, replacing this line with `#include <zephyr/toolchain/gcc.h>` avoids the warning.

https://github.com/thesofproject/sof/actions/runs/6127039819/job/16632163797?pr=8175

```
/zep_workspace/sof/src/audio/module_adapter/iadk/module_initial_settings_concrete.cpp
In file included from /zep_workspace/zephyr/include/zephyr/toolchain.h:50,
                 from /zep_workspace/zephyr/include/zephyr/sys/time_units.h:10,
                 from /zep_workspace/zephyr/include/zephyr/sys/util.h:640,
                 from /zep_workspace/sof/src/include/sof/audio/module_adapter/iadk/adsp_stddef.h:15,
                 from /zep_workspace/sof/src/audio/module_adapter/iadk/module_initial_settings_concrete.cpp:7:
/zep_workspace/zephyr/include/zephyr/toolchain/gcc.h:87: error: "ZRESTRICT" redefined [-Werror]
   87 | #define ZRESTRICT __restrict
      | 
In file included from /zep_workspace/zephyr/include/zephyr/sys/util.h:18:
/zep_workspace/zephyr/include/zephyr/toolchain/common.h:33: note: this is the location of the previous definition
   33 | #define ZRESTRICT
      | 
cc1plus: all warnings being treated as errors
```

_Originally posted by @marc-hb in https://github.com/zephyrproject-rtos/zephyr/pull/61962#discussion_r1320416524_

dcpleung, 2023-09-09:
This is introduced by commit fbf851cdc4d8f0afe60131aea6154945f126d778 where `sys/util.h` includes `sys/time_units.h`.
# `sys/unit.h` includes `toolchain/common.h` in the beginning which sets `ZRESTRICT` to empty.
# `sys/unit.h` includes `sys/time_units.h` late in the file.
# `sys/time_units.h` includes `toolchain.h` which includes `toolchain/gcc.h` and it also defines `ZRESTRICT` to actual usable value.

Normally, `toolchain/common.h` is included by `toolchain/<toolchain>.h` so each toolchain can have their own macros. `toolchain/common.h` then defines the macros used by the code but not defined in toolchain specific include file. The inclusion of `sys/time_units.h` in `sys/unit.h` reverses this order... which causes the issue.

fabiobaltieri, 2023-09-10:
@dcpleung is the fix to move the `zephyr/sys/time_units.h` up in `util.h`?

dcpleung, 2023-09-11:
> @dcpleung is the fix to move the `zephyr/sys/time_units.h` up in `util.h`?

Maybe? I haven't tried it.

IMHO, hierarchy wise, `util.h` is more fundamental that it should not include `time_units.h` as it is forcing its content on source files which have nothing to do with time units.

marc-hb, 2023-09-11:
> This is introduced by commit fbf851cdc4d8f where sys/util.h includes sys/time_units.h.

Having two .h files that each include each other directly does not make any sense to me; I think this commit should never have been merged.

```
git checkout fbf851cdc4d8f
git grep -n ...

include/zephyr/sys/time_units.h:11:#include <zephyr/sys/util.h>
include/zephyr/sys/util.h:565:#include <zephyr/sys/time_units.h>

```
Two .h files including each other means the code in these two .h files can _intentionally_ end-up in one order in some .c files but in another order in other .c file. I imagine it's technically possible for both orders to compile _depending on which macros are in use_ but does anyone seriously want that? Structuring .h files and debugging `#include` is already incredibly hard (see #41543), I think we really don't want to make it even harder by opening this sort of "varying `#include` order" possibility. This can make the .h combinatorial explosion blow up even more.

I don't fully understand what the #56006 problem was and I bet it wasn't an easy one but I'm pretty sure the one-line commit fbf851cdc4d8f was not the fix. I guess the proper fix is as usual to split/combine/shuffle the content of .h files into a proper DAG.

cc: 
- #41543

fabiobaltieri, 2023-09-12:
> Two .h files including each other means the code in these two .h files can _intentionally_ end-up in one order in some .c files but in another order in other .c file. I imagine it's technically [...]

Oh I see what you mean now, sorry took me a bit to realize that the problem was the headers including each other

> I don't fully understand what the #56006 problem was and I bet it wasn't an easy one but I'm pretty sure the one-line commit [fbf851c](https://github.com/zephyrproject-rtos/zephyr/commit/fbf851cdc4d8f0afe60131aea6154945f126d778) was not the fix. I guess the proper fix is as usual to split/combine/shuffle the content of .h files into a proper DAG.

Yeah I agree the cross-include makes no sense and is asking for troubles, obviously no one caught that in code review.

@yperess is that something you could look into? Moving the definition between headers and removing the cross include seems to be the right thing.

yperess, 2023-09-12:
> Oh I see what you mean now, sorry took me a bit to realize that the problem was the headers including each other
>
>
> Yeah I agree the cross-include makes no sense and is asking for troubles, obviously [...]

I make no guarantee on timeline but I'll be happy to add it to my list

marc-hb, 2023-09-12:
> I make no guarantee on timeline but I'll be happy to add it to my list

Thanks! May I ask what order of magnitude is your current list? I mean days-long or months-long? According to @dcpleung this is what has just turned (in an unfortunate sequence of events) our CI red.

fabiobaltieri, 2023-09-13:
Ok I'll take a stab at this myself since it's messing with CI.

marc-hb, 2023-09-14:
Thanks a lot @fabiobaltieri . I tested your #62580 and while it looks like it does fix the circular #include issue, I found it makes no difference for the ZRESTRICT redefinition bug I reported here.

So I spent a bit more more time and I found a much simpler, one-line fix for the ZRESTRICT redefinition bug:
- #62620 

Your fix and mine do not conflict with each other, I believe they're unrelated after all. Sorry for the confusion.

fabiobaltieri, 2023-09-14:
Ok thanks for checking that @marc-hb, I guess let's go ahead with both, fixing the cross include can't be a bad thing anyway.

dcpleung, 2023-09-14:
#62620 is merged. So this should have been fixed.

marc-hb, 2023-09-14:
You mean like this? https://github.com/orgs/community/discussions/17308

I'll close this when daily compilation works again: https://github.com/thesofproject/sof/actions/workflows/daily-tests.yml 

Should happen tomorrow.

marc-hb, 2023-09-15:
Daily build is passing again: https://github.com/thesofproject/sof/actions/runs/6192376904

closed 2023-09-15 by marc-hb