zephyrproject-rtos/zephyr #62465: stm32h5: samples/net/gptp doesn't work at all

blackhelicoptersdotnet, 2023-09-09:
**Describe the bug**

gptp doesn't appear to work at all on stm32h5. Peer delay always returns nonsensical value (4294967295ns).

I get exactly the same output whether testing with two identically configured `nucleo_h563zi` boards connected directly to each other, or with one board connected directly to a Linux machine runnning ptp4l v4.1 with the following configuration:

https://github.com/richardcochran/linuxptp/blob/master/configs/gPTP.cfg

**To Reproduce**
Steps to reproduce the behavior:

On the board:
1. cd samples/net/gptp
2. west build -b nucleo_h563zi
3. west flash

On the Linux system:
1. apk add linuxptp
2. wget https://github.com/richardcochran/linuxptp/blob/master/configs/gPTP.cfg
3. ptp4l -i eth1 -f gPTP.cfg -l 7 -m

**Expected behavior**
1. Linux box assumes GM role, board syncs to the GM. 
2. One Nucleo board assumes the GM role, and the other board syncs to the GM. 

**Impact**
Showstopper.

**Logs and console output**
UART output:

```
*** Booting Zephyr OS build zephyr-v3.4.0-3440-g66bf33f09008 ***
[00:00:00.002,000] <inf> net_config: Initializing network
[00:00:00.002,000] <inf> net_config: Waiting interface 1 (0x20001390) to be up...
[00:00:00.004,000] <wrn> net_if: iface 0x20001390 is down
[00:00:01.006,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:01.008,000] <wrn> net_if: iface 0x20001390 is down
[00:00:01.673,000] <inf> net_config: Interface 1 (0x20001390) coming up
[00:00:01.673,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:01.773,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:01.773,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:02.010,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:03.015,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:04.016,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:05.018,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:06.019,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:07.020,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:08.022,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:09.023,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:10.025,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:11.026,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
```

**Environment (please complete the following information):**
PTP server OS: Alpine Linux 3.18.3
Zephyr build platform: macOS 13.5.2, SDK v0.16.1, commit 66bf33f0900854a3df968e8a93a8dba4bd676c3f

**Additional context**

blackhelicoptersdotnet, 2023-09-12:
Looking at a packet capture of traffic between the board and the Linux box, it looks like the PTP peripheral clock isn't being initialised properly; it's running extremely fast.

In packet captures I can see the board putting `Peer_Delay_Resp` and `Peer_Delay_Resp_Follow_Up` onto the wire about 47µs apart, but with a delta of around 113ms in the PTP timestamps. 

[gptp.pcap.zip](https://github.com/zephyrproject-rtos/zephyr/files/12586127/gptp.pcap.zip)

Desvauxm-st, 2023-09-13:
@blackhelicoptersdotnet 
can you check if it works on a Nucleo_H7xx and on Nucleo_F7xx please
That can help me to find the issue.

Desvauxm-st, 2023-09-14:
@blackhelicoptersdotnet
I am not able to test gPtp on my laptop ( sorry)
can you please  try on your side the following modifications ( #62632 )

Ethernet: gptp doesn't work at all on STM32H5 #62632
can you test it  with only the fist commit alone please

blackhelicoptersdotnet, 2023-09-14:
@Desvauxm-st Justed tested with first commit `a23aab` only, same result.

`[00:12:03.531,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns`

I've gotten my hands on a `nucleo_f767zi` board, I can do some testing with that tomorrow also.

Desvauxm-st, 2023-09-14:
@blackhelicoptersdotnet
Thanks you very much for the test
The most important for us is to compare with the nucleo_H7xx ( normaly very similar as nucleo_H5xx ).

blackhelicoptersdotnet, 2023-09-14:
Looking much better on h5!

```
uart:~$ net gptp
Port Interface          Role
 1   0x20001390 [1]     SLAVE

SiteSyncSync state machine variables:
        Current state                  : RECEIVING_SYNC
        A PortSyncSync struct is ready : no
ClockSlaveSync state machine variables:
        Current state                  : SEND_SYNC_IND
        A PortSyncSync struct is ready : no
        The local clock has expired    : no
PortRoleSelection state machine variables:
        Current state                  : ROLE_SELECTION
ClockMasterSyncReceive state machine variables:
        Current state                  : SOURCE_TIME
        A ClockSourceTime              : yes
        The local clock has expired    : no
uart:~$ net gptp port 1
Port id    : 1 (SLAVE)
Interface  : 0x20001390 [1]
Clock id   : 00:80:e1:ff:fe:b7:12:f2
Version    : 2
AS capable : yes

Configuration:
Time synchronization and Best Master Selection enabled        : yes
The port is measuring the path delay                          : yes
One way propagation time on the link attached to this port    : 2149 ns
Propagation time threshold for the link attached to this port : 100000 ns
Estimate of the ratio of the frequency with the peer          : 1
Asymmetry on the link relative to the grand master time base  : 0
Maximum interval between sync messages                        : 375000000
Maximum number of Path Delay Requests without a response      : 3
Current Sync sequence id for this port                        : 64112
Current Path Delay Request sequence id for this port          : 38200
Current Announce sequence id for this port                    : 22341
Current Signaling sequence id for this port                   : 8303
Whether neighborRateRatio needs to be computed for this port  : yes
Whether neighborPropDelay needs to be computed for this port  : yes
Initial Announce Interval as a Logarithm to base 2            : 0
Current Announce Interval as a Logarithm to base 2            : 0
Initial Sync Interval as a Logarithm to base 2                : -4
Current Sync Interval as a Logarithm to base 2                : -4
Initial Path Delay Request Interval as a Logarithm to base 2  : 0
Current Path Delay Request Interval as a Logarithm to base 2  : 0
Time without receiving announce messages before running BMCA  : 3000 ms (3)
Time without receiving sync messages before running BMCA      : 0 ms (3)
Sync event transmission interval for the port                 : 62 ms
Path Delay Request transmission interval for the port         : 1000 ms
BMCA default priority1                                        : 248
BMCA default priority2                                        : 248

Runtime status:
Current global port state                                : SLAVE
Path Delay Request state machine variables:
        Current state                                    : WAIT_ITV_TIMER
        Initial Path Delay Response Peer Timestamp       : 1694733374249939576
        Initial Path Delay Response Ingress Timestamp    : 1694733374249941463
        Path Delay Response messages received            : 1
        Path Delay Follow Up messages received           : 1
        Number of lost Path Delay Responses              : 0
        Timer expired send a new Path Delay Request      : 0
        NeighborRateRatio has been computed successfully : 1
        Path Delay has already been computed after init  : 0
        Count consecutive reqs with multiple responses   : 0
Path Delay Response state machine variables:
        Current state                                    : INITIAL_WAIT_REQ
[... 33 lines cut ...]
        Expired announce information                     : no
PortAnnounceTransmit state machine variables:
        Current state                                    : POST_IDLE
        Trigger announce information                     : no

Statistics:
Sync messages received                 : 763
Follow Up messages received            : 862
Path Delay Request messages received   : 98
Path Delay Response messages received  : 99
Path Delay messages threshold exceeded : 0
Path Delay Follow Up messages received : 99
Announce messages received             : 96
ptp messages discarded                 : 0
Sync reception timeout                 : 16
Announce reception timeout             : 0
Path Delay Requests without a response : 3
Sync messages sent                     : 17
Follow Up messages sent                : 17
Path Delay Request messages sent       : 102
Path Delay Response messages sent      : 98
Path Delay Response FUP messages sent  : 98
Announce messages sent                 : 2
uart:~$
```

Desvauxm-st, 2023-09-15:
@blackhelicoptersdotnet
good
I will push another PR in HAL to fix this issue
Thanks you for your help

Desvauxm-st, 2023-09-15:
@blackhelicoptersdotnet
To push my new PR in our HAL branch i want associate your name so can you give me your git  signature
for exemple for me it is that  : Signed-off-by: Marc Desvaux <marc.desvaux-ext@st.com>

thanks you

blackhelicoptersdotnet, 2023-09-15:
@Desvauxm-st certainly! Thanks for all your work on this. 

`Signed-off-by: Mathew Meins <mm@blackhelicopters.net>`

blackhelicoptersdotnet, 2023-09-15:
@Desvauxm-st Definitely fixed for h5 but `nucelo_f767zi` is exhibiting the same same symptoms. 

```
[00:00:49.144,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:50.145,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:51.147,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
[00:00:52.148,000] <wrn> net_gptp: Not AS capable: 4294967295 ns > 100000 ns
uart:~$ net gptp port 1
Port id    : 1 (DISABLED)
Interface  : 0x20021420 [1]
Clock id   : 00:80:e1:ff:fe:e8:3d:b6
Version    : 2
AS capable : no

Configuration:
Time synchronization and Best Master Selection enabled        : yes
The port is measuring the path delay                          : yes
One way propagation time on the link attached to this port    : 4294967295 ns
Propagation time threshold for the link attached to this port : 100000 ns
Estimate of the ratio of the frequency with the peer          : 0
Asymmetry on the link relative to the grand master time base  : 0
Maximum interval between sync messages                        : 10000000
Maximum number of Path Delay Requests without a response      : 3
Current Sync sequence id for this port                        : 9791
Current Path Delay Request sequence id for this port          : 30448
Current Announce sequence id for this port                    : 19677
Current Signaling sequence id for this port                   : 28908
Whether neighborRateRatio needs to be computed for this port  : yes
Whether neighborPropDelay needs to be computed for this port  : yes
Initial Announce Interval as a Logarithm to base 2            : 0
Current Announce Interval as a Logarithm to base 2            : 0
Initial Sync Interval as a Logarithm to base 2                : -4
Current Sync Interval as a Logarithm to base 2                : -4
Initial Path Delay Request Interval as a Logarithm to base 2  : 0
Current Path Delay Request Interval as a Logarithm to base 2  : 0
Time without receiving announce messages before running BMCA  : 1 ms (3)
Time without receiving sync messages before running BMCA      : 0 ms (3)
Sync event transmission interval for the port                 : 62 ms
Path Delay Request transmission interval for the port         : 1000 ms
BMCA default priority1                                        : 255
BMCA default priority2                                        : 248

Runtime status:
Current global port state                                : DISABLED
Path Delay Request state machine variables:
        Current state                                    : WAIT_ITV_TIMER
        Initial Path Delay Response Peer Timestamp       : 1694770907855572216
        Initial Path Delay Response Ingress Timestamp    : 148748520000000053
        Path Delay Response messages received            : 1
        Path Delay Follow Up messages received           : 1
        Number of lost Path Delay Responses              : 0
        Timer expired send a new Path Delay Request      : 0
        NeighborRateRatio has been computed successfully : 1
        Path Delay has already been computed after init  : 0
        Count consecutive reqs with multiple responses   : 0
Path Delay Response state machine variables:
        Current state                                    : INITIAL_WAIT_REQ
SyncReceive state machine variables:
        Current state                                    : DISCARD
        A Sync Message has been received                 : no
        A Follow Up Message has been received            : no
        A Follow Up Message timeout                      : no
        Time at which a Sync Message without Follow Up
                                     will be discarded   : 0
SyncSend state machine variables:
        Current state                                    : INITIALIZING
        A MDSyncSend structure has been received         : no
        The timestamp for the sync msg has been received : no
PortSyncSyncReceive state machine variables:
        Current state                                    : DISCARD
[... 19 lines cut ...]
        Current state                                    : POST_DISABLED
        Expired announce information                     : yes
PortAnnounceTransmit state machine variables:
        Current state                                    : POST_IDLE
        Trigger announce information                     : no

Statistics:
Sync messages received                 : 243
Follow Up messages received            : 34
Path Delay Request messages received   : 34
Path Delay Response messages received  : 34
Path Delay messages threshold exceeded : 33
Path Delay Follow Up messages received : 34
Announce messages received             : 31
ptp messages discarded                 : 0
Sync reception timeout                 : 0
Announce reception timeout             : 0
Path Delay Requests without a response : 20
Sync messages sent                     : 0
Follow Up messages sent                : 0
Path Delay Request messages sent       : 54
Path Delay Response messages sent      : 34
Path Delay Response FUP messages sent  : 34
Announce messages sent                 : 0
```

erwango, 2023-10-02:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/62800

closed 2023-10-02 by erwango