zephyrproject-rtos/zephyr #62473: samples: net: wifi: esp32c3 does not connect with picolibc

martinjaeger, 2023-09-10:
**Describe the bug**

The sample does not work with `picolibc`, which is now the default C library.

**To Reproduce**

Steps to reproduce the behavior:
1. `west build -b esp32c3_devkitm samples/net/wifi -p`
2. In UART shell: `wifi connect <ssid> 0 <password>`

It will print `Connection requested` and nothing happens. After waiting a few minutes it prints `Disconnected`. `wifi scan` works fine.

It works with newlib using the following additions to `prj.conf`:

```
CONFIG_NEWLIB_LIBC=y
```

However, it does **not** work with newlib-nano either:

```
CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=y
```

**Expected behavior**

WiFi should work with any libc.

**Impact**

Showstopper if you depend on `picolibc` to reduce memory footprint, annoyance otherwise.

**Environment (please complete the following information):**

 - OS: Manjaro Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA: bc695c6df5eb8a1f8f9720b20d7e5dcd1533242b

martinjaeger, 2023-09-11:
> Does this target require a binary blob for the WiFi which internally references C library functions?

Yes, it does. And that's exactly where I stopped investigating if I can find out the root cause myself. Looks like only someone from Espressif can fix this.

keith-packard, 2023-09-11:
> Yes, it does. And that's exactly where I stopped investigating if I can find out the root cause myself. Looks like only someone from Espressif can fix this.

Yup. Would be lovely if Espressif offered a version of their binary built against the Zephyr SDK using picolibc. Who can reach out and try to make this happen?

sylvioalves, 2023-09-11:
Hi, sure. I confirmed ESP32, ESP32S2 and and ESP32S3 (all xtensa's) are properly working.  
ESp32C3 (riscv) is not. I'll check.

PR #63046 (2023-09-27): soc: esp32s3: add initial AMP support
This PRs brings initial ESP32S3 AMP support.  

- Add SoC configuration to build image for APPCPU (core 1).  
- Add board option to enable the core 1 build.  
- Add rpmsg sample code.  

Next steps: 
- Use DTS to split memory between cores (IRAM/DRAM/FLASH).
- Perform region check to avoid overlapping.
- Evaluate #51833.

This PR indirectly address the issue below as part of the hal_espressif update:  
Fixes #62473