zephyrproject-rtos/zephyr #62514: feat: support for CRT (Component RetTrimming) calibration on BMI270 incl. bounty

maxmclau, 2023-09-11:
**Is your enhancement proposal related to a problem? Please describe.**
We manufacture a few boards with BMI270 IMUs and always end up with a few boards with major gyroscope trim problems. The BMI270 supports CRT retrimming to "zero out" the gyroscope when it's at rest - saving this calibration offset to flash.

**Describe the solution you'd like**
We'd like to see an extension of the BMI270 driver with support for `SENSOR_ATTR_CALIBRATION` (or similar attribute) that completes a CRT calibration procedure. Bosch provides documentation on how this is done in their datasheet (4.14.2 Gyroscope) as well as a sample implementation in their [BMI270-Sensor-API](https://github.com/boschsensortec/BMI270-Sensor-API/blob/d270cdee303b2ed9ea4c43fb7064da8ded8a48b4/bmi2.c#L10325C1-L10325C41). 

We're offering a **$400** bounty for anyone who merges a PR with this functionality. We've used bountysource in the past, but it looks like they're in [hot water](https://github.com/bountysource/core/issues/1586) so we're open to use anyone's preferred platform. A one-time github sponsorship is another route if that's easier.

nordicjm, 2023-09-12:
> We're offering a $400 bounty for anyone who merges a PR with this functionality

@nashif @kartben this doesn't seem acceptable to be posted here?

henrikbrixandersen, 2023-09-12:
The problem with a bounty posted here like this is that even if someone steps up and attempts to solve this issue, there is no guarantee that the resulting patch will be accepted into upstream Zephyr.

abebeos, 2023-09-12:
> there is no guarantee that the resulting patch will be accepted into upstream

Nothing special, this is standard with bounties.

maxmclau, 2023-09-12:
> The problem with a bounty posted here like this is that even if someone steps up and attempts to solve this issue, there is no guarantee that the resulting patch will be accepted into upstream Zephyr.

@abebeos is right - this is standard for bounties. That being said, we would be willing to offer half of the bounty up for a initial functional PR and the other half after merge if that helps. We've been successful with that sponsorship arrangement in the past for other Zephyr contributors.

Up to you guys though. I'll close the issue if Zephyr would rather not encourage bounties.

carlescufi, 2023-09-13:
Since there seems to be some controversy regarding the bounty I've added the TSC label.
In the meantime, @maxmclau just a reminder that we also have a #job-postings channel in our [Discord](https://chat.zephyrproject.org/) that you can use to find a contractor who would do this work for you.

maxmclau, 2023-09-13:
> Since there seems to be some controversy regarding the bounty I've added the TSC label. In the meantime, @maxmclau just a reminder that we also have a #job-postings channel in our [Discord](https://chat.zephyrproject.org/) that you can use to find a contractor who would do this work for you.

Thanks for the suggestion - I'll hop over and post it now

maxmclau, 2023-09-19:
> > Thanks for the suggestion - I'll hop over and post it now
> 
> FWIW I have just updated the bug title and description. Thanks!

Understood - Was there a TSC meeting to discuss bounties? Is there a problem with posting issues in the discord channel?

kartben, 2023-09-19:
> > FWIW I have just updated the bug title and description. Thanks!
> 
> Understood - Was there a TSC meeting to discuss bounties? Is there a problem with posting issues in the discord channel?

Yes, this was discussed last week. The consensus was that bounties shouldn't be discussed/promoted through GitHub issues, and that instead people use #job-postings on Discord or the [Ecosystem Vendor Offerings](https://www.zephyrproject.org/ecosystem-vendor-offerings/) catalog to source candidates willing to help with the enhancement (keeping in mind that there will never be any guarantee that it will end up being merged upstream, as previously mentioned in this thread).

maxmclau, 2023-09-19:
> Yes, this was discussed last week. The consensus was that bounties shouldn't be discussed/promoted through GitHub issues, and that instead people use #job-postings on Discord or the [Ecosystem [...]

Good to know going forward! Had a lot of responses in the #job-postings channel so I'll be sure to use that from now on

abebeos, 2023-09-19:
> The consensus was that bounties shouldn't be discussed/promoted through GitHub issues

Your TSC (-consensus) is a bit out of reality/efficiency/practicality here.

At a minimum, you should us a "bounty" label, ideally together with the "help needed" label (thus issues are easily accessible through the main project listing (https://github.com/orgs/zephyrproject-rtos/repositories).

1.7k issues, 600 PRs => simplify things 

A workaround would be to add a "bounty" category to the "discussions here (in order to avoid forcing people to use this childish discord, which is good for gaming but ridiculously disctracting for serious work).

Me out here anyways, far to complex contribution-processes and inefficient code-structure.

henrikbrixandersen, 2023-09-19:
> Me out here anyways, far to complex contribution-processes and inefficient code-structure.

Thank you for sharing your opinions in a constructive manner.

abebeos, 2023-09-19:
> constructive manner.

Here a bit more constructive:

- #62829