zephyrproject-rtos/zephyr #62519: tests/spi loopback test fails on stm32 platforms

FRASTM, 2023-09-12:
**Describe the bug**
Testing the spi_loopback on stm32 platform fails

Please also mention any information which could help others to understand
the problem you're facing:
 -  might be a regression after merge of https://github.com/zephyrproject-rtos/zephyr/pull/62474

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_wl55jc tests/drivers/spi/spi_loopback/
2. west flash
3. connect SPI MISO to SPI MOSI on the target board
4. See error

**Expected behavior**
```
 PASS - test_spi_loopback in 0.057 seconds                                      
===================================================================             
TESTSUITE spi_loopback succeeded                                                
                                                                                
------ TESTSUITE SUMMARY START ------                                           
                                                                                
SUITE PASS - 100.00% [spi_loopback]: pass = 1, fail = 0, skip = 0, total = 1 dus
 - PASS - [spi_loopback.test_spi_loopback] duration = 0.057 seconds             
                                                                                
------ TESTSUITE SUMMARY END ------ 
```
**Impact**
CI failure

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-3488-g31cd3b1f61e9 ***
Running TESTSUITE spi_loopback
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x200001b1/0x2000086c
I: SPI test slow config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
E: Buffer contents are different: 0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,
E:                            vs: 0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/s)
Buffer contents are different
 FAIL - test_spi_loopback in 0.049 seconds
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain : Zephyr SDK zephyr-sdk-0.16.1
 - Commit SHA 662a14a25e4160d0089af81f518dac0744eb4905

erwango, 2023-09-12:
@eqvinox Can you have a look ?

eqvinox, 2023-09-12:
ah, yes, for the first half (that has a `NULL` buffer) it doesn't wait on RXNE which makes it out of sync when later the rx buffer is not NULL anymore…

need to find some better way to determine whether to RX or not :disappointed:

eqvinox, 2023-09-12:
(to be more clear, yes, the regression is caused by https://github.com/zephyrproject-rtos/zephyr/pull/62474 and reverting it would fix the issue; I'm looking for a better way to do it but if I can't find one in a reasonable amount of time this should just be reverted.)