zephyrproject-rtos/zephyr #62545: MCUBoot: lpcxpresso55s69: compilation error

butok, 2023-09-12:
**Describe the bug**
Compilation of the bootloader\mcuboot\boot\zephyr MCUBoot bootloader raises a compilation error for lpcxpresso55s69_cpu0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to bootloader\mcuboot\boot\zephyr directory
2. west build -p always -b lpcxpresso55s69_cpu0
3. See assert error "Unsupported value for MCUBOOT_BOOT_MAX_ALIGN"

 NOTE: Zephyr MCUBoot compilation guides:
- https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-zephyr.md#building-the-bootloader-itself
- https://docs.zephyrproject.org/latest/develop/west/sign.html#mcuboot-imgtool

**Expected behavior**
No errors.

**Logs and console output**
```
C:\_ddm\ZEPHYR\zephyr_vs\bootloader\mcuboot\boot\zephyr>west build -p always -b lpcxpresso55s69_cpu0
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.4
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.4", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/.cache
-- Zephyr version: 3.4.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: lpcxpresso55s69_cpu0
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (C:/_ddm/ZEPHYR/zephyr_latest/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (C:/_ddm/ZEPHYR/zephyr_latest/zephyr-sdk-0.16.1)
-- Found Dtc: C:/Users/nxa12205/.mcuxpressotools/dtc-msys2/tools/usr/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/arm/lpcxpresso55s69/lpcxpresso55s69_cpu0.dts
-- Found devicetree overlay: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/arm/lpcxpresso55s69/lpcxpresso55s69_cpu0_defconfig'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/prj.conf'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/_ddm/zephyr/zephyr_latest/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/_ddm/ZEPHYR/zephyr_latest/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for LPC55S69:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpc_gpio component is included.
driver_pint component is included.
driver_inputmux component is included.
driver_inputmux_connections component is included.
driver_flexcomm_usart component is included.
driver_flexcomm component is included.
driver_iap1 component is included.
driver_lpc_iocon component is included.
MCUBoot bootloader key file: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done (62.7s)
-- Generating done (0.9s)
-- Build files have been written to: C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build
-- west build: building application
[1/262] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr), build: zephyr-v3.4.0-3537-g4129b10b6900
[41/262] Building C object CMakeFiles/app.dir/C_/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/tlv.c.obj
FAILED: CMakeFiles/app.dir/C_/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/tlv.c.obj
[... 23 lines cut ...]
[53/262] Building C object CMakeFiles/app.dir/C_/_ddm/ZEPH...yr_vs/bootloader/mcuboot/boot/bootutil/src/image_rsa.c.obj
FAILED: CMakeFiles/app.dir/C_/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/image_rsa.c.obj
C:\_ddm\ZEPHYR\zephyr_latest\zephyr-sdk-0.16.1\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -DCPU_LPC55S69JBD100_cm33_core0 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DNDEBUG -DPICOLIBC_INTEGER_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/include -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/targets -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/include -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/include -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/soc/arm/nxp_lpc/lpc55xxx -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/../../zephyr/include -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/.. -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/source/include -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/source/include/platform -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/source/compiler -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/generate/zephyr -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/source/tools/acpiexec -IC:/_ddm/ZEPHYR/zephyr_vs/modules/lib/acpica/source/tools/acpidump -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/cmsis/CMSIS/Core/Include -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/cmsis/. -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69 -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69/drivers -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_gpio -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/pint -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/inputmux -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcomm -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/iap1 -IC:/_ddm/ZEPHYR/zephyr_vs/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_iocon -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/hal_nxp/. -IC:/_ddm/ZEPHYR/zephyr_vs/modules/crypto/mbedtls/include -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/mbedtls/configs -IC:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/mbedtls/include -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/zephyr/.. -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/zephyr/../include -IC:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/zephyr/../../zephyr/include -fno-strict-aliasing -Og -imacros C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/_ddm/ZEPHYR/zephyr_latest/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros C:/_ddm/ZEPHYR/zephyr_vs/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/zephyr=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_vs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_vs=WEST_TOPDIR -ffunction-sections -fdata-sections -mcmse --specs=picolibc.specs -std=c99 -MD -MT CMakeFiles/app.dir/C_/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/image_rsa.c.obj -MF CMakeFiles\app.dir\C_\_ddm\ZEPHYR\zephyr_vs\bootloader\mcuboot\boot\bootutil\src\image_rsa.c.obj.d -o CMakeFiles/app.dir/C_/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/image_rsa.c.obj -c C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/src/image_rsa.c
[... cut ...]
C:/_ddm/ZEPHYR/zephyr_vs/bootloader/mcuboot/boot/bootutil/include/bootutil/bootutil_public.h:88:1: error: static assertion failed: "Unsupported value for MCUBOOT_BOOT_MAX_ALIGN"
   88 | _Static_assert(MCUBOOT_BOOT_MAX_ALIGN >= 8 && MCUBOOT_BOOT_MAX_ALIGN <= 32,
      | ^~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\_ddm\ZEPHYR\zephyr_vs\bootloader\mcuboot\boot\zephyr\build'
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 0.16.1
 - Commit: 4129b10b6900133e41829893c36a01527f293548  Date: 9/12/2023

DerekSnell, 2023-09-12:
Hi @butok ,
This issue is due to a known limitation of MCUboot, documented at https://github.com/mcu-tools/mcuboot/issues/713.  Since this issue documented and tracked in the MCUboot project, you can close this issue here in the Zephyr project.

Best regards

butok, 2023-09-13:
This is the known limitation of MCUBoot.

closed 2023-09-13 by butok

raveslave, 2023-10-23:
@butok did you work on a solution to this in your own fork?

butok, 2023-10-24:
> @butok did you work on a solution to this in your own fork?
Yes.
Several PRs are still pending:
- https://github.com/zephyrproject-rtos/zephyr/pull/63869
- https://github.com/mcu-tools/mcuboot/pull/1833
- also, needed to wait for synchronization of MCUBoot.

raveslave, 2023-11-06:
thanks for info!   
how about this one? https://github.com/mcu-tools/mcuboot/issues/713

also curious if you can share NXP's long-term plan, is it SBL (https://github.com/nxp-mcuxpresso/sbl) or MCUBoot?

raveslave, 2023-11-06:
thanks! is 512-byte writes, 512-byte erase supported?

in other words, should SBL be considered legacy?

butok, 2023-11-07:
> thanks! is 512-byte writes, 512-byte erase supported?

It is supported for non-swap update modes (overwrite-only).

> in other words, should SBL be considered legacy?

It is a different story.