zephyrproject-rtos/zephyr #62546: settings shell: accept (and print) string values

sam-golioth, 2023-09-12:
**Is your enhancement proposal related to a problem? Please describe.**
The settings shell currently only supports hex input and output. This works great for non-string values (integers, floats, byte arrays, serialized structs), but is unnecessarily cumbersome for string values. Users have to convert the string value to a hex representation, and then it is converted back by the shell. E.g. to set the key `foo` to `bar` you have to use:

`settings write foo 626172`

**Describe the solution you'd like**
The settings shell should accept ASCII text input in addition to hex. This can be switched with an optional flag (e.g. `--string`), so that setting `foo` to `bar` would be:

`settings write --string foo bar`

A similar option should be added for reading values:

`settings read --string foo` should print `bar` instead of the hexdump.

**Describe alternatives you've considered**
Golioth currently uses an entirely [separate settings shell](https://github.com/golioth/golioth-zephyr-sdk/blob/main/samples/common/settings_shell.c) (which conflicts with the upstream Zephyr shell) to implement reading and writing of string values.

danielstuart14, 2023-09-12:
Hey Sam,
What do you think about adding an optional argument specifying the data type instead of doing it as a flag?
This way we could expand it to other data types later (uint32_t, float, double, etc)

Maybe something like that:
`settings write string foo bar`
`settings read string foo`

 and also:
`settings write float foo bar`
`settings read float foo`

sam-golioth, 2023-09-12:
> Hey Sam, What do you think about adding an optional argument specifying the data type instead of doing it as a flag? This way we could expand it to other data types later (uint32_t, float, double, [...]

I like it! My concern is around breaking current behavior. Would users need to switch to `settings write hex foo abcdef`? Or is hex assumed if no type is given? Feels kind of weird to me to have hex as the default. I think I'd prefer to have the type be mandatory, but I'm not sure what the general feeling is around breaking changes like that.

danielstuart14, 2023-09-12:
Yeah, I think it would be better for the type to always be explicit, but I also don't like the breaking change.
I think the best compromise would be to let the data type be optional and use hex by default. 

I will work on creating the PR, we can discuss this point there.

closed 2023-09-22 by carlescufi