zephyrproject-rtos/zephyr #62560: twister: platform exclude not working when using prepared artifacts

pavloinfineon, 2023-09-12:
**Describe the bug**
When running twister device testing on artifacts ready for testing, --platform and --exclude-platform seem to have no effect.

I would expect that with 3 platforms in the artifacts, already compiled and test cases in the .json files, that running the device-testing twister command to only test a single platform should ignore all the other platforms in the testplan.json and only test the board defined with the flag --platform. Below I am trying to test xmc47_relax_kit but ignore the 2 other boards.
It seems to just be going off the testplan.json/twister.json file and ignoring the flags

**To Reproduce**
Steps to reproduce the behavior:
1. `python3 scripts/twister -p cy8cproto_062_4343w -p cy8cproto_063_ble -p xmc47_relax_kit -T samples -v -i -j10 -n --prep-artifacts-for-testing` (meant to be run on a different computer but fails both ways)
2. `west twister -p xmc47_relax_kit --device-testing --hardware-map map.yml -T samples/ -v -W -n --test-only` OR
`west twister -p xmc47_relax_kit -P cy8cproto_062_4343w -P cy8cproto_063_ble --device-testing --hardware-map map.yml -T samples/ -v -W -n --test-only`
4. See that twister will show lots of errors and show failures because of `No device to serve as cy8cproto_063_ble platform.` error.

**Expected behavior**
Twister should only test the platforms that are defined by the --platform flag when reusing artifacts

**Impact**
Should be addressed

**Logs and console output**
Used `samples/basic` for quicker run time
Doing `$ west twister -p xmc47_relax_kit --exclude-platform cy8cproto_062_4343w --device-testing --hardware-map map.yml -T samples/basic -v -W -n --test-only` also results in the same logs
```
$ python3 scripts/twister -p cy8cproto_062_4343w -p xmc47_relax_kit -T samples/basic -v --prep-artifacts-for-testing
ZEPHYR_BASE unset, using "C:\Users\havrylyuk\zephyr\zephyr"
Renaming output directory to C:\Users\havrylyuk\zephyr\zephyr\twister-out.3
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3509-g662a14a25e41
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report C:\Users\havrylyuk\zephyr\zephyr\twister-out\testplan.json
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
see: C:\Users\havrylyuk\zephyr\zephyr\twister-out\cy8cproto_062_4343w\samples\basic\hash_map\libraries.hash_map.picolibc.separate_chaining.djb2\build.log

INFO    - 28 test scenarios (56 test instances) selected, 37 configurations skipped (34 by static filter, 3 at runtime).
INFO    - 18 of 56 test configurations passed (94.74%), 0 failed, 1 errored, 37 skipped with 0 warnings in 411.03 seconds
INFO    - 0 test configurations executed on platforms, 19 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report C:\Users\havrylyuk\zephyr\zephyr\twister-out\twister.json
INFO    - Writing xunit report C:\Users\havrylyuk\zephyr\zephyr\twister-out\twister.xml...
INFO    - Writing xunit report C:\Users\havrylyuk\zephyr\zephyr\twister-out\twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples\basic\hash_map\libraries.hash_map.picolibc.separate_chaining.djb2 on cy8cproto_062_4343w error (Build failure)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p cy8cproto_062_4343w -s samples\basic\hash_map\libraries.hash_map.picolibc.separate_chaining.djb2
INFO    - or with west:
INFO    - west build -p -b cy8cproto_062_4343w -T samples\basic\hash_map\libraries.hash_map.picolibc.separate_chaining.djb2
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed

$ west twister -p xmc47_relax_kit --device-testing --hardware-map map.yml -T samples/basic -v -W -n --test-only
Keeping artifacts untouched
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3509-g662a14a25e41
INFO    - Using 'zephyr' toolchain.

Device testing on:

| Platform        |           ID | Serial device   |
|-----------------|--------------|-----------------|
| xmc47_relax_kit | 850231222109 | COM8            |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
No device to serve as cy8cproto_062_4343w platform.
[... 30 lines cut ...]
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples\basic\hash_map\libraries.hash_map.minimal.separate_chaining.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 2) samples\basic\hash_map\libraries.hash_map.minimal.open_addressing.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 3) samples\basic\hash_map\libraries.hash_map.newlib.separate_chaining.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 4) samples\basic\hash_map\libraries.hash_map.newlib.open_addressing.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 5) samples\basic\hash_map\libraries.hash_map.newlib.cxx_unordered_map.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 6) samples\basic\hash_map\libraries.hash_map.picolibc.separate_chaining.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 7) samples\basic\hash_map\libraries.hash_map.picolibc.open_addressing.djb2 on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    - 8) samples\basic\sys_heap\sample.basic.sys_heap on cy8cproto_062_4343w failed (No device to serve as cy8cproto_062_4343w platform.)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p cy8cproto_062_4343w -s samples\basic\sys_heap\sample.basic.sys_heap
INFO    - or with west:
INFO    - west build -p -b cy8cproto_062_4343w -T samples\basic\sys_heap\sample.basic.sys_heap
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

**Environment (please complete the following information):**
 - OS: Linux/Windows
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: a7df4295e2539fdbf0938a4e755adb8f00825170

pavloinfineon, 2023-09-14:
Created a PR with purposed test-only platform filtering
https://github.com/zephyrproject-rtos/zephyr/pull/62671

nashif, 2023-09-18:
PR merged

closed 2023-09-18 by nashif