zephyrproject-rtos/zephyr #62574: tests/kernel/mem_protect fails on stm32f746 nucleo: memory partitionning error

FRASTM, 2023-09-13:
**Describe the bug**
Running the 

Error due to memory partitionning on the nucleo_f746zg platrform

Please also mention any information which could help others to understand
the problem you're facing:
 - Is this a regression ? seems to be related to  commit 
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_f746zg tests/kernel/mem_protect/userspace/
2. west flash
3. See error

**Expected behavior**
`PROJECT EXECUTION SUCCESSFUL`

**Impact**
CI test failure

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-3253-g844c78cf3a7e ***
Running TESTSUITE userspace
===================================================================
E: no free partition slots available
Failed to add default memory partition (-28)
E: r0/a1:  0x00000003  r1/a2:  0x0800e319  r2/a3:  0x40004800
E: r3/a4:  0x08003729 r12/ip:  0x0000000c r14/lr:  0x08000d99
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x08000da4
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x200113f0 (main)
Caught system error -- reason 3
Unexpected fault during test
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA 07ece0cb1dfa9a1aaf7af3fe94792cc3b7572fa0

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

fabiobaltieri, 2023-09-19:
@keith-packard could you look into this? Looks like it's been rootcaused to the picolibc commit

keith-packard, 2023-09-19:
I don't think I have this particular board, so I can't directly evaluate the issue. I can this of a couple of possible causes:

 1. Default malloc arena. Minimal libc has a zero-sized arena, picolibc has a non-zero sized. You can check this by forcing the arena to zero size under.
 2. thread local storage. https://github.com/zephyrproject-rtos/zephyr/pull/62319 allows you to disable this while using picolibc.

Both of these generate additional memory partitions which might be causing this issue? I fixed numerous test issues caused by assumptions in the code about the precise memory configuration provided in the minimal C library build, I suspect this is likely to have a similar cause.

If there's some way I can test this, let me know. I have a stack of nucleo and disco boards here, but not this particular one, alas.

keith-packard, 2023-09-19:
In the case of thread local storage on targets without a TLS base register, like many small single-core ARM SoCs, the TLS base pointer is saved in memory in the z_libc_partition, and that partition doesn't contain anything else. Is there some other partition, shared between kernel and user space, which could be used to hold this value? Creating a partition for a single integer doesn't seem ideal.

FRASTM, 2023-11-03:
Setting the `CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=0`  for the nucleo_f746zg board is not enough. The quadspi_memory `compatible = "zephyr,memory-region"` also has to be removed from the dts/arm/st/f7/stm32f7.dtsi  to PASS the test

FRASTM, 2024-02-20:
not failing since Zephyr v3.6.0-rc3

closed 2024-02-20 by FRASTM