zephyrproject-rtos/zephyr #62577: The STM32 USB device driver can only receive one message when the maximum packet length of an endpoint of the USB_EP_TYPE_BULK out type is 16

maweicoltsmart, 2023-09-13:
I discovered this issue while developing my own USB device, and I tried both USB device stacks and had the same problem. 
Through debugging, I found that the function HAL_PCD_DataOutStageCallback showed that the host sent 64 bytes, but when it was called to my program, it was only 16 bytes. I need to get the remaining 48 bytes. When I give the function ```
int usb_transfer(uint8_t ep, uint8_t *data, size_t dlen, unsigned int flags,
		 usb_transfer_callback cb, void *priv)
``` When the parameter DLEN is changed to 112, I can receive all 112 bytes, but at this time, the stm32 driver will only call back to my program if it accumulates enough 112 bytes each time.

maweicoltsmart, 2023-09-14:
@erwango I merged the two commits, but the function `HAL_PCD_DataOutStageCallback` not called when the host sent the data。`wMaxPacketSize` set to 16 bytes, function 
```
int usb_transfer(uint8_t ep, uint8_t *data, size_t dlen, unsigned int flags,
   	 usb_transfer_callback cb, void *priv) 
```
parameter `dlen` is also set to 16 bytes,
I changed the value of the parameter `dlen` to something else, such as 64 bytes, and still didn't receive any data
Data is received correctly only if both `wMaxPacketSize` and `dlen` are set to 64 bytes, 16 bytes at a time.

Desvauxm-st, 2023-09-28:
@maweicoltsmart
I have in charge to reproduce the bug
and sorry it is not clear enough for me
How can I do to reproduce the bug you face quikely
can i use amples\subsys\usb\cdc_acm ?
or do do have a special part of code to reproduce that ?

maweicoltsmart, 2023-10-08:
@Desvauxm-st The `usb_dc_ep_read_wait` function in file `usb_dc_stm32.c` causes excess data not to be processed, which creates the problem I mentioned earlier.
```
int usb_dc_ep_read_wait(uint8_t ep, uint8_t *data, uint32_t max_data_len,
			uint32_t *read_bytes)
{
	struct usb_dc_stm32_ep_state *ep_state = usb_dc_stm32_get_ep_state(ep);
	uint32_t read_count;

	if (!ep_state) {
		LOG_ERR("Invalid Endpoint %x", ep);
		return -EINVAL;
	}

	read_count = ep_state->read_count;

	LOG_DBG("ep 0x%02x, %u bytes, %u+%u, %p", ep, max_data_len,
		ep_state->read_offset, read_count, data);

	if (!USB_EP_DIR_IS_OUT(ep)) { /* check if OUT ep */
		LOG_ERR("Wrong endpoint direction: 0x%02x", ep);
		return -EINVAL;
	}

	/* When both buffer and max data to read are zero, just ignore reading
	 * and return available data in buffer. Otherwise, return data
	 * previously stored in the buffer.
	 */
	if (data) {
		read_count = MIN(read_count, max_data_len);
		memcpy(data, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)] +
		       ep_state->read_offset, read_count);
		ep_state->read_count -= read_count;
		ep_state->read_offset += read_count;
	} else if (max_data_len) {
		LOG_ERR("Wrong arguments");
	}

	if (read_bytes) {
		*read_bytes = read_count;
	}

	return 0;
}
```

When the `max_data_len` is less than `ep_state->read_count`, my callback function will only read the data of `max_data_len` length, and will not trigger my callback function to process the rest of the data and just keep it in the buffer of the USB driver.

This made it impossible for my program to continue. My program needs to process the remaining data many times in a row until all the data is processed, is there any good way to solve it?

erwango, 2023-10-09:
> This made it impossible for my program to continue. My program needs to process the remaining data many times in a row until all the data is processed, is there any good way to solve it?

@loicpoulain Maybe ?

loicpoulain, 2023-10-09:
@maweicoltsmart can you enable debug log for both with both debug log for usb_dc_stm32 and usb device stack (CONFIG_USB_DEVICE_LOG_LEVEL) highlighting the issue? Would be good if you also have an example of app code triggering the issue so that we can reproduce or at least see the use case.

maweicoltsmart, 2023-10-09:
@loicpoulain
logs as below:
```
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_start 0x200001e4
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_end 0x20000270
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Device descriptor 0x200001e4
[00:00:00.000,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 0, mps 64, type 0
[00:00:00.000,000] <dbg> usb_descriptor: usb_desc_update_mps0: Set bMaxPacketSize0 64
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Configuration descriptor 0x200001f6
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Interface descriptor 0x200001ff
[00:00:00.000,000] <dbg> can_usb: pcan_usb_interface_config: pcan_usb_interface_config, 1450
[00:00:00.000,000] <dbg> can_usb: pcan_usb_interface_config: pcan_usb_interface_config, 1459
[00:00:00.001,000] <dbg> can_usb: pcan_usb_interface_config: pcan_usb_interface_config, 1485
[00:00:00.001,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x20000208
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 1, mps 16, type 2
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 1 -> 1
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x1
[00:00:00.001,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x2000020f
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 81, mps 16, type 2
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 81 -> 81
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x81
[00:00:00.001,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x20000216
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 2, mps 64, type 2
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 2 -> 2
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x2
[00:00:00.001,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x2000021d
[00:00:00.001,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 82, mps 64, type 2
[00:00:00.001,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 82 -> 82
--- 167 messages dropped ---
[00:00:00.221,000] <dbg> usb_dc_stm32: usb_dc_ep_write: ep 0x80, len 4
[00:00:00.221,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.221,000] <dbg> usb_dc_stm32: HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.221,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.221,000] <dbg> usb_dc_stm32: HAL_PCD_SetupStageCallback: 
[00:00:00.221,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.221,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: ep 0x%02x, %u bytes, %u+%u, %p" argument:5
[00:00:00.221,000] <dbg> usb_dc_stm32: usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20003630
[00:00:00.222,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x00, len 64
[00:00:00.222,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0409
[00:00:00.222,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.222,000] <dbg> usb_dc_stm32: usb_dc_ep_write: ep 0x80, len 34
[00:00:00.222,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.222,000] <dbg> usb_dc_stm32: HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.222,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.222,000] <dbg> usb_dc_stm32: HAL_PCD_SetupStageCallback: 
[00:00:00.222,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.222,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: ep 0x%02x, %u bytes, %u+%u, %p" argument:5
[00:00:00.223,000] <dbg> usb_dc_stm32: usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20003630
[00:00:00.223,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x00, len 64
[00:00:00.223,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0409
[00:00:00.223,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.223,000] <dbg> usb_dc_stm32: usb_dc_ep_write: ep 0x80, len 18
[00:00:00.223,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.223,000] <dbg> usb_dc_stm32: HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.223,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.223,000] <dbg> usb_dc_stm32: HAL_PCD_SetupStageCallback: 
[00:00:00.223,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.224,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: ep 0x%02x, %u bytes, %u+%u, %p" argument:5
[00:00:00.224,000] <dbg> usb_dc_stm32: usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20003630
[00:00:00.224,000] <dbg> usb_dc_stm32: usb_dc_ep_start_read: ep 0x00, len 64
[00:00:00.224,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0409
[00:00:00.224,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.224,000] <dbg> usb_dc_stm32: usb_dc_ep_write: ep 0x80, len 18
[... cut ...]
[00:00:00.227,000] <dbg> usb_dc_stm32: usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x20003630
[... 325 lines cut ...]
	LOG_DBG("read continue");
	usb_ep_read_continue(ep);

}
......
/* usb.rst endpoint configuration start */
static struct usb_ep_cfg_data ep_cfg[] = {
	{
		.ep_cb = usb_cmdin_cb,
		.ep_addr = 0x82,
	},
	{
		.ep_cb = usb_cmdout_cb,
		.ep_addr = 0x01,
	},
	{
		.ep_cb = usb_transfer_ep_callback,
		.ep_addr = 0x81,
	},
	{
		.ep_cb = usb_transfer_ep_callback,
		.ep_addr = 0x02,
	},
};
```

maweicoltsmart, 2023-10-09:
@loicpoulain
In my application, `wMaxPacketSize` is 16 bytes, but in the following code `EP_MPS` 64 bytes, I guess that's where the problem lies.
```
int usb_dc_ep_read_continue(uint8_t ep)
{
	struct usb_dc_stm32_ep_state *ep_state = usb_dc_stm32_get_ep_state(ep);

	if (!ep_state || !USB_EP_DIR_IS_OUT(ep)) { /* Check if OUT ep */
		LOG_ERR("Not valid endpoint: %02x", ep);
		return -EINVAL;
	}

	/* If no more data in the buffer, start a new read transaction.
	 * DataOutStageCallback will called on transaction complete.
	 */
	if (!ep_state->read_count) {
		usb_dc_ep_start_read(ep, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],
				     EP_MPS);
	}

	return 0;
}
```

maweicoltsmart, 2023-10-10:
@loicpoulain
When I make the following changes, my program will work, but this is not a good solution, is there a better solution?

```
index c5516b73ae..38b0cd5b1b 100644

--- a/drivers/usb/device/usb_dc_stm32.c

+++ b/drivers/usb/device/usb_dc_stm32.c

@@ -889,8 +889,13 @@ int usb_dc_ep_read_continue(uint8_t ep)

         * DataOutStageCallback will called on transaction complete.

         */

        if (!ep_state->read_count) {

-               usb_dc_ep_start_read(ep, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],

+               if(ep == 0x01) {

+                       usb_dc_ep_start_read(ep, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],

+                                    16);

+               } else {

+                       usb_dc_ep_start_read(ep, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],

                                     EP_MPS);

+               }

        }

 

        return 0;

```

loicpoulain, 2023-10-10:
@maweicoltsmart  can you try that change:

```
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -751,7 +751,7 @@ int usb_dc_ep_enable(const uint8_t ep)
        if (USB_EP_DIR_IS_OUT(ep) && ep != EP0_OUT) {
                return usb_dc_ep_start_read(ep,
                                          usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],
-                                         EP_MPS);
+                                         ep_state->ep_mps);
        }
 
        return 0;
@@ -890,7 +890,7 @@ int usb_dc_ep_read_continue(uint8_t ep)
         */
        if (!ep_state->read_count) {
                usb_dc_ep_start_read(ep, usb_dc_stm32_state.ep_buf[USB_EP_GET_IDX(ep)],
-                                    EP_MPS);
+                                    ep_state->ep_mps);
        }
 
        return 0;
```

maweicoltsmart, 2023-10-10:
@loicpoulain
Only modifying the function `int usb_dc_ep_read_continue (uint8_t ep) `can run perfectly, but if you modify the function `int usb_dc_ep_enable (const uint8_t ep)` together, the host will die after running for a period of time and the device will run normally, I don't know why the host is like this, because the host's program is not designed by me. However, it cannot be ruled out whether such a change will cause any other problems.

maweicoltsmart, 2023-10-10:
@loicpoulain
To correct, both of the changes in the patch you gave will cause the host to freeze, but it does solve the problem mentioned earlier that the data cannot be fully received, and I think this patch can be committed.

maweicoltsmart, 2023-11-03:
> @maweicoltsmart can you try that change:
>
> ```
> +++ b/drivers/usb/device/usb_dc_stm32.c
> @@ -751,7 +751,7 @@ int usb_dc_ep_enable(const uint8_t ep)
>         if (USB_EP_DIR_IS_OUT(ep) && ep != EP0_OUT) [...]

This patch solved my problem perfectly, my program is running very well now, can you submit it?

Desvauxm-st, 2023-12-13:
@loicpoulain
can you push a PR with the modifications you propose because this solves the problem encountered ?