zephyrproject-rtos/zephyr #62584: wifi_mgmt: `struct wifi_scan_params` is now very large

JordanYates, 2023-09-13:
**Is your enhancement proposal related to a problem? Please describe.**

With the addition of #60686, the `struct wifi_scan_params` field is now huge, over 1kB.
This makes it very awkward to create an instance of this in a function call.

The `chan` parameter is extremely un-optimised.
According to [Wikipedia](https://en.wikipedia.org/wiki/List_of_WLAN_channels), the 2.4 GHz band only has 14 channels total.
Therefore `chan[0]` contains 219 slots that can never contain useful information, or 438 bytes. By my count, there are 60 valid channels in the 5GHz band, resulting in 173 useless slots, or 346 bytes.

`WIFI_CHANNEL_MAX` is not the correct define to be using in the instantiation of this array.

**Describe the solution you'd like**

The structure should not be larger than many thread stacks.

**Describe alternatives you've considered**

There are a few options, which can be mixed and matched.

* Immediately swap `WIFI_CHANNEL_MAX` to 60 (`WIFI_CHANNEL_NUM_ASSIGNED`)
* Split the two bands into separate arrays so 2.4 GHz doesn't pay the cost of 5GHz channels
* Use defines to compress the structure to only assigned channels.
* Use a bitmask to specify channels instead of integers.

Implementing all of these suggestions would reduce the `chan` field to 10 bytes, down from 932.

JordanYates, 2023-09-13:
@sachinthegreen

jukkar, 2023-09-13:
Very good points, I have to say I missed the memory usage problem during the review, sorry about that :disappointed:

PR #62735 (2023-09-25): net: wifi_mgmt: reduce `struct wifi_scan_params` size
These commits reduce the overall size of `struct wifi_scan_params` in its default configuration from 1412 bytes to 20 bytes.

The only driver I am aware of that implements these advanced scanning functions is the Nordic nRF700x [OOT driver](https://github.com/nrfconnect/sdk-nrf/blob/main/drivers/wifi/nrf700x/zephyr/src/zephyr_wifi_mgmt_scan.c).
The two API changes are trivial to integrate into that driver as:
 * `ssids` is being memcpy'ed to local memory
 * The `chan` nested array is iterated over to create a linear frequency array

Shouldn [... cut]
closed 2023-09-25 by carlescufi