zephyrproject-rtos/zephyr #62597: Getting 1 second difference in the output of the RTC when using stm32h743ZIT controller

ashutoshpandey-eaton, 2023-09-13:
**I am using Zephyr 3.2.0 in our project and When I am trying to get RTC output at every 10 seconds**,
Sometimes I am getting 1 sec less than expected.(Refer "Result" in the attachment)

**Steps to reproduce**

Zephyr version used:3.2.0
Hardware used:Nucleo-H743ZI2 development board
Build command:west build -b nucleo_h743zi ./ --pristine

1.For demo purpose,Used blinky folder which is at "zephyr-v3.2.0\zephyr\samples\basic\blinky",
slightly modified 2 files in the bliky sample with my code 
1.main.c (attached)
2.prj.conf (attached)[in this added two more symbols CONFIG_PRINTK=y CONFIG_COUNTER=y]

2.Reading rtc output (seconds) value from main.c(attached) by calling counter_get_value(..) function at every 10 sec interval and 
using a print statement in rtc_stm32_read()in counter_ll_stm32_rtc.c, which is at "zephyr-v3.2.0\zephyr\drivers\counter" (refer attachement :print in counter_ll_stm32_rtc.c )

3.When looking at the serial terminal using above mentioned setup: most of the times I am getting the 10-second difference,
but sometimes the difference is 9 seconds (so getting 1-second difference)

**NOTE:** 
[blinky.zip](https://github.com/zephyrproject-rtos/zephyr/files/12597148/blinky.zip)
**Refer: blinky.zip:for the files used (This contain changed main.c and prj.conf and final build folder which got created)**

print in counter_ll_stm32_rtc.c 
[image]

**Result:**
[image]

erwango, 2023-09-14:
I've change the sample to iterate each 2 sec (vs 10 initially)
Adding subsecond reading in the rtc counter driver, we can see there is a 6/7 subsecond drift each 2 seconds.
Once ss > 256, the correction happens:
```
tick:71, uptime:2004                              
now.tm_sec=13, subseconds:239 
tick:73, uptime:2005                           > 2 sec
now.tm_sec=15, subseconds:246 
tick:75, uptime:2005                           > 2 sec
now.tm_sec=17, subseconds:253      > Subsecond wrapup
tick:77, uptime:2004                           >>> 1 sec
now.tm_sec=18, subseconds:4 
tick:78, uptime:2005                           > 2 sec
now.tm_sec=20, subseconds:11 
tick:80, uptime:2004                           > 2 sec
now.tm_sec=22, subseconds:17 
```
Note, I've used LSE instead of LSI, which doesn't bring improvement on the drift.

erwango, 2023-09-14:
@ashutoshpandey-eaton Before digging further into his, please note that this is likely due to the fact that RTC is programmed with a non configurable 1Hz clock, which opens the way to some dift in any case.
Please note that you can also use counter API with STM32 timer with configurable prescaler ending in better accuracy. Would it better fit your need ?

ashutoshpandey-eaton, 2023-09-14:
- "RTC is programmed with a non-configurable 1Hz clock", I didn't get what exactly you are referring to, if you can explain that will be helpful.
- I cannot use counter API with STM32 timer, it doesn't suffice my purpose, I want to use RTC only. If you can suggest any other way by which this drift can be avoided/reduced.

erwango, 2023-10-02:
Need to check H7 errata sheet about SSR reload

FRASTM, 2023-10-09:
@ashutoshpandey-eaton is there any possibility for you to test with a more recent zephyr version and with LSE as rtc clock source for the RTC os the stm32h743. I do not see any shift
```
*** Booting Zephyr OS build v3.5.0-rc1-160-*
RTC device OK
 time now = 0 seconds
ticks=0
 time now = 10 seconds
ticks=10
 time now = 20 seconds
ticks=20
 time now = 30 seconds
ticks=30
 time now = 40 seconds
ticks=40
 time now = 50 seconds
ticks=50
 time now = 0 seconds
ticks=60
 time now = 10 seconds
ticks=70
 time now = 20 seconds
ticks=80
 time now = 30 seconds
```
Compared with RTC clocked by LSI:
```
*** Booting Zephyr OS build v3.5.0-rc1-160-*
RTC device OK
 time now = 0 seconds (248)
ticks=0
 time now = 10 seconds (187)
ticks=10
 time now = 20 seconds (126)
ticks=20
 time now = 30 seconds (65)
ticks=30
 time now = 40 seconds (5)
ticks=40
 time now = 51 seconds (194)
ticks=51
 time now = 1 seconds (133)
ticks=61
 time now = 11 seconds (72)
ticks=71
```

erwango, 2023-12-11:
@ashutoshpandey-eaton Any feedback on the impact on changing the clock source ?

closed 2024-02-26 by github-actions[bot]