zephyrproject-rtos/zephyr #62612: New build warning since switching to Picolibc

fabiobaltieri, 2023-09-13:
**Describe the bug**
Hi, bumped into few new warnings in the weekly run related to the picolibc switch:

1 - `west build -p -b s32z270dc2_rtu0_r52 -T samples/net/sockets/can/sample.net.sockets.can.two_sockets`

```
zephyrproject/zephyr/drivers/can/can_nxp_s32_canxl.c:757:9: warning: '__builtin___memcpy_chk' forming offset [16, 71] is out of the bounds [0, 16] of object 'frame' with type 'struct can_frame' [-Warray-bounds]
```

`can_dlc_to_bytes` can potentially return up to 64 bytes, the receiving structure holds either 8 or 64 depending on the configuration, opened https://github.com/zephyrproject-rtos/zephyr/pull/62611 for it

2 - `west build -p -b mg100 -T samples/net/telnet/sample.net.telnet`

```
zephyrproject/zephyr/drivers/modem/hl7800.c:1786:74: warning: '%s' directive output may be truncated writing up to 53 bytes into a region of size 21 [-Wformat-truncation=]
```

This is more interesting, the `snprintk` return code is handled, and if I understand the warning, it only triggers if the compiler detects that the return value is not used. That said, the handling is a `LOG_WRN`, which becomes a noop and I guess causes the return value to not being used and the warning to fire if logging is off.

I could not figure out a workaround for this, so it may need a `#pragma GCC diagnostic ignored "-Warray-bounds"`. Open to ideas.

3 -``west build -p -b cc3220sf_launchxl -T tests/net/lib/mqtt_publisher/net.mqtt`

`zephyrproject/zephyr/include/zephyr/net/socket_select.h:111:16: error: redefinition of 'struct zsock_fd_set'`

and

```
/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/picolibc/include/sys/select.h:109:5: error: conflicting types for 'select'; have 'int(int,  zsock_fd_set *, zsock_fd_set *, zsock_fd_set *, struct timeval *)'
...
/usr/local/google/home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/net/socket_select.h:114:19: note: previous definition of 'select' with type 'int(int,  zsock_fd_set *, zsock_fd_set *, zsock_fd_set *, struct timeval *)'
```

This one seems to trigger because there's an `fd_set` and `select` defined in `/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/picolibc/include/sys/select.h`, the Zephyr specific select code is then defining its own.

Not too sure how to handle that either, maybe we should use the picolibc definitions when enabled? Tried to hack together a change to do that but it did not work, this may need more than a oneline hotfifx. :-)

@keith-packard do you have any idea of how we could handle the ones above?

**To Reproduce**
See the commands above.

**Expected behavior**
No warnings.

**Impact**
Weekly CI breakage, no super urgent but should be fixed soon, ideally before rc1.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 16.1
 - a6756c36e9

keith-packard, 2023-09-13:
> **Describe the bug** Hi, bumped into few new warnings in the weekly run related to the picolibc switch:
>
> 1 - `west build -p -b s32z270dc2_rtu0_r52 -T [...]

Maybe just switch to strncpy (yeah, it's hard to use correctly) instead of using snprintk to copy a string?

> This one seems to trigger because there's an `fd_set` and `select` defined in `/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/picolibc/include/sys/select.h`, the Zephyr specific select code is then defining its own.

Yeah, we pretty clearly want to use the zephyr definition instead of the picolibc one -- the zephyr version provides a static inline implementation on top of zsock_select. 

> Not too sure how to handle that either, maybe we should use the picolibc definitions when enabled? Tried to hack together a change to do that but it did not work, this may need more than a oneline hotfifx. :-)

How is your code managing to include both versions of sys/select.h ? I would have though the one in include/zephyr/posix would have replaced the picolibc one?

fabiobaltieri, 2023-09-13:
> How is your code managing to include both versions of sys/select.h ? I would have though the one in include/zephyr/posix would have replaced the picolibc one?

Good question, looks like the include chain is something like:
```
zephyrproject/zephyr/drivers/wifi/simplelink/simplelink_support.c
zephyrproject/modules/hal/ti/simplelink/source/ti/net/slnetif.h
zephyrproject/modules/hal/ti/simplelink/source/ti/net/slnetsock.h
/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/picolibc/include/sys/time.h
/opt/zephyr-sdk-0.16.1/arm-zephyr-eabi/picolibc/include/sys/select.h
```

There's no `sys/select.h` in the Zephyr `include`, there's `include/zephyr/posix/sys/select.h` and then `include/zephyr/net/socket_select.h`.

Should the posix file there take priority? (cc @cfriedt)

Dropping https://github.com/zephyrproject-rtos/hal_ti/blob/master/simplelink/source/ti/net/slnetsock.h#L201 fixes it as well.

keith-packard, 2023-09-13:
> Should the posix file there take priority over the library one? (cc @cfriedt)

I think so -- the library one is just providing a declaration of a function which might exist in the underlying OS -- the Zephyr one is what the underlying OS actually provides.

> Dropping https://github.com/zephyrproject-rtos/hal_ti/blob/master/simplelink/source/ti/net/slnetsock.h#L201 fixes it as well.

Hrm. I wonder how sys/time.h ends up pulling in sys/select.h. Twisty include files.

cfriedt, 2023-09-13:
Should probably just remove this from that header.

cfriedt, 2023-09-13:
> Dropping https://github.com/zephyrproject-rtos/hal_ti/blob/master/simplelink/source/ti/net/slnetsock.h#L201 fixes it as well.

I think there are a couple of instances around the codebase that have a similar fix.

Might more more of an if / else situation rather than just if.

I'll take a peak in about an hour

fabiobaltieri, 2023-09-14:
@cfriedt I realize only now that that chunk of code in `simplelink/source/ti/net/slnetsock.h` was meant to prevent exactly this --- is this right? https://github.com/zephyrproject-rtos/hal_ti/pull/32/files#diff-321a64fa87bae1847e9c0db1783858b4c704a99de9b8e7a9f3d591a533ad1922 feels like it unintentionally inverted the condition.

fabiobaltieri, 2023-09-19:
All fixed, thanks for the help!