zephyrproject-rtos/zephyr #62619: Hard Fault in `sdmmc_spi_read_cxd()` due to `sd_card.card_buffer` alignment.

attie-argentum, 2023-09-13:
**Describe the bug**

Since a85ffa8, I reliably get a hard-fault in `sdmmc_spi_read_cxd()` when dereferencing `&sd_card.card_buffer` when it's cast to `uint32_t *` (line 221 below).

https://github.com/zephyrproject-rtos/zephyr/blob/38c918164217965c92b04f23ca894f926bc11219/subsys/sd/sd_ops.c#L195-L224

Note that `sd_card.card_buffer` is tagged for alignment using `CONFIG_SDHC_BUFFER_ALIGNMENT`, which defaults to `1`, and is set to `4` in some instances...

https://github.com/zephyrproject-rtos/zephyr/blob/38c918164217965c92b04f23ca894f926bc11219/include/zephyr/sd/sd.h#L43-L70

It appears that this probably only worked by chance previously, with `.card_buffer` happening to fall on a 32-bit word boundary...
Peeling back the changes from my starting point of 38c9181 shows the change in size of the structure, and the offset to that member:

- HEAD / 38c9181: `sizeof(struct sd_card): 180,  __builtin_offsetof (struct sd_card, card_buffer): 113`
- reverting a85ffa8: `sizeof(struct sd_card): 176,  __builtin_offsetof (struct sd_card, card_buffer): 112`
- also reverting b2172a9: `sizeof(struct sd_card): 192,  __builtin_offsetof (struct sd_card, card_buffer): 128`

I'm surprised an alignment of `1` is ever used, let alone the default (given that this is intended to be accessed as 32-bit values).

**To Reproduce**
I don't have publicly available dev boards to hand, but will see if I can put something together...

**Expected behavior**
The SD card is accessible, and no hard fault presents

**Impact**
"_Annoyance_", and derailed development to chase down the bug... will likely be "_showstopper_" for those less experienced.

**Logs and console output**
```
PK: WEST_TOPDIR/zephyr/subsys/sd/sd_ops.c:222 sdmmc_spi_read_cxd(): sizeof(struct sd_card): 180,  __builtin_offsetof (struct sd_card, card_buffer): 113
PK: WEST_TOPDIR/zephyr/subsys/sd/sd_ops.c:224 sdmmc_spi_read_cxd(): i: 0,  &(cxd_be[i]): 0x20000161
PK: WEST_TOPDIR/zephyr/subsys/sd/sd_ops.c:225 sdmmc_spi_read_cxd(): i: 0,  ((uint8_t*)cxd_be)[i]: 0x74
[00:00:00.255,000] <err> os: ***** HARD FAULT *****
[00:00:00.255,000] <err> os: r0/a1:  0x2000035c  r1/a2:  0x0000001c  r2/a3:  0x00000000
[00:00:00.255,000] <err> os: r3/a4:  0x00007f17 r12/ip:  0x00000003 r14/lr:  0x00011dab
[00:00:00.255,000] <err> os:  xpsr:  0x81000000
[00:00:00.255,000] <err> os: Faulting instruction address (r15/pc): 0x00007f16
[00:00:00.256,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.256,000] <err> os: Current thread: 0x20001b60 (main)
[00:00:00.343,000] <err> os: Halting system
```

The fault above occurs inside `sdmmc_spi_read_cxd()`, while attempting to print (and thus dereference) `cxd_be[i]`.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: `zephyr-sdk-0.16.1` & `arm-gnu-toolchain-12.3.rel1-x86_64-arm-none-eabi`
 - Commit: 38c9181

attie-argentum, 2023-09-13:
# Steps to Reproduce

1. Use an ATSAML21-XPRO, and connect an ATIO1-XPRO to the EXT1 header.
2. Place `atsaml21_xpro.overlay` (below) in `.../zephyr/samples/subsys/fs/fs_sample/boards/`
3. `cmake -GNinja .../zephyr/samples/subsys/fs/fs_sample -D BOARD=atsaml21_xpro && ninja && ninja flash`
4. Observe "_hard fault_" output (below)
5. Apply [`0001-DEBUG-force-sd_card.card_buffer-alignment-to-4-bytes.patch`](https://github.com/zephyrproject-rtos/zephyr/files/12601920/0001-DEBUG-force-sd_card.card_buffer-alignment-to-4-bytes.patch)
6. Rebuild, and observe "_success_" output (below)

### `atsaml21_xpro.overlay`

```
#include <freq.h>

&sercom0 {
	status = "okay";
	cs-gpios = <&porta 5 GPIO_ACTIVE_LOW>;

	sdhc0: sdhc@0 {
		compatible = "zephyr,sdhc-spi-slot";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <DT_FREQ_M(1)>;

		mmc {
			compatible = "zephyr,sdmmc-disk";
			status = "okay";
		};
	};
};
```

### Hard Fault

```
*** Booting Zephyr OS build zephyr-v3.4.0-3659-g4b70723faafb ***
[00:00:00.041,000] <err> os: ***** HARD FAULT *****
[00:00:00.041,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:00.041,000] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000003 r14/lr:  0x0000662b
[00:00:00.041,000] <err> os:  xpsr:  0x01000000
[00:00:00.041,000] <err> os: Faulting instruction address (r15/pc): 0x000080ac
[00:00:00.041,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.041,000] <err> os: Current thread: 0x200007e8 (unknown)
[00:00:00.092,000] <err> os: Halting system
```

<details>
<summary>Disassembly</summary>

```
00008068 <sdmmc_spi_read_cxd>:
{
    8068:   b5f0        push    {r4, r5, r6, r7, lr}
    struct sdhc_data data = {0};
    806a:   2510        movs    r5, #16
{
    806c:   0004        movs    r4, r0
    806e:   b091        sub sp, #68 @ 0x44
    8070:   000f        movs    r7, r1
    8072:   0016        movs    r6, r2
    struct sdhc_command cmd = {0};
    8074:   2100        movs    r1, #0
    8076:   221c        movs    r2, #28
    8078:   a808        add r0, sp, #32
    807a:   f7ff fe7f   bl  7d7c <memset>
    struct sdhc_data data = {0};
    807e:   002a        movs    r2, r5
    8080:   2100        movs    r1, #0
    8082:   a801        add r0, sp, #4
    8084:   f7ff fe7a   bl  7d7c <memset>
    data.blocks = 1U;
    8088:   2201        movs    r2, #1
    cmd.response_type = SD_SPI_RSP_TYPE_R1;
    808a:   950d        str r5, [sp, #52]   @ 0x34
    data.block_size = 16;
    808c:   9502        str r5, [sp, #8]
    uint32_t *cxd_be = (uint32_t *)card->card_buffer;
    808e:   0025        movs    r5, r4
    cmd.timeout_ms = CONFIG_SD_CMD_TIMEOUT;
    8090:   23c8        movs    r3, #200    @ 0xc8
    ret = sdhc_request(card->sdhc, &cmd, &data);
    8092:   6820        ldr r0, [r4, #0]
    data.blocks = 1U;
    8094:   9203        str r2, [sp, #12]
    uint32_t *cxd_be = (uint32_t *)card->card_buffer;
    8096:   3571        adds    r5, #113    @ 0x71
    ret = sdhc_request(card->sdhc, &cmd, &data);
    8098:   aa01        add r2, sp, #4
    809a:   a907        add r1, sp, #28
    cmd.opcode = opcode;
    809c:   9707        str r7, [sp, #28]
    cmd.timeout_ms = CONFIG_SD_CMD_TIMEOUT;
    809e:   930f        str r3, [sp, #60]   @ 0x3c
    data.data = cxd_be;
    80a0:   9505        str r5, [sp, #20]
    data.timeout_ms = CONFIG_SD_CMD_TIMEOUT;
    80a2:   9306        str r3, [sp, #24]
    ret = sdhc_request(card->sdhc, &cmd, &data);
    80a4:   f7ff ffcc   bl  8040 <sdhc_request>
    for (i = 0; i < 4; i++) {
    80a8:   360c        adds    r6, #12
    80aa:   3481        adds    r4, #129    @ 0x81
        cxd[3 - i] = sys_be32_to_cpu(cxd_be[i]);
    80ac:   cd02        ldmia   r5!, {r1}
    for (i = 0; i < 4; i++) {
    80ae:   3e04        subs    r6, #4
    80b0:   ba09        rev r1, r1
        cxd[3 - i] = sys_be32_to_cpu(cxd_be[i]);
    80b2:   6071        str r1, [r6, #4]
    for (i = 0; i < 4; i++) {
    80b4:   42a5        cmp r5, r4
    80b6:   d1f9        bne.n   80ac <sdmmc_spi_read_cxd+0x44>
}
    80b8:   2000        movs    r0, #0
    80ba:   b011        add sp, #68 @ 0x44
    80bc:   bdf0        pop {r4, r5, r6, r7, pc}
```
</details>

### Success!

```
*** Booting Zephyr OS build zephyr-v3.4.0-3659-g718aa882bde2 ***
[00:00:00.042,000] <inf> sd: Maximum SD clock is under 25MHz, using clock of 1000000Hz
[00:00:00.045,000] <inf> main: Block count 15523840
Sector size 512
Memory Size(MB) 7580
Disk mounted.

Listing dir /SD: ...
[FILE] TESTING.TXT (size = 0)
[FILE] HELLO.TXT (size = 0)
```

attie-argentum, 2023-09-13:
Ping @danieldegrasse

danieldegrasse, 2023-09-14:
Hi @attie-argentum- thank you for narrowing this down. I think you are correct that the best way to fix this is to make the default `4`. We can also add a `BUILD_ASSERT` in `subsys/sd/sd.c` to insure the value is always greater than 4.

If you're willing to submit a patch to fix this, I'd appreciate it, otherwise I'm happy to.

attie-argentum, 2023-09-14:
No worries!

Can you expand on the reasoning for having the alignment configurable?

I wonder if it would be better to just align the buffer to the machine's word-size and be done with it (i.e: `sizeof(size_t)`, or something else if it's available in Zephyr).

I see `ARCH_IRQ_VECTOR_TABLE_ALIGN` and `ARCH_SW_ISR_TABLE_ALIGN` both use a default of `4`, and aside from RSICV there are no overrides visible... so perhaps that's the way to go...?

danieldegrasse, 2023-09-15:
> Can you expand on the reasoning for having the alignment configurable?

Of course- this is intended to enable DMA support in SD host controller drivers. You can see in the [`card_write_blocks` function](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/sd/sd_ops.c#L692) that `CONFIG_SDHC_BUFFER_ALIGNMENT` is used to make sure any data buffer passed to the SD host controller is aligned. This can be used to enforce additional alignment requirements if a given DMA controller needs them

commit a97b61e7 (2024-01-18): include: sd: revise the minimum alignment of card_buffer to 4-bytes

Since a85ffa8, `struct sd_card.card_buffer` has naturally fallen at an
offset that is unaligned. This meant that on systems without support for
unaligned access (e.g: Cortex-M0), a hard fault would present when
executing code that casts the buffer to `uint32_t` (such as
`sdmmc_spi_read_cxd()`, `card_query_written()`, etc...)

Historically, it appears that the alignment of the `card_buffer` member
was good and operational only by chance.

Altering the default value of `CONFIG_SDHC_BUFFER_ALIGNMENT` was
rejected, as this has wi [... cut]
closed 2024-01-18 by aescolar