zephyrproject-rtos/zephyr #62627: intel,adsp-mem-window driver is using EARLY init level

gmarull, 2023-09-14:
**Describe the bug**

`intel,adsp-mem-window` driver is using the EARLY init level, however, this level can't be used by device drivers because when it is called, the Kernel hasn't called `z_device_state_init`.

**To Reproduce**

https://github.com/zephyrproject-rtos/zephyr/blob/3fb434a6530897772e40922e780446b2f6ed4a24/soc/xtensa/intel_adsp/common/mem_window.c#L48-L71

**Expected behavior**

EARLY level is not used by device drivers

**Impact**

Blocks https://github.com/zephyrproject-rtos/zephyr/pull/62594, progress in init improvements.

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used 3fb434a6530897772e40922e780446b2f6ed4a24

**Additional context**

It looks like the code is not really a device driver, but a kind of initializer that seems to use the device infrastructure. There is nothing wrong with this, but, if this code is meant to be run early and devicetree does not capture the dependencies of such mem windows with other devices, it could be moved to a regular SYS_INIT.

nashif, 2023-09-14:
sorry, but this is not a bug,  the statement "EARLY level is not used by device drivers" is where you probably want to be, but it is not something established that this code violates. 

Actually, EARLY was introduced because of this and the dependencies at this very early stage of the boot process.

nashif, 2023-09-14:
> It is a bug because EARLY can't be safely used by devices: Kernel has not yet initialized some relevant structures. The fact that this has been allowed in devices is really an oversight.

what relevant structures are not initialized? How is this different from PRE_KERNEL, beside the fact it is done a bit earlier?

gmarull, 2023-09-14:
`z_device_state_init`. Also, expecting to have a working device infrastructure just after entering the C domain seems like a limitation for potential future changes. Moreover, `arch_kernel_init` has not been called at this point. I always understood EARLY as a SYS_INIT level to run something very early, not to be used by devices, precisely because you can't offer many guarantees at that point. Note that this is also a weird driver (under soc/...), I believe this can be solved in a better manner.

dcpleung, 2023-09-14:
The "memory windows" are not exactly a devices. Since they are defined in device tree, they simply piggy-back onto the device model to do init, with all the `DT_*` helpers and `struct device` provided. These can done via SoC hooks if we are going that route. Just need to know which one.

nashif, 2023-09-15:
> The "memory windows" are not exactly a devices. Since they are defined in device tree, they simply piggy-back onto the device model to do init, with all the `DT_*` helpers and `struct device` provided. These can done via SoC hooks if we are going that route. Just need to know which one.

we have a few things in drivers/ that are the same.

PR #62835 (2023-09-20): intel_adsp: move memory window setup to PRE_KERNEL_1
PRE_KERNEL_1 is more suited for dealing with devices, so move out of
EARLY.
Verified on hardware and things seem to behave the same. Something was
changed since this was first introduced as this was not possible for
some reason.

Fixes #62627
commit 759e07be (2023-09-20): intel_adsp: move memory window setup to PRE_KERNEL_1

PRE_KERNEL_1 is more suited for dealing with devices, so move out of
EARLY.
Verified on hardware and things seem to behave the same. Something was
changed since this was first introduced as this was not possible for
some reason.

Fixes #62627