zephyrproject-rtos/zephyr #62649: [RFC] Sharing built applications across multiple tests

gopiotr, 2023-09-14:
## Introduction

To work efficiently with multiple applications in one test it would be nice to have a mechanism for sharing once built application across several tests.

### Problem description

Last time I started work on porting BabbleSim tests from bash scripts into Twister’s yaml files. Some of those tests require building multiple applications per one test. I was convinced that all those cases can be handled by Sysbuild. But during work on this I face with situations when I need to reuse the same application image across several tests and so far, I don’t see efficient way for doing this with Twister or Sysbuild.

### Proposed change

Twister or Sysbuild (or both of them) would support sharing once built application across several tests. This could help to avoid rebuilding the same application several times across many tests, and finally save time.

## Detailed RFC

To illustrate my problem better I created following diagram with several build/test scenarios:

[image]

In general, there is no problem with `Scenario 1` or with `Scenario 2` – currently Twister and Sysbuild support those test configurations.
 
The problem occurs when I would like to reuse the once built application in several tests (like in `Scenario 3`). Building the same application twice is a waste of time and resources. 

In `Scenario 4` I propose an alternative solution – try to build all applications at once and after that try to run `test I` and `test II`. The disadvantage of this approach is that execution `test I` depends on building `Application C` which can fail and thus Twister will not run `test I`. Moreover, when user would like to run only `test I` he/she has to waste time for building `Application C` which is not necessary in this case.

Finally, `Scenario #5` shows how this problem could be solved by implementing an “sharing built applications” mechanism. In this case Twister or Sysbuild could verify if the needed application was already built in the previous test and then reuse them in next one. 

### Proposed change (Detailed)

I have no experience in Sysbuild (as a developer) so, here I can describe only my idea how to implement this mechanism in Twister. If someone else will have an idea how to achieve something similar in Sysbuild, then I'm open for propositions.

I propose to add new field `shared_app` in `testcase.yaml`/`sample.yaml` which will inform Twister that this particular test could reuse already built application from previous test. Let's assume that I would like to write a test which requires two applications: `sample.bluetooth.central` (from [samples/bluetooth/central/sample.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/central/sample.yaml) and `sample.bluetooth.peripheral_hr` (from [samples/bluetooth/peripheral_hr/sample.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/peripheral_hr/sample.yaml)). Then my `testcase.yaml` could looks like:

```yaml
bluetooth.central.peripheral_hr:
    shared_app:  # <--- if it was already built do not build it once again
      - sample.bluetooth.central
      - sample.bluetooth.peripheral_hr
    harness: pytest
    sysbuild: true
    # 1 - sample.bluetooth.central
    # 2 - sample.bluetooth.peripheral_hr
```

If the above test is run standalone, then Twister has to build all applications from scratch. But when the above test is run together with `sample.bluetooth.central` or `sample.bluetooth.peripheral_hr`, then there is possibility of reuse some of them in `bluetooth.central.peripheral_hr` test. In second situation there is a risk, that Twister starts build two the same applications in the same time (due to parallelism) -  this require some locking mechanism on Twister side. For me this is strong argument to have this functionality done on the Twister side - Twister as a test runner of plenty of tests can easier manage what should be build or what shouldn't in particular moment. I think that this could be hard to achieve only on the build system (Sysbuild) side.

### Dependencies

Even if the main functionality could be implemented in Twister, then I believe that some changes on the Sysbuild side have to be done to make it possible to "tell" Sysbuild to build for example only one application instead of two.

### Concerns and Unresolved Questions

I'm not sure where this mechanism should be implemented - in Twister or in Sysbuild (or both of them). I think that this could be quite discussable, because at the first glance this seems to be implemented on the build system side, but on the other hand in my opinion it would be more convenient to have it implemented on the Twister side - to avoid complication with building applications in parallel.

## Alternatives

At this moment any alternative which I see is ignore building the same application several times in row - maybe at this moment this is not a big issue (I need this sharing mechanism to rewrite BabbleSim tests), but I believe that in the future this could be a potential problem when number of such tests will increase.

gopiotr, 2023-09-14:
@tejlmand @57300 @nashif @carlescufi @golowanow @PerMac @gchwier - I would be grateful for your opinion about this

PerMac, 2023-09-14:
I think there is an alternative without sysbuild in the picture, i.e. enhancing twisters scheduler.
Twister prepares a list of test instances to proceed. Then multiple workers are spawned and fed with instances from the list. 
"Upgrade": If an instance depending on shared images is detected, scheduler positions it after the shared images and holds its proceeding until all dependencies are available. If a user points twister only to a test with dependencies, scheduler will detect that those dependencies are not in the scope and add them to the list to build in advance. 
I believe with the above approach nothing would have to be changed on the build system side, just adding some logic to the scheduler.

aescolar, 2023-09-25:
Thanks for the proposal @gopiotr 
(Note that I just saw this now, and @nordicjm may not have seen it at all, as editing comments to add users does not notify those users)
I agree with @PerMac that a version that relies on twister and does not involve sysbuild for this should be possible, and in my eyes preferable:
In general we have a situation where 1 "test" requires N devices images, and M tests may share some subsets of these images. And where L tests may all share the same set of devices images.

One option in line of what is proposed here is that some images that are shared between tests could be described in twister as a new category beyond _test_ and _sample_ (say _image_*1).
We could then have a new key for tests similar to what you mentioned like `requires_image: <list of images>`

In line with this, we'd need to have a new key that signifies a test does not contain in itself an image. Say `no_own_image:true` (As some tests will share all their images with other tests).

Conceptually the "old" tests and samples we have now describe both  one _test_ and one _image_ in one go.

I think the keys for filtering (depends, allow, exclude, levels, integration_platform, filter, slow, skip, tags), harness, and build_only could be left at the _test_/_sample_ level;
And the image would keep the extra_args, extra_configs, extra_sections & sysbuild keys.

As @PerMac proposes, twister would need to get smarter in its scheduling, decoupling the set of build and runs jobs, and know which run jobs depends on which build jobs (as now a build job gets a N->1 dependency)

I would involve sysbuild only for building images for a single device, meaning either multi-core/AMP or images with bootloaders or the like.

*1 these _image_ entries would not produce a build or test unless used by a selected test. Also these _image_ entries would need to be searched in the zephyr tree (not just under what `-T` may select)

aescolar, 2023-09-25:
For context, today the bsim tests build 85 different images, and run 726 separate tests (not counting the EDTT subtests). Where each of these tests uses in general several of those images.
So just building for each test all images that test needs irrespectively of what had just been built for other tests in the same run would not be a reasonable option.

aescolar, 2023-10-10:
An extra remark: Today twister can generate a set of tests lists to split the work for CI minions. Ideally these would be generated in a way in which separate sub lists do not share images (i.e. tests that share images are grouped together). Though one could consider this an optimization.

nashif, 2023-10-11:
> @nashif what is your take on the topic? Maybe you'd like to discuss it during some meeting?

needs some thinking, I am not sure tinkering with the scheduler and queue management is the right approach though.

Without having to look into bsim, can someone provide a way using some simple and general test to show the usecase, i.e. if you had to create a test for this "feature", how would it look like?

PerMac, 2023-10-11:
I think a non bsim example is mentioned in the "Proposed change (Detailed)" section, but let's have here a short example:
We have a single sample for bluetooth cetral. We also have several samples for bluetooth peripherals. We want to test if those peripheral samples actually works with the central. Also the tests for each peripheral needs to be an individual entity. E.g. failing heart rate test, shouldn't influence health thermometer test. We don't want to build the same central, for each test. Just build central once and be able to use the image in tests for each peripheral.

nashif, 2023-10-17:
@PerMac  my point is that

> ```yaml
> bluetooth.central_and_peripheral:
>     shared_app:  # <--- if it was already built do not build it once again
>       - sample.bluetooth.central
>       - sample.bluetooth.peripheral_hr
>     harness: pytest
>     sysbuild: true
>     # 1 - sample.bluetooth.central
>     # 2 - sample.bluetooth.peripheral_hr
> ```

makes a lot of assumptions, i.e. that you have started twister on the top level of the tree and that shared_app items might not be available in the test scope. What happens if you point twister to `-T tests/`? Having a dependency on some sample in the tree is very weak here. So we need to define this at the test level somehow and ideally make it reproducible outside of twister if possible as well.

aescolar, 2023-10-17:
Thanks @nashif 

> makes a lot of assumptions, i.e. that you have started twister on the top level of the tree and that shared_app items might not be available in the test scope.

I think this one is solved with my proposal, or having `image`s you can refer to (which are not tests in themselves). After the testplan has been built based on the tests, when you chose a try to build a test, you see what images you need, and if not yet built for that platform, you build them (after building the test defined image (if any) and applying the `filter`).

> What happens if you point twister to -T tests/?

Required images would need to be searched outside of -T, we could say that by default that is zephyr base, but we can also add an extra command line option to say where that search is based.

PerMac, 2023-10-17:
Thanks for the clarification. We are thinking about alternatives or some more refinement in the process. However, I already have a question regarding
>"So we need to define this at the test level somehow and ideally make it reproducible outside of twister if possible as well."

I assume you mean preparation of all required images, without twister, correct? Being able to execute the test without twister, doesn't seem reasonable (I guess it would be possible to reproduce it with pytest without it being wrapped in twister, but I believe it still counts as twister).