zephyrproject-rtos/zephyr #62676: tests: drivers: flash: common: drivers.flash.common.spi_nor and drivers.flash.common.spi_nor_wp_hold fail

katgiadla, 2023-09-15:
**Describe the bug**
The tests `drivers.flash.common.spi_nor` and `drivers.flash.common.spi_nor_wp_hold` fail.

Observed for `nrf52840dk_nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/flash/common/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
W: Waiting until flash is ready
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.0
- Commit SHA or Version used: zephyr-v3.4.0-3678-gdb52d27eb568

fabiobaltieri, 2023-09-19:
@katgiadla is this a regression? setting as low for now @de-nordic feel free to change that

katgiadla, 2023-09-20:
The issues still occur but currently, tests has sent this:

```
E: Device id 00 00 00 does not match config c2 23 15
*** Booting Zephyr OS build zephyr-v3.4.0-3926-g4ba97c22555e ***
Running TESTSUITE flash_driver
===================================================================
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/flash/common/src/main.c:59: flash_driver_setup: device_is_ready(flash_dev) is false
START - test_read_unaligned_address
FAIL - test_read_unaligned_address in 0.000 seconds
===================================================================
TESTSUITE flash_driver failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.000 seconds
- FAIL - [flash_driver.test_read_unaligned_address] duration = 0.000 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 245ca324b97180fe5114662dbcf157ed
PROJECT EXECUTION FAILED
```

fabiobaltieri, 2023-09-20:
> E: Device id 00 00 00 does not match config c2 23 15

Oh I see this now and the error is that the device is not ready, the two are probably tied. The flash has not been recognized, so either something is wrong with the initialization or something is wrong with the board.

Could you:
- check if this works in v3.4.0 or some other stable version, if it does do a bisect to figure what broke it
- verify on a different board if you have one available?

@de-nordic/@anangl maybe you can run this on a board and see if you can reproduce this on your side?

de-nordic, 2023-09-28:
Give in moment I am overwhelmed at this point.

de-nordic, 2023-10-09:
Looks like this is the same issue as here: https://github.com/zephyrproject-rtos/zephyr/issues/57324#issuecomment-1736209879

anangl, 2023-11-21:
The problem here is that these two tests cannot be performed on the nRF52840 DK without additional components. According to related overlays, they require an external flash chip (other than the onboard one) that is connected to specific pins.

PR #65536 (2023-11-23): tests: drivers: flash: Fix a few test scenarios on nRF52840 DK
- tests: drivers: flash: Update nrf52840dk_mx25r_high_perf.overlay
- tests: drivers: flash: Use fixtures for tests requiring external chips

See the commit messages for details.

Fixes #62676.
closed 2023-11-23 by carlescufi