zephyrproject-rtos/zephyr #62687: Central / Broadcast Assistant fails to discover BASS on the receiver after a disconnect / connect, if the receiver had been syncing to an audio broadcast stream

pmullt, 2023-09-15:
**Describe the bug**

Central / Broadcast Assistant fails to discover BASS on the receiver after a disconnect / connect, if the receiver had been syncing to an audio broadcast stream.

If a Broadcast Assistant (BA) application has an ACL connection to a receiver, and the receiver starts to synchronize to an audio broadcast stream (with or without PAST), the connection between the Broadcast Assistant and receiver drops regularly. This is a known issue for some time. 

However, if the connection drops, the BA application can detect this and re-connect successfully . The problem here is that then the BASS discover fails on the BA, until the receiver gets reset. Here are logs from the BA application when this happens : 

```
Disconnected from device EE:0B:7C:30:59:12 (random) (reason 0x3e)
About to connect to EE:0B:7C:30:59:12
Connection pending..
Successfully connected to device EE:0B:7C:30:59:12 (random)
Now discovering..
[00:01:04.517,425] <wrn> bt_gatt: gatt_exchange_mtu_func: conn 0x200018c8 err 0x0e
[00:01:04.517,456] <dbg> bt_bap_broadcast_assistant: service_discover_func: Could not discover BASS
Error! BASS discover failed (-6)
```
**NOTE : This only happens when using the Nordic soft-device controller or the Packetcraft controller. With the Zephyr controller it's not an issue and the BASS re-discovery works fine**  

It seems that there is a BT_ATT_ERR_AUTHORIZATION. 
In the _proj.conf_ in both the receiver application and BA application, even bonding is disabled :  `CONFIG_BT_BONDABLE=n`

The targets used are the nRF5340 audio development kits.

**To Reproduce**

1. Unzip the attached broadcast_assistant.zip into zephyr\samples\bluetooth
2. cd into this broadcast_assistant sample folder
3. README.rst explains how to build and flash the Broadcast Assistant application 
4. Flash the Packetcraft controller to the network core 
5. Connect the Broadcast Assistant application to a receiver and add sources to the Scan Delegator on the receiver (information in the README.rst)
6. Keep trying until the connection drops and this BASS discover fails. 

**Expected behavior**
Connection between the BA and receiver drops, and BASS discover fails on the BA with : `Error! BASS discover failed (-6)`

**Impact**

This is a showstopper.

One solution is to just use the Zephyr controller for now, but unable to use PAST with that. 

**Logs and console output**

```
Disconnected from device EE:0B:7C:30:59:12 (random) (reason 0x3e)
About to connect to EE:0B:7C:30:59:12
Connection pending..
Successfully connected to device EE:0B:7C:30:59:12 (random)
Now discovering..
[00:01:04.517,425] <wrn> bt_gatt: gatt_exchange_mtu_func: conn 0x200018c8 err 0x0e
[00:01:04.517,456] <dbg> bt_bap_broadcast_assistant: service_discover_func: Could not discover BASS
Error! BASS discover failed (-6)
```
**Environment :**
 OS : Windows 11
SDK: nRF Connect SDK v2.4.1

**Additional context**

[broadcast_assistant.zip](https://github.com/zephyrproject-rtos/zephyr/files/12617824/broadcast_assistant.zip)

pmullt, 2023-09-15:
@Thalley

Thalley, 2023-09-15:
Thanks @pmullt - I'll have a look at it in the near future :)

Thalley, 2023-10-24:
Closing as I cannot reproduce, and the issue seems to be related to something else than Zephyr.

closed 2023-10-24 by Thalley