zephyrproject-rtos/zephyr #62704: ARC: NSIM: tests/kernel/context/kernel.context fails.

gbarkadiusz, 2023-09-15:
**To Reproduce**
`twister -p nsim_em -s tests/kernel/context/kernel.context -v`

**Expected behavior**
Test should pass.

**Impact**
Affects our CI.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-3[869](https://github.com/intel-innersource/os.rtos.zephyr.zephyr/actions/runs/6197499291/job/16826525671#step:7:870)-ga9d75fc56ec5 ***
Running TESTSUITE context
===================================================================
START - test_ctx_thread
Testing k_current_get() from an ISR and thread

    Assertion failed at WEST_TOPDIR/os.rtos.zephyr.zephyr/tests/kernel/context/src/main.c:620: context_test_ctx_thread: (isr_info.data not equal to (void *)self_thread_id)
ISR context ID mismatch
 FAIL - test_ctx_thread in 0.005 seconds
===================================================================
START - test_interrupts
 SKIP - test_interrupts in 0.001 seconds
===================================================================
TESTSUITE context failed.
Running TESTSUITE context_cpu_idle
===================================================================
START - test_cpu_idle
 PASS - test_cpu_idle in 0.147 seconds
===================================================================
START - test_cpu_idle_atomic
 PASS - test_cpu_idle_atomic in 0.148 seconds
===================================================================
TESTSUITE context_cpu_idle succeeded
Running TESTSUITE context_one_cpu
===================================================================
START - test_busy_wait
 PASS - test_busy_wait in 0.041 seconds
===================================================================
START - test_k_sleep
Testing k_thread_create() without cancellation
 thread (q order: 2, t/o: 500) is running
 got thread (q order: 2, t/o: 500) as expected
 thread (q order: 3, t/o: 750) is running
 got thread (q order: 3, t/o: 750) as expected
 thread (q order: 0, t/o: 1000) is running
 got thread (q order: 0, t/o: 1000) as expected
 thread (q order: 6, t/o: 1250) is running
 got thread (q order: 6, t/o: 1250) as expected
 thread (q order: 1, t/o: 1500) is running
 got thread (q order: 1, t/o: 1500) as expected
 thread (q order: 4, t/o: 1750) is running
 got thread (q order: 4, t/o: 1750) as expected
 thread (q order: 5, t/o: 2000) is running
 got thread (q order: 5, t/o: 2000) as expected
Testing k_thread_create() with cancellations
 cancelling [q order: 0, t/o: 1000, t/o order: 0]
 thread (q order: 3, t/o: 750) is running
 got (q order: 3, t/o: 750, t/o order 1) as expected
 thread (q order: 0, t/o: 1000) is running
 got (q order: 0, t/o: 1000, t/o order 2) as expected
 cancelling [q order: 3, t/o: 750, t/o order: 3]
 cancelling [q order: 4, t/o: 1750, t/o order: 4]
 thread (q order: 4, t/o: 1750) is running
 got (q order: 4, t/o: 1750, t/o order 5) as expected
 cancelling [q order: 6, t/o: 1250, t/o order: 6]
 PASS - test_k_sleep in 5.343 seconds
===================================================================
START - test_k_yield
 PASS - test_k_yield in 0.002 seconds
===================================================================
START - test_thread

    Assertion failed at WEST_TOPDIR/os.rtos.zephyr.zephyr/tests/kernel/context/src/main.c:665: _test_kernel_thread: ((isr_info.error || (isr_info.data != (void *)self_thread_id)) is true)
Thread ID taken during ISR != calling thread
 FAIL - test_thread in 0.005 seconds
===================================================================
START - test_timer_interrupts
 PASS - test_timer_interrupts in 0.349 seconds
===================================================================
TESTSUITE context_one_cpu failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [context]: pass = 0, fail = 1, skip = 1, total = 2 duration = 0.006 seconds
 - FAIL - [context.test_ctx_thread] duration = 0.005 seconds
 - SKIP - [context.test_interrupts] duration = 0.001 seconds

SUITE PASS - 100.00% [context_cpu_idle]: pass = 2, fail = 0, skip = 0, total = 2 duration = 0.295 seconds
 - PASS - [context_cpu_idle.test_cpu_idle] duration = 0.147 seconds
 - PASS - [context_cpu_idle.test_cpu_idle_atomic] duration = 0.148 seconds

SUITE FAIL -  80.00% [context_one_cpu]: pass = 4, fail = 1, skip = 0, total = 5 duration = 5.740 seconds
 - PASS - [context_one_cpu.test_busy_wait] duration = 0.041 seconds
 - PASS - [context_one_cpu.test_k_sleep] duration = 5.343 seconds
 - PASS - [context_one_cpu.test_k_yield] duration = 0.002 seconds
 - FAIL - [context_one_cpu.test_thread] duration = 0.005 seconds
 - PASS - [context_one_cpu.test_timer_interrupts] duration = 0.349 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 2a64c439d8aa2b2b4fb175545729e24d
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux.
 - Toolchain Zephyr SDK v16.1
 - Zephyr version: zephyr-v3.4.0-3867-g772eaf98eaed

nashif, 2023-10-25:
Can be bisected to 

```
commit f0daf904bb0202c9247cdcf4d1186e0762a9db67 (HEAD)
Author: Keith Packard <keithp@keithp.com>
Date:   Tue Jan 10 21:17:32 2023 -0800

    libc: Default to picolibc where supported

    This switches the default C library to picolibc for all targets which
    support it.

    Signed-off-by: Keith Packard <keithp@keithp.com>
```

@keith-packard  FYI

keith-packard, 2023-10-25:
I wonder if this is TLS related; would be useful for someone with access to nSIM to try with TLS disabled by applying #62319 and then building with TLS disabled using the picolibc module.

ruuddw, 2023-10-25:
Thanks for the suggestions and looking into this. There have been some other priorities, but with the suggestions and joint effort we should be able to find the root cause. Will see what we can do.

nashif, 2023-10-25:
> I wonder if this is TLS related; would be useful for someone with access to nSIM to try with TLS disabled by applying #62319 and then building with TLS disabled using the picolibc module.

this passes, so it is TLS related...

ruuddw, 2023-10-26:
Not sure about the stack size or TLS stuff. Could it be a similar root cause as #39948? i.e. different scheduler/timer behavior with ISS?

keith-packard, 2023-10-26:
yeah, that's my concern -- TLS changes a bunch of stuff in task switching and requires careful register management to make sure the right TLS base pointer is in place everywhere. You should rule out stack size issues by increasing the relevant stack allocations first though; stack overflows in consecutive stack areas can actually smash the TLS variables in adjacent allocations and cause all kinds of weird problems. These are often asymptomatic in the non-TLS case as the first few things pushed on the thread's stack are often never used again.

ruuddw, 2023-10-27:
ok, digged into it a bit more. The issue is indeed that the TLS base pointer (gp) has the wrong value at the moment of the `if (k_current_get() != &alt_thread)` check. k_current_get() gets a cached thread pointer from TLS.
Reason for the incorrect value of gp is an ARC hardware feature for low-latency interrupts: double register bank. As this check is executed still in "fast IRQ" ISR context (the ISR handler triggers the detection of the stack smashing, in firq_exit when calling get_next_switch_handle; then continuing with an exception/exception context), it's using the second register bank which doesn't have the valid gp.
Not sure yet what's the best fix for this corner case: updating the test, no TLS caching for the thread pointer, copying gp to 2nd register bank, etc.
I guess relying on TLS in ISR/exception context is not correct, so either the generic `k_current_get()` may need fixing, or the test?

ruuddw, 2023-10-27:
Replacing `k_current_get()` with `k_sched_current_thread_query()` does the trick. Note the remark in the API documentation:

>  "Use [k_current_get()](https://docs.zephyrproject.org/latest/kernel/services/threads/index.html#group__thread__apis_1ga7ef1ed0fb9513df8096ede1e52fc76b2) unless absolutely sure this is necessary. This should only be used directly where the thread local variable cannot be used or may contain invalid values if thread local storage (TLS) is enabled. If TLS is not enabled, this is the same as [k_current_get()](https://docs.zephyrproject.org/latest/kernel/services/threads/index.html#group__thread__apis_1ga7ef1ed0fb9513df8096ede1e52fc76b2)."

Maybe grep and check if there are more places where k_current_get() is used in interrupt/exception context?

keith-packard, 2023-10-27:
Awesome debugging. I'd encourage you to copy the regular thread base pointer to the ISR register set -- that will avoid potential regressions in the future when changes are made to core kernel code.

kokas-a, 2024-01-30:
The issue related changes applied here: https://github.com/zephyrproject-rtos/zephyr/pull/67831

closed 2024-01-30 by evgeniy-paltsev