zephyrproject-rtos/zephyr #62705: ARC: NSIM: tests/kernel/fatal/exception/kernel.common.stack_sentinel fails.

gbarkadiusz, 2023-09-15:
**To Reproduce**
`twister -p nsim_em -s tests/kernel/fatal/exception/kernel.common.stack_sentinel -v`

**Expected behavior**
Test should pass.

**Impact**
Affect our CI.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-3867-g772eaf98eaed ***
Running TESTSUITE fatal_exception
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0xc4
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0xbc  pc: 0xc4
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x0 lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 0
test alt thread 1: generic CPU exception divide zero
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0x1b0
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0x1a8  pc: 0x1b0
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
E:  r0: 0x3  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0xd8  pc: 0xe6
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 3
test alt thread 3: initiate kernel panic
E:  r0: 0x4  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0xe8  pc: 0xf6
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 4
test alt thread 4: fail assertion
ASSERTION FAIL [0] @ WEST_TOPDIR/os.rtos.zephyr.zephyr/tests/kernel/fatal/exception/src/main.c:155
	intentionally failed assertion
E:  r0: 0x4  r1: 0x9b  r2: 0x80000058  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0xffffffff r13: 0x50d4  pc: 0x9ce
E:  blink: 0x150 status32: 0x80082806
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 4
test alt thread 5: initiate arbitrary SW exception
E:  r0: 0x7fffffff  r1: 0x0  r2: 0x7fffffff  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0xf8  pc: 0x10e
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 2147483647: Unknown error on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason 2147483647
test alt thread 6: initiate arbitrary SW exception negative
E:  r0: 0xfffffffe  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x80000290 r13: 0x110  pc: 0x11e
E:  blink: 0x736 status32: 0x80082006
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR -2: Unknown error on CPU 0
E: Current thread: 0x80000058 (unknown)
Caught system error -- reason -2
test stack sentinel overflow - timer irq
posting 1024 bytes of junk to stack...
E:  r0: 0x2  r1: 0xf0f0f0f0  r2: 0xbbbbbbbb  r3: 0x14b8
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x0
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x800002c4 r13: 0x0  pc: 0x2bdc
E:  blink: 0x37e8 status32: 0x80091606
E: lp_end: 0x0 lp_start: 0x2 lp_count: 0x1
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0xef8 (@1)
Caught system error -- reason 2
Wrong thread crashed
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK v16.1
 - Zephyr version: zephyr-v3.4.0-3867-g772eaf98eaed

fabiobaltieri, 2023-09-19:
> E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0

Maybe it's a configuration issue and needs a bigger stack?

nashif, 2023-10-25:
ok, bisected to 

```
commit f0daf904bb0202c9247cdcf4d1186e0762a9db67
Author: Keith Packard <keithp@keithp.com>
Date:   Tue Jan 10 21:17:32 2023 -0800

    libc: Default to picolibc where supported

    This switches the default C library to picolibc for all targets which
    support it.

    Signed-off-by: Keith Packard <keithp@keithp.com>

```

@keith-packard FYI

nashif, 2023-10-25:
@evgeniy-paltsev  @ruuddw  Can you please take a look! No activity for over a month.

keith-packard, 2023-10-25:
Could try #62319 and then disable TLS; that saves a bit of stack space and makes the memory configuration much more similar to minimal libc.

keith-packard, 2023-10-25:
> Yes, this seems to fix the issue, but then I have to build picolibc as a module,

So perhaps the test just needs a bit more stack space (for any TLS variables), or perhaps the test isn't configuring the memory regions correctly?

kokas-a, 2024-01-30:
The issue related changes applied here: https://github.com/zephyrproject-rtos/zephyr/pull/67831

closed 2024-01-30 by evgeniy-paltsev