zephyrproject-rtos/zephyr #62714: upsquared: tests/kernel/threads/dynamic_thread_stack timeout

KamilxPaszkiet, 2023-09-15:
**Describe the bug**
Out of 10 tests in the test suite, 2 fail.
`tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user`
and
`tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.no_alloc.user`
timeout.

**To Reproduce**
```twister --device-testing --hardware-map up2.map -T tests/kernel/threads/dynamic_thread_stack/ -vv```

**Logs and console output**
```
2023-09-15 15:07:06,405 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-15 15:07:06,408 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-15 15:07:06,410 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-15 15:07:06,793 - twister - DEBUG - DEVICE:
2023-09-15 15:07:06,795 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-15 15:07:06,863 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-15 15:07:06,865 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-15 15:07:06,869 - twister - DEBUG - DEVICE: Zeroing 4867200 bytes of memory at 0x10e000
2023-09-15 15:07:06,917 - twister - DEBUG - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
2023-09-15 15:07:06,923 - twister - DEBUG - DEVICE: Copying 57344 data bytes to 0x100000 from image offset 32768
2023-09-15 15:07:06,930 - twister - DEBUG - DEVICE: Copying 2268032 data bytes to 0x5b2480 from image offset 90112
2023-09-15 15:07:06,956 - twister - DEBUG - DEVICE: Jumping to Entry Point: 0x1153 (48 31 c0 48 31 d2 48)
2023-09-15 15:07:07,193 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v3.4.0-3755-gd42d14e392e7 ***
2023-09-15 15:07:07,207 - twister - DEBUG - DEVICE: Running TESTSUITE dynamic_thread_stack
2023-09-15 15:07:07,214 - twister - DEBUG - DEVICE: ===================================================================
2023-09-15 15:07:07,218 - twister - DEBUG - DEVICE: START - test_dynamic_thread_stack_alloc
o023-09-15 15:07:07,222 - twister - DEBUG - DEVICE: Hello,H edlylnoa,m idcy nwaomrilcd !w

2023-09-15 15:07:29,484 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-15 15:07:29,486 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-15 15:07:29,489 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-15 15:07:29,868 - twister - DEBUG - DEVICE:
2023-09-15 15:07:29,870 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-15 15:07:29,937 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-15 15:07:29,940 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-15 15:07:29,944 - twister - DEBUG - DEVICE: Zeroing 4873056 bytes of memory at 0x10e000
2023-09-15 15:07:29,992 - twister - DEBUG - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
2023-09-15 15:07:29,998 - twister - DEBUG - DEVICE: Copying 57344 data bytes to 0x100000 from image offset 32768

2023-09-15 15:08:55,663 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-15 15:08:55,665 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-15 15:08:55,668 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-15 15:08:56,038 - twister - DEBUG - DEVICE:
2023-09-15 15:08:56,040 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-15 15:08:56,108 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-15 15:08:56,110 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-15 15:08:56,115 - twister - DEBUG - DEVICE: Zeroing 4867200 bytes of memory at 0x10e000

2023-09-15 15:09:18,880 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-15 15:09:18,882 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-15 15:09:18,885 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-15 15:09:19,545 - twister - DEBUG - DEVICE:
2023-09-15 15:09:19,548 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-15 15:09:19,617 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-15 15:09:19,619 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-15 15:09:19,624 - twister - DEBUG - DEVICE: Zeroing 11198464 bytes of memory at 0x120000
2023-09-15 15:09:19,726 - twister - DEBUG - DEVICE: Zeroing 3008 bytes of memory at 0xdf7440
2023-09-15 15:09:19,731 - twister - DEBUG - DEVICE: Copying 32768 data bytes to 0x1000 from image offset

2023-09-15 15:09:42,641 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-15 15:09:42,644 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-15 15:09:42,646 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-15 15:09:43,029 - twister - DEBUG - DEVICE:
2023-09-15 15:09:43,032 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-15 15:09:43,101 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-15 15:09:43,103 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-15 15:09:43,107 - twister - DEBUG - DEVICE: Zeroing 11198464 bytes of memory at 0x120000
2023-09-15 15:09:43,209 - twister - DEBUG - DEVICE: Zeroing 3008 bytes of memory at 0xdf7440
[... 42 lines cut ...]
2023-09-15 15:11:10,589 - twister - DEBUG - DEVICE: ===================================================================
2023-09-15 15:11:10,593 - twister - DEBUG - DEVICE: RunID: 01aca5074b481867bdb0ac0326c1db3a
2023-09-15 15:11:10,596 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
2023-09-15 15:11:10,699 - twister - DEBUG - Expected suite names:['dynamic_thread_stack', 'dynamic_thread_stack']
2023-09-15 15:11:10,699 - twister - DEBUG - Detected suite names:['dynamic_thread_stack']
2023-09-15 15:11:10,700 - twister - DEBUG - run status: up_squared/tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.no_user passed
2023-09-15 15:11:10,703 - twister - INFO - 8/8 up_squared                tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.no_user PASSED (device: up2, 23.043s)
2023-09-15 15:11:10,725 - twister - INFO - 8 test scenarios (8 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
2023-09-15 15:11:10,727 - twister - INFO - 6 of 8 test configurations passed (75.00%), 2 failed, 0 errored, 0 skipped with 0 warnings in 325.64 seconds
2023-09-15 15:11:10,727 - twister - INFO - In total 10 test cases were executed, 6 skipped on 1 out of total 615 platforms (0.16%)
2023-09-15 15:11:10,727 - twister - INFO - 8 test configurations executed on platforms, 0 test configurations were only built.
2023-09-15 15:11:10,728 - twister - INFO - Saving reports...
2023-09-15 15:11:10,728 - twister - INFO - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/twister.json
2023-09-15 15:11:10,736 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
2023-09-15 15:11:10,741 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
2023-09-15 15:11:10,741 - twister - INFO - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
2023-09-15 15:11:10,741 - twister - INFO - The following issues were found (showing the top 10 items):
2023-09-15 15:11:10,741 - twister - INFO - 1) tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.no_alloc.user on up_squared failed (Timeout)
2023-09-15 15:11:10,741 - twister - INFO - 2) tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user on up_squared failed (Timeout)
```

**Environment:**
- OS: Linux
- Toolchain: zephyr-sdk-0.16.0
- zephyr-v3.4.0-3755-gd42d14e392e7

nashif, 2023-10-24:
@KamilxPaszkiet  that log you provide does not show any failures, it shows tests passing, please provide accurate logs of the test that is failing.

KamilxPaszkiet, 2023-10-26:
This [PR](https://github.com/zephyrproject-rtos/zephyr/pull/64362) fix this issue

jhedberg, 2023-11-22:
@KamilxPaszkiet I don't see any errors when running this test app multiple times with the latest tree (just building the full app and booting it through USB on my up_squared board):
```
------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [dynamic_thread_stack]: pass = 1, fail = 0, skip = 1, total = 2 duration = 0.005 seconds
 - SKIP - [dynamic_thread_stack.test_dynamic_thread_stack_alloc] duration = 0.001 seconds
 - PASS - [dynamic_thread_stack.test_dynamic_thread_stack_pool] duration = 0.004 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```
I don't have a way to run this with twister however. Could this be an issue with your test harness? Upstream doesn't seem have any documentation or claimed support for any `twister --device-testing` features with this board, so if it's an issue with your setup then an upstream GitHub issue is probably not the right place for this report.

KamilxPaszkiet, 2023-11-22:
> @KamilxPaszkiet I don't see any errors when running this test app multiple times with the latest tree (just building the full app and booting it through USB on my up_squared board):
>
> ```
> ------ [...]

You checked the wrong tests, the problem occurs in:
`tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.no_alloc.user`
and
`tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user`

jhedberg, 2023-11-22:
@KamilxPaszkiet thanks, I can see the issue now

jhedberg, 2023-11-22:
@KamilxPaszkiet and now the issue is gone. Maybe merging #65428 fixed this one as well for `up_squared`. Could you try with latest main branch?

KamilxPaszkiet, 2023-11-23:
> @KamilxPaszkiet and now the issue is gone. Maybe merging #65428 fixed this one as well for `up_squared`. Could you try with latest main branch?

You're right, the problem has been fixed. I'm closing the issue.

closed 2023-11-23 by KamilxPaszkiet