zephyrproject-rtos/zephyr #62717: upsquared: tests/subsys/logging/log_immediate/logging.log_immediate

KamilxPaszkiet, 2023-09-15:
**Describe the bug**
Out of 2 tests in the test suite, 2 fail.
`tests/subsys/logging/log_immediate/logging.log_immediate`
and
`tests/subsys/logging/log_immediate/logging.log_immediate.clean_output`
fails.

**To Reproduce**
```twister --device-testing --hardware-map up2.map -T tests/subsys/logging/log_immediate/ -vv```

**Logs and console output**
```
DEBUG   - DEVICE: >>Checking Media Presence......
DEBUG   - DEVICE: >>Media Present......
DEBUG   - DEVICE: Downloading NBP file...
DEBUG   - DEVICE: ASSERTION FAIL [device_is_ready(uart_dev)] @ WEST_TOPDIR/zephyr/subsys/logging/backends/log_backend_uart.c:127
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Succeed to download NBP file.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x79c2e000
DEBUG   - DEVICE: Zeroing 4972568 bytes of memory at 0x110000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 65536 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2265056 data bytes to 0x5ce020 from image offset 98304
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: ASSERTION FAIL [device_is_ready(uart_dev)] @ WEST_TOPDIR/zephyr/subsys/logging/backends/log_backend_uart.c:127
DEBUG   - Timed out while monitoring serial output on up_squared
DEBUG   - Process ssh -o StrictHostKeyChecking=no -tt zephyr@10.123.239.214 stty raw 115200 -F /dev/ttyUP2 && cat /dev/ttyUP2_2 terminated outs: None errs None
DEBUG   - Expected suite names:['log_immediate', 'log_immediate']
DEBUG   - Detected suite names:[]
DEBUG   - run status: up_squared/tests/subsys/logging/log_immediate/logging.log_immediate failed
INFO    - 1/2 up_squared                tests/subsys/logging/log_immediate/logging.log_immediate  FAILED Timeout (device: up2_2, 90.201s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/up_squared/tests/subsys/logging/log_immediate/logging.log_immediate/handler.log
DEBUG   - Timed out while monitoring serial output on up_squared
DEBUG   - Process ssh -o StrictHostKeyChecking=no -tt zephyr@10.123.239.214 stty raw 115200 -F /dev/ttyUP2 && cat /dev/ttyUP2 terminated outs: None errs None
DEBUG   - Expected suite names:['log_immediate', 'log_immediate']
DEBUG   - Detected suite names:[]
DEBUG   - run status: up_squared/tests/subsys/logging/log_immediate/logging.log_immediate.clean_output failed
INFO    - 2/2 up_squared                tests/subsys/logging/log_immediate/logging.log_immediate.clean_output  FAILED Timeout (device: up2, 90.365s)
ERROR   - see: /home/zephyr/zephyrproject/zephyr/twister-out/up_squared/tests/subsys/logging/log_immediate/logging.log_immediate.clean_output/handler.log

INFO    - 2 test scenarios (2 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
--------------------------------
Total test suites: 2
Total test cases: 2
Executed test cases: 2
Skipped test cases: 0
Completed test suites: 2
Passing test suites: 0
Failing test suites: 2
Skipped test suites: 0
Skipped test suites (runtime): 0
Skipped test suites (filter): 0
Errors: 0
--------------------------------
INFO    - 0 of 2 test configurations passed (0.00%), 2 failed, 0 errored, 0 skipped with 0 warnings in 103.30 seconds
INFO    - In total 2 test cases were executed, 0 skipped on 1 out of total 617 platforms (0.16%)
INFO    - 2 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board      | ID    |   Counter |
|------------|-------|-----------|
| up_squared | up2   |         1 |
| up_squared | up2_2 |         1 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/zephyr/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/zephyr/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/zephyr/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/logging/log_immediate/logging.log_immediate on up_squared failed (Timeout)
INFO    - 2) tests/subsys/logging/log_immediate/logging.log_immediate.clean_output on up_squared failed (Timeout)
```

**Environment:**
- OS: Linux
- Toolchain: zephyr-sdk-0.16.0
- zephyr-v3.4.0-3801-g4856fd4cb6ce

KamilxPaszkiet, 2023-10-26:
This [PR](https://github.com/zephyrproject-rtos/zephyr/pull/64362) fix this issue

jhedberg, 2023-11-22:
I can reproduce this:
```
Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
ASSERTION FAIL [device_is_ready(uart_dev)] @ ZEPHYR_BASE/subsys/logging/backends/log_backend_uart.c:154
```
Seems like this is indeed a UART init priority issue.

jhedberg, 2023-11-22:
The issue can be easily seen when looking at the initlevels output:
```
❯ west build -t initlevels
-- west build: running target initlevels
[0/1] cd /Users/jhedberg/src/zephyr/zephyr/build/zephyr && /usr/local/opt/python@3.11/bin/python3.11 /Users/jhedberg/src/zephyr/zephyr/scripts/build/check_init_priorities.py --elf-file=/Users/jhedberg/src/zephyr/zephyr/build/zephyr/zephyr.elf --initlevels
EARLY
PRE_KERNEL_1
  __init_efi_console_init: efi_console_init(NULL)
  __init_statics_init_pre: statics_init(NULL)
  __init___device_dts_ord_7: loapic_init(__device_dts_ord_7)
  __init___device_dts_ord_12: ioapic_init(__device_dts_ord_12)
  __init_pcie_init: pcie_init(NULL)
  __init_uart_console_init: uart_console_init(NULL)
PRE_KERNEL_2
  __init_sys_clock_driver_init: sys_clock_driver_init(NULL)
POST_KERNEL
  __init_enable_logger: enable_logger(NULL)
  __init_malloc_prepare: malloc_prepare(NULL)
  __init___device_dts_ord_22: uart_ns16550_init(__device_dts_ord_22)
  __init___device_dts_ord_23: uart_ns16550_init(__device_dts_ord_23)
  __init___device_dts_ord_24: uart_ns16550_init(__device_dts_ord_24)
  __init___device_dts_ord_25: uart_ns16550_init(__device_dts_ord_25)
APPLICATION
SMP
```

jhedberg, 2023-11-22:
There doesn't seem to be any reason to keep ns16550 in POST_KERNEL even if PCIe is enabled. The important part is that it's initialised after PCIe, and that will hold true even if both are in PRE_KERNEL_1:
```
EARLY
PRE_KERNEL_1
  __init_efi_console_init: efi_console_init(NULL)
  __init_statics_init_pre: statics_init(NULL)
  __init___device_dts_ord_7: loapic_init(__device_dts_ord_7)
  __init___device_dts_ord_12: ioapic_init(__device_dts_ord_12)
  __init_pcie_init: pcie_init(NULL)
  __init___device_dts_ord_22: uart_ns16550_init(__device_dts_ord_22)
  __init___device_dts_ord_23: uart_ns16550_init(__device_dts_ord_23)
  __init___device_dts_ord_24: uart_ns16550_init(__device_dts_ord_24)
  __init___device_dts_ord_25: uart_ns16550_init(__device_dts_ord_25)
  __init_uart_console_init: uart_console_init(NULL)
PRE_KERNEL_2
  __init_sys_clock_driver_init: sys_clock_driver_init(NULL)
POST_KERNEL
  __init_enable_logger: enable_logger(NULL)
  __init_malloc_prepare: malloc_prepare(NULL)
APPLICATION
SMP
```

I just tried this diff:
```diff
--- a/drivers/serial/uart_ns16550.c
+++ b/drivers/serial/uart_ns16550.c
@@ -1423,9 +1423,7 @@ static const struct uart_driver_api uart_ns16550_driver_api = {
        };                                                                           \
        DEVICE_DT_INST_DEFINE(n, &uart_ns16550_init, NULL,                           \
                              &uart_ns16550_dev_data_##n, &uart_ns16550_dev_cfg_##n, \
-                             COND_CODE_1(CONFIG_UART_NS16550_PARENT_INIT_LEVEL,     \
-                                         (POST_KERNEL), (PRE_KERNEL_1)),            \
-                             CONFIG_SERIAL_INIT_PRIORITY,                           \
+                             PRE_KERNEL_1, CONFIG_SERIAL_INIT_PRIORITY,             \
                              &uart_ns16550_driver_api);                             \
        UART_NS16550_PCIE_IRQ_FUNC_DEFINE(n)
```

With the above change the test passes fine, although with (expected garbled output):
```
*** Booting Zephyr OS build zephyr-v3.5.0-1947-g7be1a8119bfd ***
Running TESTSUITE log_immediate
===================================================================
START - test_log_immediate_preemption
[00:00:00.028,000] <inf> test: CONFIG_LOG_IMMEDIATE_CLEAN_OUTPUT not enabled. Text output will be garbled.
...quite a lot of garbage...
  [ 0 0 0 PASS - test_log_immediate_preemption in 0.313 seconds
===================================================================
TESTSUITE log_immediate succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [log_immediate]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.313 seconds
 - PASS - [log_immediate.test_log_immediate_preemption] duration = 0.313 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

PR #65428 (2023-11-22): drivers: ns16550: remove parent init level dependency
remove parent init level dependency for ns16550 driver such as PRE_KERNEL or POST_KERNEL. ns16550 driver init level change always to PRE_KERNEL and also removed Alder Lake board configuration with parent init dependency configuration.

Fixes #62717
Fixes #62761
Fixes #65516
closed 2023-11-22 by fabiobaltieri