zephyrproject-rtos/zephyr #62720: Thread safety of the random subsystem

hubertmis, 2023-09-15:
**Describe the bug**

All of the threads in Zephyr RTOS share a common random number subsystem instance to generate random numbers and CSPRNGs. RNG and CSPRNG backends are selected independently.

Some of these backends are thread-safe:

* ctr_drbg (irq_lock)
* timer (spin_lock)

Some are not:

* xoshiro

For others, it depends on the device driver thread safety:

* entropy_device

**Expected behavior**

At least all CSPRNG backends are thread-safe.

It seems currently there are two CSPRNG backends available:

* ctr_drbg
* entropy_device

*ctr_drbg* is already thread-safe. For *entropy_device* we would need to either add some synchronization primitives to the *entropy_device* backend implementation or require all entropy device driver implementations to be thread-safe. I checked a few selected driver implementations and they seem already thread-safe by using spin_locks or irq_locks. So maybe it's just a matter of documenting this requirement in entropy device driver API docs to fix the problem I'm describing.

**Impact**
Security

fabiobaltieri, 2023-09-19:
@ceolin flagging as medium since this may have some security impact, could you take a look?

ceolin, 2023-09-19:
> @ceolin flagging as medium since this may have some security impact, could you take a look?

sure, will take a look.

ceolin, 2023-09-21:
@hubertmis thanks for reporting it. I see some items that need to be done: 

1) ctr_dbrg is not thread-safe in SMP systems, so instead of just locking irq we need to use some mechanism that covers smp like a mutex (need to check though if is used in ISR)
2) rand32_entropy_device itself does not need any change and is thread-safe assuming that the entropy driver is thread-safe, so we need to make it a requirement

hubertmis, 2023-09-22:
@ceolin, thanks for looking at this!

Do you think we should also make non-CS PRNGs thread safe (xoshiro)? If not maybe we should explicitly state in the rand subsystem docs they are not and describe the potential side effects of using them from different contexts.

ceolin, 2023-09-22:
> @ceolin, thanks for looking at this!

Of course !

> 
> Do you think we should also make non-CS PRNGs thread safe (xoshiro)? If not maybe we should explicitly state in the rand subsystem docs they are not and describe the potential side effects of using them from different contexts.

My initial idea is to make both thread safe to avoid further problems.

ceolin, 2023-09-22:
@hubertmis I've linked a pr that address this issue. I will still look xoshiro. I avoided just put a lock there because it seems that only one small piece can be problematic and we could make it work using TLS, so I will think a little bit more.

I found other things in the subsys that require some love too and will work on them tomorrow.

closed 2023-09-27 by MaureenHelm