zephyrproject-rtos/zephyr #62723: Twister: Unconnected platforms are not filtered out with --test-only and --hardware-map option

mbrunnen, 2023-09-15:
**Describe the bug**
When I run twister tests with the options `--test-only` and `--hardware-map`, all built tests are attempted. This is not intended, according to this comment in [testplan.py](https://github.com/zephyrproject-rtos/zephyr/blob/4856fd4cb6ce13d7218d7993cce658463ef702c0/scripts/pylib/twister/twisterlib/testplan.py#L215). The option `--platform` was not set in the command line.

> Get list of connected hardware and filter tests to only be run on connected hardware
> in cases where no platform was specified when running the tests.
> If the platform does not exist in the hardware map, just skip it.

Instead I see the test failed:

```
ERROR   - No device to serve as frdm_k64f platform.
ERROR   - frdm_k64f                 tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c     FAILED : No device to serve as frdm_k64f platform.
```

**Use Case**

I have a "build-runner", which builds all tests and a "test-runner", which tests a subset of these tests on the according hardware. Therefore I need separate calls with `--build-only` and `--test-only`.
 
**To Reproduce**
Steps to reproduce the behavior:

1. Build tests in the zephyr repository, any test root should do: 
```sh
west twister \
        --build-only \
        --testsuite-root tests/boards/frdm_k64f
```
2. [Generate a hardware map](https://docs.zephyrproject.org/3.4.0/guides/test/twister.html#executing-tests-on-multiple-devices) for `frdm_k64f` and set it `connected: false`:
```yaml
- platform: frdm_k64f
  connected: false
  id: '000000000000'
  product: J-Link
  runner: jlink
  serial: /dev/ttyUSB0
```

3. Now run the tests:
```sh
west twister \
      --test-only \
      --integration \
      --device-testing \
      --hardware-map hardware-map.yml \
      --testsuite-root tests/boards/frdm_k64f
```

**Expected behavior**

Filter tests to only be run on connected hardware, according to the hardware map. Tests on `integration_platforms`, which are not connected or even mentioned in the hardware map, should be skipped.

**Impact**

Increased testing time, since quarantine lists are also not working with `--test-only` option. [apply_filters](https://github.com/zephyrproject-rtos/zephyr/blob/4856fd4cb6ce13d7218d7993cce658463ef702c0/scripts/pylib/twister/twisterlib/testplan.py#L227) is not called, when `--test-only` is set.

**Logs and console output**
Build log:
```
west twister \
        --build-only \
        --testsuite-root tests/boards/frdm_k64f
Renaming output directory to /home/mbrunnen/projects/src/firmware/zephyr/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3801-g4856fd4cb6ce
INFO    - Using 'zephyr' toolchain.
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/testplan.json
INFO    - JOBS: 16
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - frdm_k64f                 tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c     ERROR : Build failure
ERROR   - see: /home/mbrunnen/projects/src/firmware/zephyr/twister-out/frdm_k64f/tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c/build.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0, error:    1
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 0 failed, 1 errored, 0 skipped with 0 warnings in 3.13 seconds
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c on frdm_k64f error (Build failure)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p frdm_k64f -s tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c
INFO    - or with west:
INFO    - west build -p -b frdm_k64f -T tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

Test log:
```
west twister \
      --test-only \
      --integration \
      --device-testing \
      --hardware-map hardware-map.yml \
      --testsuite-root tests/boards/frdm_k64f
Keeping artifacts untouched
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3801-g4856fd4cb6ce
INFO    - Using 'zephyr' toolchain.

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|

INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - No device to serve as frdm_k64f platform.
ERROR   - frdm_k64f                 tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c     FAILED : No device to serve as frdm_k64f platform.
ERROR   - see: /home/mbrunnen/projects/src/firmware/zephyr/twister-out/frdm_k64f/tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c/build.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    1, error:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 errored, 0 skipped with 0 warnings in 0.77 seconds
INFO    - In total 3 test cases were executed, 0 skipped on 0 out of total 622 platforms (0.00%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.

Hardware distribution summary:

| Board   | ID   | Counter   |
|---------|------|-----------|
INFO    - Saving reports...
INFO    - Writing JSON report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/mbrunnen/projects/src/firmware/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c on frdm_k64f failed (No device to serve as frdm_k64f platform.)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p frdm_k64f -s tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c
INFO    - or with west:
INFO    - west build -p -b frdm_k64f -T tests/boards/frdm_k64f/i2c/boards.frdm_k64f.i2c
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

**Environment:**
 - OS: Linux
 - Toolchain: zephyr 0.16.1
 - Zephyr: `main` or 4856fd4cb6ce13d7218d7993cce658463ef702c0
 - target platform: `frdm_k64f`, but the hardware is not important

mbrunnen, 2023-09-15:
It would work as intended if [HardwareMap.discover()](https://github.com/zephyrproject-rtos/zephyr/blob/4856fd4cb6ce13d7218d7993cce658463ef702c0/scripts/pylib/twister/twisterlib/hardwaremap.py#L164) wouldn't populate the command line option `platform`. But I do not know how to fix that.

nashif, 2023-09-18:
duplicate of #62560

closed 2023-09-18 by nashif

mbrunnen, 2023-09-18:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/62671