zephyrproject-rtos/zephyr #62738: zbus: faulty __ASSERT when publishing and subscriber queue is full and timeout is expired

vai-mikar, 2023-09-16:
**Describe the bug**

Assertion at [`_zbus_vded_exec`](https://github.com/zephyrproject-rtos/zephyr/blob/c7255cf3749a122c30f82a7077f06ac87431b4bf/subsys/zbus/zbus.c#L88-L90) hits on "normal timeout error cases". Instead of assertion failure,  an error code should be returned to the caller of `zbus_chan_pub` as documented in the [API documentation](https://docs.zephyrproject.org/latest/services/zbus/index.html#c.zbus_chan_pub).

That assertion should maybe be completely removed, or at least it should not consider  `-ENOMSG` or `-EAGAIN` as a failure. The code is okay when assertions are not enabled.

**More details**

When publishing by calling `zbus_chan_pub`, notifying subscribers is done by [`_zbus_vded_exec`](https://github.com/zephyrproject-rtos/zephyr/blob/c7255cf3749a122c30f82a7077f06ac87431b4bf/subsys/zbus/zbus.c#L62
).

That in turn calls [`_zbus_notify_observer`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/zbus/zbus.c#L86). 

Finally, [that functions calls](https://github.com/zephyrproject-rtos/zephyr/blob/c7255cf3749a122c30f82a7077f06ac87431b4bf/subsys/zbus/zbus.c#L55) `k_msgq_put` to notify a subscriber. The timeout given by the caller of `zbus_chan_pub` is carried all along there.

Now, assume that the subscriber queue is full. `k_msgq_put` returns either `-ENOMSG` or `-EAGAIN`. This is what `zbus_chan_pub` should return as well.

However, [there's an assertion in `_zbus_vded_exec`](https://github.com/zephyrproject-rtos/zephyr/blob/c7255cf3749a122c30f82a7077f06ac87431b4bf/subsys/zbus/zbus.c#L88-L90) which fires in such cases preventing `zbus_chan_pub` to return an error the the caller.

**Logs and console output**

The code calling `zbus_chan_pub` with `K_NO_WAIT` can see this happening:
```
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/zbus/zbus.c:88
	could not deliver notification to observer . Error code -35
[00:33:58.346,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000058  r2/a3:  0x0000000e
[00:33:58.346,000] <err> os: r3/a4:  0x20044770 r12/ip:  0x00001a4a r14/lr:  0x08021b25
[00:33:58.346,000] <err> os:  xpsr:  0x41000000
[00:33:58.346,000] <err> os: Faulting instruction address (r15/pc): 0x0805edb6
[00:33:58.346,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:33:58.346,000] <err> os: Current thread: 0x20046d48 (nimbusrapp_id)
[00:33:58.370,000] <err> os: Halting system
```

**Expected behavior**
Error code should be returned to the API caller even when assertions are turned on.

**Impact**
Kernel panic when asserts are enabled.

**Environment**
 - Commit SHA or Version used c7255cf3749a122c30f82a7077f06ac87431b4bf

vai-mikar, 2023-09-16:
Unit test demonstrating the issue in commit https://github.com/mkaranki/zephyr/commit/4d8b98ab30a99619d4dcfc3c6eb140f49f69f414

rodrigopex, 2023-09-18:
@vai-mikar, thank you for your contribution. 

The `__ASSERT` there is not a defect. It is there intentionally. Notice the issue you are pointing out is a design problem that must be solved during development, correct? 

In production, the assert will be disabled, and the error code will be returned as expected If you don't solve the design problem. I strongly suggest solving that before production. The assert is there to help. 

With ZBus, we should always avoid this notification issue to keep consistency. The message queue for the subscriber must be big enough to fit all the "simultaneous" notifications until the subscriber can consume them. The timeout should also be sufficient to make the message queue available again.

If you have a different view of the situation, let me know. There is no single truth in life. I love to hear other points of view. :smile:

rodrigopex, 2023-09-18:
> Unit test demonstrating the issue in commit [mkaranki@4d8b98a](https://github.com/mkaranki/zephyr/commit/4d8b98ab30a99619d4dcfc3c6eb140f49f69f414)

Regarding the unit test, the `_ZBUS_ASSERT` always returns `-EFAULT` when the condition is false. Maybe you could not see that. This is why the `-42` is never returned.

vai-mikar, 2023-09-18:
> If you have a different view of the situation, let me know. There is no single truth in life. I love to hear other points of view. 😄

Thanks, I'll try. :)

I think that the handling of run time errors should not differ whether assertions are on or off, and that subscriber queue getting full is a run-time error, not a programming logic error per se.

As far as I know, Zephyr hasn't defined what an assertion means, but I have learned that most commonly it is a predicate that is always thought to be true when the program works as expected. Assertion failure is always a sign of a bug - never a way to perform error handling. Errors may happen, might be even expected, there might be reasonable run-time actions to do when those happens, and they are not a sign of a bug in the software.

Now, lets think an example application where a producer thread is reading measurements from a sensor and publishing those to zbus channel. Then there's a consumer that is sending the measurement somewhere over the internet.

Sending data over the network has often non-deterministic delays. It could be a valid design to use a subscriber queue as a buffer to compensate varying speed of the network connection. Sometimes the network might just be too slow and the queue gets full nevertheless how big it is. That is not necessarily a programming error. It could of course be if there is some other throttling mechanism in place that should always prevent the situation. But in that case I think the caller of `zbus_chan_pub` should be responsible of the assertion regarding the function return code, not the API function itself, as some other application could just start dropping the data if it is only promising best-effort delivery of it, or throttle the sensor reading to happen slower (e.g. by increasing the timeout of `zbus_chan_pub` on next iteration), and work "correctly" by that.

Now, with the assertions enabled build, run-time behavior of recovering from queue full situations cannot be tested (yeah, I noticed there possible to mock the zbus assertions for that, but that is changing the error code zbus_chan_pub returns compared to the production build. We could live with that, and additional -EFAULT check, but that could turn other very useful assertions off?).

Another reason is API consistency: `k_msgq_put`, for instance,  is not failing an assertion when putting a new item to queue fails due to timeout. Why `zbus_chan_pub` is good to behave differently? It surprised at least me a little.

I'll close the issue as not planned as the assert is not just accidentally there. Thank you for the zbus anyways. It's pretty neat way to decouple producers and consumers, especially with the help of recently added `ZBUS_CHAN_ADD_OBS` macro.

closed 2023-09-18 by vai-mikar

rodrigopex, 2023-09-18:
@vai-mikar, thank you for the insight. 

You are right when you say that ZBus differs from the `k_msgq_put. This is because ZBus has a much more complex message delivery net topology. In ZBus, missing a notification delivery may indicate a bad thing. So, the users must rely on that, trusting the messages will be delivered. 

In your case, it is better to use a different approach. If you cannot lose a message, use a listener to aggregate that. After aggregating, you can use an additional channel to send that to the internet. I like to work with a channel pair `send_packet` and `transmission_done` to which I submit the payload to be sent and await confirmation on the `transmission_done` channel. The aggregation buffer is large enough to deal with faulty communication a few times. I can retry to send the data as well.

> Thank you for the zbus anyways. It's pretty neat way to decouple producers and consumers, especially with the help of recently added ZBUS_CHAN_ADD_OBS macro.

Thank you very much for the feedback. It makes my work worth it.  :handshake: