zephyrproject-rtos/zephyr #62741: ppp driver to change baud rate with AT+IPR

LubomirBogdanov, 2023-09-16:
**Is your feature request related to a problem? Please describe.**
Our team wants to send tens of kilobytes of data through a GSM modem. However, the sending lasts 65 kB for 14 seconds and this drains the battery of the device that we're developing. 

**Describe the solution you'd like**
I know that our modem supports 1000 000 bauds and that our GSM network provider supports high upload rates. 
If you could add an API function that changes the default PPP communication baud rate at run time, this would be great and would solve our problem. 
The AT command is: AT+IPR=1000000

**Describe alternatives you've considered**
Currently the PPP driver takes 115200 bauds from the DTB which is the default rate for the GSM modem. If we simply change this rate, we would have to re-flash the GSM modem but we do not have AT firmware for it, so we're stuck.

We tried writing such function on our own like that:
`int gsm_ppp_extended_change_baud(struct gsm_modem *gsm){
	int ret = 0;

	ret = modem_cmd_send_nolock(
			&gsm->context.iface,
			&gsm->context.cmd_handler,
			NULL, 0,
			"AT+IPR=1000000\r\n",
			&gsm->sem_response,
			K_SECONDS(8));

	return ret;
}`

and then on the host microcontroller side (ESP32):

`	const struct device *const uart = DEVICE_DT_GET(DT_NODELABEL(uart1));
	cfg.baudrate = 1000000;
	cfg.parity = UART_CFG_PARITY_NONE;
	cfg.stop_bits = UART_CFG_STOP_BITS_1;
	cfg.data_bits = UART_CFG_DATA_BITS_8;
	cfg.flow_ctrl = UART_CFG_FLOW_CTRL_NONE;

	int error = uart_configure(uart, &cfg);
	if (error) {
		LOG_ERR("uart_line_ctrl_set failed (error %d)", error);
		return error;
	}`

but this didn't work. It returned error -11 (EAGAIN) and -5 (EIO) when using the websocket_send_msg( ) API. The communication between the ESP32 and the Quectel EG915U does change (we have sniffed the link with an external USB-to-UART dongle) but for some reason we get the errors mentioned above.

**Additional context**
None

LubomirBogdanov, 2023-09-20:
Hello again,
I've managed to change the baud rate at run time by modifying the PPP driver with the code in my original post. However, there is one important thing to note:

the command AT+IPR=1000000 must be issued before any other commands in the "static void gsm_configure(struct k_work *work)" API. 

Looks like some of the modem initialization commands disable the feature of baud rate changing. Currently we are using Quectel EG915U modem.

Here is my exact code:

void gsm_ppp_set_1Mbit(void){
	int ret = 0;
	struct uart_config cfg;
	const struct device *const uart = DEVICE_DT_GET(DT_NODELABEL(uart1));

	ret = modem_cmd_send_nolock(
			&gsm.context.iface,
			&gsm.context.cmd_handler,
			NULL, 0,
			"AT+IPR=1000000",
			&gsm.sem_response,
			K_SECONDS(8));

	LOG_INF("gsm_ppp_set_1Mbit %d", ret);

	cfg.baudrate = 1000000;
	cfg.parity = UART_CFG_PARITY_NONE;
	cfg.stop_bits = UART_CFG_STOP_BITS_1;
	cfg.data_bits = UART_CFG_DATA_BITS_8;
	cfg.flow_ctrl = UART_CFG_FLOW_CTRL_NONE;

	ret = uart_configure(uart, &cfg);
	if (ret) {
		LOG_ERR("uart_line_ctrl_set failed (error %d)", ret);

	}

	k_msleep(500);

	uart_config_get(uart, &cfg);
	LOG_INF("Testing modem baud rate ... %d", cfg.baudrate);

	ret = modem_cmd_send_nolock(
			&gsm.context.iface,
			&gsm.context.cmd_handler,
			NULL, 0,
			"АТI",
			&gsm.sem_response,
			GSM_CMD_AT_TIMEOUT);

	k_msleep(3000);
}`
`
and I invoke the gsm_ppp_set_1Mbit( ) on the first line of gsm_configure( ) in zephyr/drivers/modem/gsm_ppp.c

rerickson1, 2023-09-20:
Hello, the GSM PPP driver is now deprecated. I would recommend basing your driver off of the new `cellular_modem.c` driver for the best support going forward.
https://github.com/zephyrproject-rtos/zephyr/pull/62777

LubomirBogdanov, 2023-09-21:
OK, thanks. But where can I find it? I've cloned the latest and greatest version of Zephyr but it is not in the source tree ...

LubomirBogdanov, 2023-09-21:
Oh, I found it in zephyr/drivers/modem/modem_cellular.c

closed 2024-07-10 by rerickson1