zephyrproject-rtos/zephyr #62753: lvgl: hello world sample crashes

faxe1008, 2023-09-17:
**Describe the bug**
Running the zephyr lvgl sample on any commit after `c7292efbda9902579966657ae5bc514bcda7033b` (target: native_posix) leads to a  segfault.

Please also mention any information which could help others to understand
the problem you're facing:
- The issue is that the drivers are now initialized before the default display [lvgl.c#L252](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/lvgl.c#L252). LVGL requires the presence of the display before any drivers which are group and therefore display related are instantiated.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b native_posix -d build vendor/zephyr/samples/subsys/display/lvgl
2. west build -t run -d build 

**Expected behavior**
The sample should not crash.

**Impact**
Encoder devices are not usable anymore.

faxe1008, 2023-09-17:
@gmarull what would be the prefered way of resolving this? Undoing the change or moving the other lvgl setup code to `POST_KERNEL`?

nashif, 2023-09-17:
@gmarull FYI

closed 2023-09-18 by carlescufi