zephyrproject-rtos/zephyr #62761: upsquared: tests/subsys/logging/log_core_additional/ fail

KamilxPaszkiet, 2023-09-18:
**Describe the bug**
Out of 3 tests in the test suite, 2 fail.
`tests/subsys/logging/log_core_additional/logging.add.async`
and
`tests/subsys/logging/log_core_additional/logging.add.sync`
fails.

**To Reproduce**
```twister --device-testing --hardware-map up2.map -T tests/subsys/logging/log_core_additional/ -vv```

**Logs and console output**
```
2023-09-18 09:43:15,647 - twister - DEBUG - DEVICE: >>Checking Media Presence......
2023-09-18 09:43:15,649 - twister - DEBUG - DEVICE: >>Media Present......
2023-09-18 09:43:15,652 - twister - DEBUG - DEVICE: Downloading NBP file...
2023-09-18 09:43:16,024 - twister - DEBUG - DEVICE: 
2023-09-18 09:43:16,027 - twister - DEBUG - DEVICE: Succeed to download NBP file.
2023-09-18 09:43:16,095 - twister - DEBUG - DEVICE: *** Zephyr EFI Loader ***
2023-09-18 09:43:16,097 - twister - DEBUG - DEVICE: RSDP found at 0x79c34000
2023-09-18 09:43:16,101 - twister - DEBUG - DEVICE: Zeroing 4902936 bytes of memory at 0x115000
2023-09-18 09:43:16,149 - twister - DEBUG - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
2023-09-18 09:43:16,156 - twister - DEBUG - DEVICE: Copying 86016 data bytes to 0x100000 from image offset 32768
2023-09-18 09:43:16,162 - twister - DEBUG - DEVICE: Copying 2265056 data bytes to 0x5c2020 from image offset 118784
2023-09-18 09:43:16,188 - twister - DEBUG - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
2023-09-18 09:43:16,430 - twister - DEBUG - DEVICE: ASSERTION FAIL [device_is_ready(uart_dev)] @ WEST_TOPDIR/zephyr/subsys/logging/backends/log_backend_uart.c:127
2023-09-18 09:43:57,267 - twister - DEBUG - Timed out while monitoring serial output on up_squared
2023-09-18 09:43:57,371 - twister - DEBUG - Expected suite names:['test_log_core_additional', 'test_log_core_additional']
2023-09-18 09:43:57,371 - twister - DEBUG - Detected suite names:[]
2023-09-18 09:43:57,372 - twister - DEBUG - run status: up_squared/tests/subsys/logging/log_core_additional/logging.add.async failed
2023-09-18 09:43:57,377 - twister - INFO - 3/3 up_squared                tests/subsys/logging/log_core_additional/logging.add.async  FAILED Timeout (device: up2, 63.638s)
2023-09-18 09:43:57,378 - twister - ERROR - see: /home/ztest/zephyrproject/zephyr/twister-out/up_squared/tests/subsys/logging/log_core_additional/logging.add.async/handler.log
2023-09-18 09:43:57,388 - twister - INFO - 3 test scenarios (3 test instances) selected, 1 configurations skipped (1 by static filter, 0 at runtime).
2023-09-18 09:43:57,388 - twister - INFO - 0 of 3 test configurations passed (0.00%), 2 failed, 0 errored, 1 skipped with 0 warnings in 145.58 seconds
2023-09-18 09:43:57,388 - twister - INFO - In total 20 test cases were executed, 16 skipped on 1 out of total 617 platforms (0.16%)
2023-09-18 09:43:57,389 - twister - INFO - 2 test configurations executed on platforms, 0 test configurations were only built.
2023-09-18 09:43:57,390 - twister - INFO - Saving reports...
2023-09-18 09:43:57,390 - twister - INFO - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/twister.json
2023-09-18 09:43:57,393 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
2023-09-18 09:43:57,402 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
2023-09-18 09:43:57,412 - twister - INFO - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
2023-09-18 09:43:57,412 - twister - INFO - The following issues were found (showing the top 10 items):
2023-09-18 09:43:57,412 - twister - INFO - 1) tests/subsys/logging/log_core_additional/logging.add.async on up_squared failed (Timeout)
2023-09-18 09:43:57,412 - twister - INFO - 2) tests/subsys/logging/log_core_additional/logging.add.sync on up_squared failed (Timeout)
```

**Environment:**
- OS: Linux
- Toolchain: zephyr-sdk-0.16.0
- zephyr-v3.4.0-3801-g4856fd4cb6ce

nashif, 2023-10-24:
this is the most relevant line:
`DEVICE: ASSERTION FAIL [device_is_ready(uart_dev)] @ WEST_TOPDIR/zephyr/subsys/logging/backends/log_backend_uart.c:127`

Someone, (@najumon1980?) has changed init levels and we now have uart console start before the uart driver. Also, it seems we have uart driver come up in post kernel, which is very bad.

KamilxPaszkiet, 2023-10-26:
This [PR](https://github.com/zephyrproject-rtos/zephyr/pull/64362) fix this issue

PR #65428 (2023-11-22): drivers: ns16550: remove parent init level dependency
remove parent init level dependency for ns16550 driver such as PRE_KERNEL or POST_KERNEL. ns16550 driver init level change always to PRE_KERNEL and also removed Alder Lake board configuration with parent init dependency configuration.

Fixes #62717
Fixes #62761
Fixes #65516
closed 2023-11-22 by fabiobaltieri