zephyrproject-rtos/zephyr #62772: samples: subsys: pm: device_pm: sample.power.ospm.dev_idle_pm fails

katgiadla, 2023-09-18:
**Describe the bug**
The test `sample.power.ospm.dev_idle_pm` fails.

Observed for:
- `nrf52dk_nrf52832`
- `nrf52840dk_nrf52840`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/pm/device_pm/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-09-18T03:40:26.192Z] ERROR   - parent not found
[2023-09-18T03:40:26.192Z] child suspending..
[2023-09-18T03:40:26.192Z] parent suspending..
[2023-09-18T03:40:26.192Z] *** Booting Zephyr OS build zephyr-v3.4.0-3830-gad8f3e5b4e28 ***
[2023-09-18T03:40:26.192Z] Device PM sample app start
[2023-09-18T03:40:26.192Z] child resuming..
[2023-09-18T03:40:26.192Z] Dummy device resumed
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.0
 - Commit SHA or Version used: zephyr-v3.4.0-3830-gad8f3e5b4e28

gmarull, 2023-09-18:
could you send a PR deleting such sample? it looks like it adds no value, there are already tests and docs covering the same functionality.

nashif, 2023-09-20:
commit that broke this:

4dd7bc997ffa6480afea93bb47236df04d36f7c4

nashif, 2023-09-20:
> could you send a PR deleting such sample? it looks like it adds no value, there are already tests and docs covering the same functionality.

There is value and probably it needs some love. I do not understand why it was locked on nordic hardware, this sample runs fine in qemu.

PR #62836 (2023-09-21): samples: pm: fix init priority after init level change
Moving from APPLICATION level to POST_KERNEL and keeping both parent and
child on the same level is causing problems, so fix this by putting them
on different priorities in POST_KERNEL.

Also, this sample runs fine in qemu, so make it the integration
platform.

Fixes #62772
commit 42d8f7ff (2023-09-21): samples: pm: fix init priority after init level change

Moving from APPLICATION level to POST_KERNEL and keeping both parent and
child on the same level is causing problems, so fix this by putting them
on different priorities in POST_KERNEL.

Also, this sample runs fine in qemu, so make it the integration
platform.

Fixes #62772
closed 2023-09-21 by carlescufi