zephyrproject-rtos/zephyr #62806: EFM32PG and EFM32WG broken by latest Silabs hal update

fabiobaltieri, 2023-09-19:
**Describe the bug**
Hi, looks like the latest Silabs HAL update broke the build for a couple of platforms (maybe more), these are failing in CI:

```
$ west build -p -b efm32pg_stk3401a samples/basic/blinky
...
CMake Error at /home/fabiobaltieri/zephyrproject/modules/hal/silabs/gecko/CMakeLists.txt:54 (message):
  Unknown gecko series type.  Is CONFIG_SOC_GECKO_SERIES1/2 set?
```

```
$ west build -p -b efm32wg_stk3800 samples/basic/blinky
...
CMake Error at /home/fabiobaltieri/zephyrproject/modules/hal/silabs/gecko/CMakeLists.txt:54 (message):
  Unknown gecko series type.  Is CONFIG_SOC_GECKO_SERIES1/2 set?
```

The specific change requires a config option to be set for all EFM32 (https://github.com/zephyrproject-rtos/hal_silabs/pull/35/commits/f8b233f281284dd6404b7143306b927509e0a7f8) but, few of them have none.

Could you look into this? It's breaking the main CI push so it would be a hotfix.

**To Reproduce**
See the build error above and https://github.com/zephyrproject-rtos/zephyr/runs/16908324631

**Expected behavior**
Builds correctly.

**Impact**
CI breakage, potentially blocking other PRs.

fabiobaltieri, 2023-09-19:
cc @yonsch @tgorochowik @sateeshkotapati @mnkp

fkokosinski, 2023-09-19:
Thanks for reporting that! I'm working on it.

commit 8131a67d (2023-09-19): soc/arm/silabs_exx32: select missing `CONFIG_SOC_GECKO_SERIESx`

This commit:
* adds the `CONFIG_SOC_GECKO_SERIES0` Kconfig option for Gecko Series 0
  SoCs
* selects the proper `CONFIG_SOC_GECKO_SERIESx` option where it's currently
  missing

Fixes #62806.
closed 2023-09-19 by nashif