zephyrproject-rtos/zephyr #62819: tests/posix/common/portability.posix.common.picolibc broken on main on some platforms

aescolar, 2023-09-19:
**Describe the bug**
tests/posix/common/portability.posix.common.picolibc fails to build on some platforms

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake ../tests/posix/common/ -DBOARD=native_sim -GNinja -DCONFIG_PICOLIBC=y
3. ninja
4. See error

**Expected behavior**
No build errors

**Impact**
CI blocked when these tests are triggered

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/6235662389/job/16925534892#step:13:1650
```
 in function `get_posix_cond': lib/posix/cond.c:54: undefined reference to `log_const_pthread_cond'
 in function `to_posix_cond': lib/posix/cond.c:79: undefined reference to `log_const_pthread_cond'
 in function `cond_wait': lib/posix/cond.c:105: undefined reference to `log_const_pthread_cond'
/usr/bin/ld: lib/posix/cond.c:108: undefined reference to `log_const_pthread_cond'
 in function `pthread_cond_signal': lib/posix/cond.c:131: undefined reference to `log_const_pthread_cond'
lib/posix/cond.c:153: more undefined references to `log_const_pthread_cond' follow
 in function `to_posix_mutex': lib/posix/mutex.c:91: undefined reference to `log_const_pthread_mutex'
 in function `get_posix_mutex': lib/posix/mutex.c:66: undefined reference to `log_const_pthread_mutex'
 in function `acquire_mutex': lib/posix/mutex.c:131: undefined reference to `log_const_pthread_mutex'
/usr/bin/ld: lib/posix/mutex.c:136: undefined reference to `log_const_pthread_mutex'
/usr/bin/ld: lib/posix/mutex.c:144: undefined reference to `log_const_pthread_mutex'
lib/posix/mutex.c:149: more undefined references to `log_const_pthread_mutex' follow
 in function `is_posix_policy_prio_valid': lib/posix/pthread.c:145: undefined reference to `log_const_pthread'
 in function `to_posix_thread': lib/posix/pthread.c:102: undefined reference to `log_const_pthread'
 in function `pthread_attr_setschedparam': lib/posix/pthread.c:195: undefined reference to `log_const_pthread'
 in function `pthread_attr_setstack': lib/posix/pthread.c:213: undefined reference to `log_const_pthread'
 in function `pthread_attr_is_valid': lib/posix/pthread.c:237: undefined reference to `log_const_pthread'
lib/posix/pthread.c:416: more undefined references to `log_const_pthread' follow 
collect2: error: ld returned 1 exit status
```

**Environment (please complete the following information):**
 - OS: Linux CI, or Ubuntu 22.04
 - Toolchain gcc 11 or 12
 - Commit SHA or Version used: main after https://github.com/zephyrproject-rtos/zephyr/pull/62771 

**Additional context**
Introduced by https://github.com/zephyrproject-rtos/zephyr/pull/62771

cfriedt, 2023-09-19:
This particular configuration is probably filtered-out in `testcase.yaml`, since it's passing CI for all other platforms.

cfriedt, 2023-09-19:
I'll take a quick look to see if I can fix it.

aescolar, 2023-09-19:
> This particular configuration is probably filtered-out in testcase.yaml, since it's passing CI for all other platforms.

Nope. Note the linked CI failure.

> I'll take a quick look to see if I can fix it.

Thanks :)

cfriedt, 2023-09-19:
Ah, it was an easy fix. Will make a PR.

cfriedt, 2023-09-19:
> > This particular configuration is probably filtered-out in testcase.yaml, since it's passing CI for all other platforms.
> 
> Nope. Note the linked CI failure.

right - .. well it passed CI for code that was already merged. I can't make CI run with the ability to predict what future code might be merged.

https://github.com/zephyrproject-rtos/zephyr/pull/62771/checks

PR #62821 (2023-09-19): posix: logging: use log module register instead of declare
Log modules should be registered with LOG_MODULE_REGISTER rather than LOG_MODULE_DECLARE. It seems the latter works on most platforms (at least with in minimal mode as configured with ZTest).

Fixes #62819
closed 2023-09-19 by fabiobaltieri