zephyrproject-rtos/zephyr #62828: rx_timeout_slab shouldn't consider CONFIG_SYS_CLOCK_TICKS_PER_SEC when hardware timers are used

pres2300, 2023-09-19:
I'm using hardware timers for an ASYNC UART implementation on an nRF52840.  I have a very small timeout I want to use (28us) for the driver to notify with `UART_RX_RDY`.  Because this timeout is small, the current code implementation in `uart_nrfx_uarte.c` (referenced below) can potentially set `rx_timeout_slab` to a much greater value than `rx_timeout` if `CONFIG_SYS_CLOCK_TICKS_PER_SEC` is a relatively low value.  I've noticed that when this happens you can get spurious `UART_RX_RDY` events.  That can be explained due to these lines of code (1045-1055):

```
               if (clipped ||
			(data->async->rx_timeout_left
				< data->async->rx_timeout_slab)) {
			/* rx_timeout us elapsed since last receiving */
			notify_uart_rx_rdy(dev, len);
			data->async->rx_offset += len;
			data->async->rx_total_user_byte_cnt += len;
		} else {
			data->async->rx_timeout_left -=
				data->async->rx_timeout_slab;
		}
```

The `else` condition above will set `rx_timeout_left` to a much smaller value than expected causing `notify_uart_rx_rdy` to be called on the next iteration.

This can be fixed by modifying the lines where `rx_timeout_slab` gets set to check for the use of hardware timers.  I'm assuming that when hardware timers are used the consideration for sys tick time need not be made.  I ran a test on my setup by quickly changing the code to set `rx_timeout_slab` to the following:
`data->async->rx_timeout_slab = timeout / RX_TIMEOUT_DIV;`

It appears to work, but I'm not sure how best to check for the use of hardware timers in all scenarios.

https://github.com/zephyrproject-rtos/zephyr/blob/2ea2d37d60ada33bdf051eb9d699f4db9123c3ff/drivers/serial/uart_nrfx_uarte.c#L855

henrikbrixandersen, 2023-09-19:
If you are reporting a bug (difficult to tell from the description above), please use our bug template. You can edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

anangl, 2023-10-16:
> From looking at the code, the current code implementation in `uart_nrfx_uarte.c` (referenced below) can potentially set `rx_timeout_slab` to a much greater value than `rx_timeout` if `CONFIG_SYS_CLOCK_TICKS_PER_SEC` is a relatively low value. I've noticed that when this happens you can get spurious `UART_RX_RDY` events. That can be explained due to these lines of code (1045-1055):
> 
> [..]
> 
> The `else` condition above will set `rx_timeout_left` to a much smaller value than expected causing `notify_uart_rx_rdy` to be called on the next iteration.

The actual problem here is that when `rx_timeout_slab` is greater than `rx_timeout_left` (and it can be because of that rounding up to 3 system clock ticks), the timeout is reported (i.e. the `UART_RX_RDY` is generated) in every entry to the `rx_timeout` function, even if new data is continuously received. And this is obviously not the way that `UART_RX_RDY` is supposed to be used.

> This can be fixed by modifying the lines where `rx_timeout_slab` gets set to check for the use of hardware timers. I'm assuming that when hardware timers are used the consideration for sys tick time need not be made.

This assumption is incorrect. Hardware timers are used (as counters) for counting of incoming bytes, not for checking the RX timeouts - for this, always the kernel timers are used.

> I ran a test on my setup by quickly changing the code to set `rx_timeout_slab` to the following:
> `data->async->rx_timeout_slab = timeout / RX_TIMEOUT_DIV;`
> 
> It appears to work, but I'm not sure how best to check for the use of hardware timers in all scenarios to make an official fix for this issue

Yes, that works and it should be used always, no matter if a hardware timer is used for counting of incoming bytes or not.

PR #64094 (2023-10-21): drivers: uart: nrf: rx_timeout_slab incorrectly set
When rx_timeout is set to a sufficiently small value, rx_timeout_slab could potentially get set to a greater than necessary value that causes spurious UART_RX_RDY events.

Fixes #62828
commit 52c55177 (2023-10-21): drivers: uart: nrf: rx_timeout_slab incorrectly set

When rx_timeout is set to a sufficiently small value,
rx_timeout_slab could potentially get set to a greater
than necessary value that causes spurious UART_RX_RDY
events.

Fixes #62828
closed 2023-10-21 by carlescufi