zephyrproject-rtos/zephyr #62829: [TSC] In Source Documentation, Source Modularization

abebeos, 2023-09-19:
**Is your enhancement proposal related to a problem? Please describe.**

I had difficulties to do some (low effort) trial-coding within

- #62752

**Describe the solution you'd like**

### In-Source Docs

- in-source documentation (ideally a README.md per major folder)
  - zephyr/drivers
  - zephyr/drivers/sensors

Describes quickly the architecture and development process (modify/add code)

### Modularization

- a driver should be compilable stand-alone (at minimum with a mock during development)
- ideally, a driver can be linked

henrikbrixandersen, 2023-09-20:
I'd suggest starting with getting familiar with how to do development with Zephyr: https://docs.zephyrproject.org/latest/develop/index.html

Your bullet points about modularization are far-off how the Zephyr build environment works, but patches are always welcome.

closed 2023-09-20 by henrikbrixandersen

abebeos, 2023-09-20:
Wow, issue-tracking-process is really messy.

Giving up here!